lvgl-micropython/lvgl_micropython #111: ST7701 don't works,need examples

jd3096-mpy, 2024-09-04:
Recently, the ST7701 RGB screen seems to be favored by various hardware manufacturers. Therefore, I think it's necessary to specifically test the ST7701 and provide an example. Below is the ST7701 test code I wrote. It seems that there are still some issues with the st7701.py, but even after I corrected these problems, the screen still doesn't show any response after running the code.

`from micropython import const # NOQA
from machine import sleep

_WIDTH = const(480)
_HEIGHT = const(480)

_BUFFER_SIZE = int(_WIDTH * _HEIGHT * 2)

import lcd_bus # NOQA

spi3wire = lcd_bus.SPI3Wire(
sclk=48,
mosi=47,
cs=39,
use_dc_bit=True,
dc_zero_on_data=False,
lsb_first=False,
cs_high_active=False,
keep_cs_inactive=True
)

display_bus = lcd_bus.RGBBus(
hsync=16,
vsync=17,
de=18,
disp=-1,
pclk=21,
data0=11,
data1=12,
data2=13,
data3=14,
data4=0,
data5=8,
data6=3,
data7=46,
data8=9,
data9=10,
data10=20,
data11=4,
data12=5,
data13=6,
data14=7,
data15=15,
freq=12000000
)

import st7701
import lvgl as lv # NOQA

lcd_bus.MEMORY_SPIRAM
fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM)#lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM)#lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

display = st7701.ST7701(
display_bus,
spi3wire,
_WIDTH,
_HEIGHT,
frame_buffer1=fb1,
frame_buffer2=fb2,
backlight_pin=38,
color_space=lv.COLOR_FORMAT.RGB565,
# this is if you are sharing the rgb data lines with the 3wire
bus_shared_pins=False
)

print('olaaa')

display.set_power(True)
display.init()
display.set_backlight(100)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)

import task_handler
th = task_handler.TaskHandler()
`

`AttributeError: 'ST7701' object has no attribute '_power_pin'`

I'm quite certain that the issue lies in the SPI initialization phase of the RGB screen because I tested rgb_display.py, and it works fine for RGB screens that don't require SPI initialization. For the ST7701, if I flash the LVGL MicroPython firmware and run the rgb_display code without powering off after running the test firmware, it still works fine. This indicates a problem in the initialization phase. I also compared it with other ST7701 C language code available online and found some differences in the initialization commands. I attempted to fix them, but the screen still didn't display any image. Therefore, I earnestly request an ST7701 initialization demo that others can also use.
Here's the translation:

I'm also willing to continuously track, debug, and test the ST7701 until it works properly.

kdschlosser, 2024-09-04:
I don't have one of these displays to test with. I have something that is similiar that I can mess about with to get the 3 wire SPI working.

kdschlosser, 2024-09-04:
OK so I found an error in the code.

This is what I want you to do...

Attached are 2 files, you need to remove the `.txt` form the end of the filename. Upload them to your ESP32 like you would any other file. DO NOT change the file names other than removing the `.txt`. Hopefully that should fix the problem.

[display_driver_framework.py.txt](https://github.com/user-attachments/files/16880363/display_driver_framework.py.txt)

[st7701.py.txt](https://github.com/user-attachments/files/16880366/st7701.py.txt)

jd3096-mpy, 2024-09-05:
> OK so I found an error in the code.
> 
> This is what I want you to do...
> 
> Attached are 2 files, you need to remove the `.txt` from the end of the filename. Upload them to your ESP32 like you would any other file. DO NOT change the file names other than removing the `.txt`. Hopefully that should fix the problem.
> 
> [display_driver_framework.py.txt](https://github.com/user-attachments/files/16880363/display_driver_framework.py.txt)
> 
> [st7701.py.txt](https://github.com/user-attachments/files/16880366/st7701.py.txt)

After updating these two Python files, there are indeed no more errors, but the old problem persists: the screen still doesn't display anything. I further confirmed that the issue is still with SPI initialization. The reason is that when I deliberately enter incorrect SPI pin numbers, flash the LVGL test firmware first, and then run the Python file, the screen displays normally. However, when I enter the correct SPI pin numbers, the screen does not display anything again. This proves that the SPI initialization is indeed running but causing errors.

By the way, dealing with this 3-wire SPI is quite frustrating. I couldn't find any examples of using 3-wire SPI with MicroPython online; all the examples I found are for 4-wire SPI. The final solution is to use a logic analyzer to capture the data from successful initialization code. Although it's quite cumbersome, it is indeed the most effective method. I've already purchased one and it's on the way.

kdschlosser, 2024-09-05:
There are no "errors" occurring It's going to be a parameter that is being passed that is causing an issue.

Now.. If the display works without needing to use the 3 wire SPI then just use the rgb_display driver instead. It is going to be the same thing just without the spi portions of the code.

kdschlosser, 2024-09-05:
give that a shot. I change some of the init parameters and also added some other features while I was at it. You need to use this with the display_driver_framework file I uploaded previously.

[st7701.py.txt](https://github.com/user-attachments/files/16884507/st7701.py.txt)

jd3096-mpy, 2024-09-05:
> give that a shot. I change some of the init parameters and also added some other features while I was at it. You need to use this with the display_driver_framework file I uploaded previously.
> 
> [st7701.py.txt](https://github.com/user-attachments/files/16884507/st7701.py.txt)

The screen still isn't displaying anything. I noticed you've written many adjustment command functions, but none of them are being used. I see you modified this function, and I believe it could be the key to the issue.

```
def set_params(self, cmd, params=None):
        if (
            not self._initilized or
            not self._bus_shared_pins
        ):
            self._spi_3wire.tx_param(cmd << 8, params)
```
My friend captured the SPI initialization signals using a logic analyzer.
[image]

SPI_WriteComm(0xFF);
SPI_WriteData(0x77);
SPI_WriteData(0x01);
SPI_WriteData(0x00);
SPI_WriteData(0x00);
SPI_WriteData(0x13);

SPI_WriteComm(0xEF);
SPI_WriteData(0x08);
[init](https://github.com/eggfly/SoundBerry/blob/main/docs/screens/EP28060S_init.c)
It looks like a 9-bit SPI data, where the first bit determines whether it's a command or data. I'm not sure if this is helpful to you, but I feel that this line might be problematic
` self._spi_3wire.tx_param(cmd << 8, params)`

kdschlosser, 2024-09-05:
I am working off of the spec sheet for the driver IC. It states the commands are 16 bits.

I changed a few things. Give it a try and se if it works. I set the command bits to 8 and removed the bit shifting. Originally I had it set to 16 without the bit shifting and that didn't work. So now I have it set without the bit shifting and lowered the command bits. 

[st7701.py.txt](https://github.com/user-attachments/files/16899245/st7701.py.txt)

One of the things I have noticed is the datasheet for this display is a mess. there are sooo many errors in it it's crazy..That makes it exceedingly hard to get things working properly.

kdschlosser, 2024-09-05:
That link you have in your last post is for a display running in MIPI DSI mode which is a completely different interface and if doing to have completely different init commands. It is also written to be used with an STM32 so the code used is going to be using a bunch of bit macros instead of making function calls to do things.

jd3096-mpy, 2024-09-05:
> I am working off of the spec sheet for the driver IC. It states the commands are 16 bits.
>
> I changed a few things. Give it a try and se if it works. I set the command bits to 8 and removed the bit [...]

There's still no image, and it's really difficult to debug without the hardware. Is there any way I can send you an ESP32-S3 RGB board? Next week, I will use the logic analyzer to thoroughly check the SPI signal output from the current MicroPython initialization to see if there's any issue.

kdschlosser, 2024-09-06:
You can send me a board if you want to. It will be easier to have the hardware right in front of me when doing the work. If you have PayPal and tell me how much the board cost you I can send ya the money for it.

I am not going to pass my address on GitHub. If you can set up an account on  LVGL's forum if you don't already have one and send me a message there I will get you my address.

Here is my profile on the forum. 
https://forum.lvgl.io/u/kdschlosser/summary

jd3096-mpy, 2024-09-06:
> You can send me a board if you want to. It will be easier to have the hardware right in front of me when doing the work. If you have PayPal and tell me how much the board cost you I can send ya the [...]

No problem, I will send you a brand-new 86Box ESP32-S3 hardware for free, as a token of my appreciation for the work you've done. I'll contact you later on the LVGL forum.

jd3096-mpy, 2024-09-06:
> [jd3096-mpy](/jd3096-mpy)

I’ve already registered an account on the LVGL forum, but I’m not sure how to send you a private message. This is my account.
https://forum.lvgl.io/u/jd3096/summary

kdschlosser, 2024-09-06:
I sent you a message. You have the chat feature turned off for some reason... You might want to enable that.. It's easier with the chat.

jd3096-mpy, 2024-09-06:
I received your private message in lvgl forum and reply,but I can't find where to sending private message.
Or you can send me an email with your address.
1091329318@qq.com

jd3096-mpy, 2024-09-06:
> I sent you a message. You have the chat feature turned off for some reason... You might want to enable that.. It's easier with the chat.

I received your private message in lvgl forum and reply,but I can't find where to sending private message.
Or you can send me an email with your address.
[1091329318@qq.com](mailto:1091329318@qq.com)

jd3096-mpy, 2024-09-08:
I manually wrote a low-level SPI Python code to implement 9-bit SPI transmission. Now the screen is displaying normally. Although it's a bit clumsy, it works, which proves that the issue was indeed with the SPI initialization code.

closed 2024-09-08 by jd3096-mpy

kdschlosser, 2024-09-08:
well that doesn't help me at all to know where the problem could possibly be. If I could see your code I could figure out where the issue is stemming form.