lvgl-micropython/lvgl_micropython #116: GT911 Initialization failed and it cannot be used with board `ESP32-8048S043`

zjalen, 2024-09-08:
Hello, I have some trouble on gt911 Initialization with board `ESP32-8048S043`. Is someone could help me.
My lvgl-mircopython git log version is `78cc29392253fa6e1054f1e93ca402380b6e2889` on main branch.I tried to init the `gt911` failed.The product_id is none.

```python
import time, os, urandom, lcd_bus, rgb_display, lvgl as lv
from machine import Pin, SPI, ADC, PWM, SDCard, SoftSPI
from micropython import const
import i2c 
import gt911
import task_handler
import _thread

class Demo(object):

    def __init__(self):
        _WIDTH, _HEIGHT = 800, 480
        self.indev = None
        self.cnt = 0
        self.rgb_bus = lcd_bus.RGBBus(de=40, vsync=41, hsync=39, pclk=42,
                                      data11=45, data12=48, data13=47, data14=21, data15=14,
                                      data5=5, data6=6, data7=7, data8=15, data9=16, data10=4,
                                      data0=8, data1=3, data2=46, data3=9, data4=1,
                                      freq=13000000, hsync_front_porch=8, disp=-1,
                                      hsync_pulse_width=4, hsync_back_porch=8, hsync_idle_low=True,
                                      vsync_front_porch=8, vsync_pulse_width=4, vsync_back_porch=8,
                                      vsync_idle_low=True, de_idle_high=False, pclk_idle_high=False,
                                      disp_active_low=False, refresh_on_demand=False, pclk_active_low=1)

        _BUF1 = self.rgb_bus.allocate_framebuffer(_WIDTH * _HEIGHT * 2, lcd_bus.MEMORY_SPIRAM)
        _BUF2 = self.rgb_bus.allocate_framebuffer(_WIDTH * _HEIGHT * 2, lcd_bus.MEMORY_SPIRAM)
        self.display = rgb_display.RGBDisplay(data_bus=self.rgb_bus, display_width=_WIDTH, display_height=_HEIGHT,
                                              frame_buffer1=_BUF1, frame_buffer2=_BUF2, backlight_pin=2,
                                              color_space=lv.COLOR_FORMAT.RGB565, rgb565_byte_swap=False)
        self.display.set_power(True)
        self.display.init()

        self.display.set_backlight(100)
        self.task_handler = task_handler.TaskHandler()

        self.screen = lv.screen_active()

        self.test2()
        time.sleep_ms(500)
        self.screen.set_style_bg_color(lv.color_make(0xff, 0, 0), 0)
        time.sleep_ms(500)
        self.screen.set_style_bg_color(lv.color_make(0, 0xff, 0), 0)
        time.sleep_ms(500)
        self.screen.set_style_bg_color(lv.color_make(0, 0, 0xff), 0)
        time.sleep_ms(500)

        self.init_touch()

    def init_touch(self):
        _WIDTH = const(800)
        _HEIGHT = const(480)

        i2c_bus = i2c.I2C.Bus(host=1, scl=20, sda=19)
        touch_i2c = i2c.I2C.Device(i2c_bus, gt911.I2C_ADDR, gt911.BITS)
        indev = gt911.GT911(touch_i2c, reset_pin=None, startup_rotation=0)
        if indev.hw_size != (_WIDTH, _HEIGHT):
            fw_config = indev.firmware_config
            fw_config.width = _WIDTH
            fw_config.height = _HEIGHT
            fw_config.save()

    def test2(self):
        btn = lv.button(self.screen)
        btn.align(lv.ALIGN.CENTER, 0, 0)
        btn.add_event_cb(self.event_cb, lv.EVENT.CLICKED, None)
        label = lv.label(btn)
        label.set_text('Hello World!')

    def event_cb(self, e):
        print("Clicked")

        btn = e.get_target_obj()
        label = btn.get_child(0)
        label.set_text(str(self.cnt))
        self.cnt += 1
```

This is the output result and the touch is not working.

```bash
>>> from demo import Demo
>>> d=Demo()
Touch Product id:
Touch Firmware version: 0x0
Touch Vendor id: 0x0
Touch resolution: width=0, height=0
Touch Product id:
Touch Firmware version: 0x0
Touch Vendor id: 0x0
Touch resolution: width=0, height=0
```
=======

I have also tried to replace `i2c` and `gt911` with the `develop branch` version.
```python
import i2co as i2c   # checkout from develop branch
import gt911o as gt911   # checkout from develop branch
...
    def init_touch(self):
        _WIDTH = const(800)
        _HEIGHT = const(480)

        i2c_bus = i2c.I2CBus(host=1, scl=20, sda=19)
        self.indev = gt911.GT911(i2c_bus)
        if self.indev.hw_size != (_WIDTH, _HEIGHT):
            fw_config = self.indev.firmware_config
            fw_config.width = _WIDTH
            fw_config.height = _HEIGHT
            fw_config.save()
```
This is the output result, but screen feedback is not in the correct position.
```
>>> from demo import Demo
>>> d=Demo()
Touch Product id: bytearray(b'911\x00')
Touch Firmware version: 0x1060
Touch Vendor id: 0x0
Touch resolution: width=480, height=272
Clicked
```
[image]

Hope some could give me some advices. Thank you.

kdschlosser, 2024-09-08:
Give me a bit to take a look and see what is happening with it. Someone had mentioned something about the bytes that are being sent needing to be flipped around so I will check into that. I made some changes to the I2C drivers recently that may have caused an issue.

zjalen, 2024-09-09:
> Give me a bit to take a look and see what is happening with it. Someone had mentioned something about the bytes that are being sent needing to be flipped around so I will check into that. I made some changes to the I2C drivers recently that may have caused an issue.

Thank you.I have an Arduino demo that runs perfectly.I wish the code could help you to find the issue.
```ino
/* demo.ino*/
#include <lvgl.h>
/*******************************************************************************
 ******************************************************************************/
#include <Arduino_GFX_Library.h>
#define TFT_BL 2
#define GFX_BL DF_GFX_BL

Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
  GFX_NOT_DEFINED /* CS */, GFX_NOT_DEFINED /* SCK */, GFX_NOT_DEFINED /* SDA */,
  40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
  45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
  5 /* G0 */, 6 /* G1 */, 7 /* G2 */, 15 /* G3 */, 16 /* G4 */, 4 /* G5 */,
  8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */
);
// option 1:
// ST7262 IPS LCD 800x480
Arduino_RPi_DPI_RGBPanel *gfx = new Arduino_RPi_DPI_RGBPanel(
  bus,
  800 /* width */, 0 /* hsync_polarity */, 8 /* hsync_front_porch */, 4 /* hsync_pulse_width */, 8 /* hsync_back_porch */,
  480 /* height */, 0 /* vsync_polarity */, 8 /* vsync_front_porch */, 4 /* vsync_pulse_width */, 8 /* vsync_back_porch */,
  1 /* pclk_active_neg */, 14000000 /* prefer_speed */, true /* auto_flush */);

#include "touch.h"
static uint32_t screenWidth;
static uint32_t screenHeight;
static lv_disp_draw_buf_t draw_buf;
static lv_color_t *disp_draw_buf;
static lv_disp_drv_t disp_drv;

void setup() {
  Serial.begin(115200);
  // while (!Serial);
  Serial.println("LVGL Widgets Demo");

  // Init Display
  gfx->begin();
#ifdef TFT_BL
  pinMode(TFT_BL, OUTPUT);
  digitalWrite(TFT_BL, HIGH);
#endif
  gfx->fillScreen(RED);
  delay(500);
  gfx->fillScreen(GREEN);
  delay(500);
  gfx->fillScreen(BLUE);
  delay(500);
  gfx->fillScreen(BLACK);
  delay(500);
  lv_init();
  delay(10);

// Init Touch
  touch_init();
  screenWidth = gfx->width();
  screenHeight = gfx->height();
#ifdef ESP32
  disp_draw_buf = (lv_color_t *)heap_caps_malloc(sizeof(lv_color_t) * screenWidth * screenHeight / 4, MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
#else
  disp_draw_buf = (lv_color_t *)malloc(sizeof(lv_color_t) * screenWidth * screenHeight / 4);
#endif
  if (!disp_draw_buf) {
    Serial.println("LVGL disp_draw_buf allocate failed!");
  } else {
    lv_disp_draw_buf_init(&draw_buf, disp_draw_buf, NULL, screenWidth * screenHeight / 4);

    /* Initialize the display */
    lv_disp_drv_init(&disp_drv);
    /* Change the following line to your display resolution */
    disp_drv.hor_res = screenWidth;
    disp_drv.ver_res = screenHeight;
    disp_drv.flush_cb = my_disp_flush;
    disp_drv.draw_buf = &draw_buf;
    lv_disp_drv_register(&disp_drv);

    /* Initialize the (dummy) input device driver */
    static lv_indev_drv_t indev_drv;
    lv_indev_drv_init(&indev_drv);
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read_cb = my_touchpad_read;
    lv_indev_drv_register(&indev_drv);

    // lv_demo_widgets();
    lv_example_btn_1();

    Serial.println("Setup done");
  }
}
...
```

```c
/* touch.h for GT911 */
 #define TOUCH_GT911
 #define TOUCH_GT911_SCL 20
 #define TOUCH_GT911_SDA 19
 #define TOUCH_GT911_INT -1
 #define TOUCH_GT911_RST 38
 #define TOUCH_GT911_ROTATION ROTATION_NORMAL
 #define TOUCH_MAP_X1 480
 #define TOUCH_MAP_X2 0
 #define TOUCH_MAP_Y1 272
 #define TOUCH_MAP_Y2 0

int touch_last_x = 0, touch_last_y = 0;
#include <Wire.h>
#include <TAMC_GT911.h>
TAMC_GT911 ts = TAMC_GT911(TOUCH_GT911_SDA, TOUCH_GT911_SCL, TOUCH_GT911_INT, TOUCH_GT911_RST, max(TOUCH_MAP_X1, TOUCH_MAP_X2), max(TOUCH_MAP_Y1, TOUCH_MAP_Y2));

void touch_init()
{
  Wire.begin(TOUCH_GT911_SDA, TOUCH_GT911_SCL);
  ts.begin();
  ts.setRotation(TOUCH_GT911_ROTATION);
}

bool touch_has_signal()
{
  return true;
}

bool touch_touched()
{
  ts.read();
  if (ts.isTouched)
  {
#if defined(TOUCH_SWAP_XY)
    touch_last_x = map(ts.points[0].y, TOUCH_MAP_X1, TOUCH_MAP_X2, 0, gfx->width() - 1);
    touch_last_y = map(ts.points[0].x, TOUCH_MAP_Y1, TOUCH_MAP_Y2, 0, gfx->height() - 1);
#else
    touch_last_x = map(ts.points[0].x, TOUCH_MAP_X1, TOUCH_MAP_X2, 0, gfx->width() - 1);
    touch_last_y = map(ts.points[0].y, TOUCH_MAP_Y1, TOUCH_MAP_Y2, 0, gfx->height() - 1);
#endif
    return true;
  }
  else
  {
    return false;
  }
}
bool touch_released()
{
  return true;
}
```

kdschlosser, 2024-09-09:
I just updated the code,m hopefully it will work not. clone the repo and give it a try.

zjalen, 2024-09-09:
> I just updated the code,m hopefully it will work not. clone the repo and give it a try.

@kdschlosser I tried. But here's another crash message.

```bash
>>> from demo import Demo
>>> d=Demo()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "demo.py", line 58, in __init__
  File "demo.py", line 81, in init_touch
  File "gt911.py", line 83, in __init__
  File "gt911.py", line 113, in hw_reset
AttributeError: 'int' object has no attribute 'decode'
>>>
```

```python
# GT911 init code
def init_touch(self):
        _WIDTH = const(800)
        _HEIGHT = const(480)

        i2c_bus = i2c.I2C.Bus(host=1, scl=20, sda=19)
        touch_i2c = i2c.I2C.Device(i2c_bus, gt911.I2C_ADDR, gt911.BITS)
        indev = gt911.GT911(touch_i2c, reset_pin=None, startup_rotation=0)
        if indev.hw_size != (_WIDTH, _HEIGHT):
            fw_config = indev.firmware_config
            fw_config.width = _WIDTH
            fw_config.height = _HEIGHT
            fw_config.save()
```

zjalen, 2024-09-10:
@kdschlosser Hello, I have some new information about this issue.
First, there is a bug in `gt911_extension.py` ，**line 39** should be `self._buf = bytearray(3)` because **line 137** is `self._buf[2] = checksum`.And in the `gt911.py`, **line 113** I replace `product_id += item.decode('utf-8')` with `product_id += str(item)`.

Second, I found `task_handler.TaskHandler()` and `GT911 init` could not be used in the same time.
```python
class Demo(object):
   ...
    def __init__(self):
        ...
        self.task_handler = task_handler.TaskHandler()  # this will make `fw_config = indev.firmware_config` unable to continue
        self.init_touch()  #
        ...

    def init_touch(self):
        _WIDTH = const(800)
        _HEIGHT = const(480)

        i2c_bus = i2c.I2C.Bus(host=1, scl=20, sda=19)
        touch_i2c = i2c.I2C.Device(i2c_bus, gt911.I2C_ADDR, gt911.BITS)
        indev = gt911.GT911(touch_i2c, reset_pin=None, startup_rotation=0)
        if indev.hw_size != (_WIDTH, _HEIGHT):
            fw_config = indev.firmware_config   # hang here
            fw_config.width = _WIDTH
            fw_config.height = _HEIGHT
            fw_config.save()
```

I have located the code that is causing the hang.
```python
# gt911_extension.py  line 45
self._indev._read_reg(_CONFIG_START_REG, buf=self._config_mv[2:])

# gt911.py line 37
    def _read_reg(self, reg, num_bytes=None, buf=None):
        self._tx_buf[0] = reg >> 8
        self._tx_buf[1] = reg & 0xFF
        if num_bytes is not None:
            self._device.write_readinto(self._tx_mv[:2], self._rx_mv[:num_bytes])
        else:
            self._device.write_readinto(self._tx_mv[:2], buf)
```
If I comment out `task_handler.TaskHandler()`, it will run normally. But the display screen will not refresh and  the output resolution is not the resolution I was expecting.

```python
class Demo(object):
   ...
    def __init__(self):
        ...
        # self.task_handler = task_handler.TaskHandler()  # comment
        self.init_touch()  #
        ...

    def init_touch(self):
        _WIDTH = const(800)
        _HEIGHT = const(480)

        i2c_bus = i2c.I2C.Bus(host=1, scl=20, sda=19)
        touch_i2c = i2c.I2C.Device(i2c_bus, gt911.I2C_ADDR, gt911.BITS)
        indev = gt911.GT911(touch_i2c, reset_pin=None, startup_rotation=0)
        if indev.hw_size != (_WIDTH, _HEIGHT):
            fw_config = indev.firmware_config 
            fw_config.width = _WIDTH
            fw_config.height = _HEIGHT
            fw_config.save()
```

```bash
// output
>>> from demo import Demo
>>> d=Demo()
Touch Product id: 574949
Touch Firmware version: 0x1060
Touch Vendor id: 0x0
Touch resolution: width=480, height=272
Touch Product id:
Touch Firmware version: 0x2b7b
Touch Vendor id: 0x70
Touch resolution: width=28672, height=54  // should be width=800,height=480
```

kdschlosser, 2024-09-10:
OK i fixed the gt911_extension driver. See if that works.

I am not sure why the task_handler is not working properly.. Actually I think I might know why it is misbehaving. I mam going to correct it now.

kdschlosser, 2024-09-10:
I just pushed a commit that might fix the hang that is happening with the touch driver.

kdschlosser, 2024-09-10:
I also fixed the decoding of an integer.. I was brain dead when I did that.

zjalen, 2024-09-10:
> I also fixed the decoding of an integer.. I was brain dead when I did that.

@kdschlosser Thank you very much! It did work.The only issue I have now is that the touch screen's feedback location is incorrect.
```python
    def init_touch(self):
        _WIDTH = const(800)
        _HEIGHT = const(480)

        i2c_bus = i2c.I2C.Bus(host=1, scl=20, sda=19)
        touch_i2c = i2c.I2C.Device(i2c_bus, gt911.I2C_ADDR, gt911.BITS)
        indev = gt911.GT911(touch_i2c)
        if indev.hw_size != (_WIDTH, _HEIGHT):
            fw_config = indev.firmware_config 
            fw_config.width = _WIDTH
            fw_config.height = _HEIGHT
            fw_config.save()
```

```bash
// output
>>> from demo import Demo
>>> d=Demo()
Touch Product id: 911
Touch Firmware version: 0x1060
Touch Vendor id: 0x0
Touch resolution: width=480, height=272
Touch Product id: "
Touch Firmware version: 0x2b7b
Touch Vendor id: 0x70
Touch resolution: width=28672, height=54  // should be width=800,height=480
```

[image]

zjalen, 2024-09-10:
I think there is a bug in ``

```python
# gt911_extension.py -> line 125
    def save(self):
        checksum = ((~sum(self._config_data[2:])) + 1) & 0xFF

        self._config_data[0] = _CONFIG_CHKSUM_REG >> 8
        self._config_data[1] = _CONFIG_CHKSUM_REG & 0xFF
        self._indev._write_reg(_CONFIG_CHKSUM_REG, buf=self._config_mv)

        self._config_data[2] = checksum
        self._indev._write_reg(_CONFIG_CHKSUM_REG, buf=self._config_mv[:3])

        self._config_data[0] = _CONFIG_FRESH_REG >> 8
        self._config_data[1] = _CONFIG_FRESH_REG & 0xFF
        self._config_data[2] = 0x01
        self._indev._write_reg(_CONFIG_FRESH_REG, buf=self._config_mv[:3])

        self._indev.hw_reset()
```

I tried using the following code as a replacement.
```python
   def save(self):
        checksum = ((~sum(self._config_data[2:])) + 1) & 0xFF

        self._config_data[0] = _CONFIG_CHKSUM_REG >> 8
        self._config_data[1] = _CONFIG_CHKSUM_REG & 0xFF
        self._config_data[2] = checksum
        self._indev._write_reg(_CONFIG_CHKSUM_REG, buf=self._config_mv[:3])

        self._config_data[0] = _CONFIG_FRESH_REG >> 8
        self._config_data[1] = _CONFIG_FRESH_REG & 0xFF
        self._config_data[2] = 0x01
        self._indev._write_reg(_CONFIG_FRESH_REG, buf=self._config_mv[:3])

        self._config_data[0] = const(800) >> 8
        self._config_data[1] = const(800) & 0xFF
        self._indev._write_reg(const(0x8146), value=self._config_data[1])
        self._indev._write_reg(const(0x8147), value=self._config_data[0])

        self._config_data[0] = const(480) >> 8
        self._config_data[1] = const(480) & 0xFF
        self._indev._write_reg(const(0x8148), value=self._config_data[1])
        self._indev._write_reg(const(0x8149), value=self._config_data[0])

        self._indev.hw_reset()
```
The the output resolution now is correct, but the touch screen's feedback location is still incorrect.
```bash
// debug=True
>>> from demo import Demo
>>> d=Demo()
Touch Product id: 911
Touch Firmware version: 0x1060
Touch Vendor id: 0x0
Touch resolution: width=480, height=272
GT911(raw_x=-1, raw_y=-1, x=0, y=0, state=RELEASED)
Touch Product id: 911
Touch Firmware version: 0x1060
Touch Vendor id: 0x0
Touch resolution: width=800, height=480  // ouput correct
>>> GT911(raw_x=0, raw_y=0, x=472, y=263, state=PRESSED)
GT911(raw_x=472, raw_y=263, x=472, y=263, state=RELEASED)  // touched the right-bottom corner
```
I think the real resolution is still `480*272`. Because when I touched the right-bottom corner. I got `GT911(raw_x=472, raw_y=263, x=472, y=263, state=RELEASED)`.

kdschlosser, 2024-09-11:
was it my updates that did it? They shouldn't have....

zjalen, 2024-09-12:
@kdschlosser I spent some time reading the documentation and trying different approaches, and now I have resolved the issue.It was indeed a code bug. The problem was caused by an offset in the register write position, which led to discrepancies in resolution and border distance information. The code and my solution are provided here.

```python
# gt911_extension.py
class GT911Extension(object):

    # -----> copy and edit from gt911.py, because`self._indev._read_reg` result is incorrect
    def _read_reg(self, reg, num_bytes=None, buf=None):
        self._tx_buf[0] = reg >> 8
        self._tx_buf[1] = reg & 0xFF
        if num_bytes is not None:
            self._i2c.write_readinto(self._tx_mv[:2], self._rx_mv[:num_bytes])
        else:
            self._i2c.write_readinto(self._tx_mv[:2], buf)

    # -----> copy and edit from gt911.py  because`self._indev._write_reg` write failed
    def _write_reg(self, reg, value=None, buf=None):
        if value is not None:
            self._tx_buf[0] = value
            self._i2c.write_mem(reg, self._tx_mv[:1])
        elif buf is not None:
            self._i2c.write_mem(reg, buf)

    def __init__(self, indev, i2c):
        self._indev = indev
        self._i2c = i2c

        # self._config_data = bytearray(_CONFIG_FRESH_REG - _CONFIG_START_REG)
        self._config_data = bytearray(_CONFIG_FRESH_REG - _CONFIG_START_REG + 1)  # ----> fix 1
        self._config_mv = memoryview(self._config_data)

        # -----> copy from gt911.py
        self._tx_buf = bytearray(3)
        self._tx_mv = memoryview(self._tx_buf)
        self._rx_buf = bytearray(6)
        self._rx_mv = memoryview(self._rx_buf)

        # self._indev._read_reg(_CONFIG_START_REG, buf=self._config_mv[2:])
        self._read_reg(_CONFIG_START_REG, buf=self._config_mv[:-2])   # ----> fix 2
...
...
    def save(self):
        # calculate the checksum
        self._config_data[_CONFIG_CHKSUM_REG - _CONFIG_START_REG] = ((~sum(self._config_data[:-2])) + 1) & 0xFF

        # set the flag to save the data the data
        self._config_data[-1] = 0x01  # _CONFIG_FRESH_REG

        # write all config data to the touch IC
        self._write_reg(_CONFIG_START_REG, buf=self._config_mv)    # ----> fix 3

        self._indev.hw_reset()
```

kdschlosser, 2024-09-12:
Try the driver now. I made some of the changes you have above. I didn't do the copy of the methods and instance attributes from the gt911 driver. I am curious to see if it works. There is no reason why it shouldn't work because the code is identical to what is in the  driver.

I did make the modifications for getting the `bytearray` correct. 

Give it a try and see what happens.

kdschlosser, 2024-09-12:
I do thank you for the time you have spent getting this to work correctly. That is a HUGE help. I am trying to get caught up with fixes and trying to finish up some of the core bus drivers. I had back issues that I ended up having to go into surgery for and I was out of commission for several months. And after that I damned near cut one of my fingers off of my right hand. It's been a heck of a year this year....

I wanted to let you know that the time you have spent on this is not unnoticed and is greatly appreciated.

zjalen, 2024-09-13:
@kdschlosser Sorry to hear about the unfortunate situation you’re going through. I hope you recover and stay safe. Thank you for maintaining this project.I only provided some basic testing and I am glad this was helpful to you.
I believe that two different methods produce different effects, even though the code looks the same.And I don't know why.Here is the code and the output result.

```python
class GT911Extension(object):
    def _read_reg(self, reg, num_bytes=None, buf=None):
        self._tx_buf[0] = reg >> 8
        self._tx_buf[1] = reg & 0xFF
        if num_bytes is not None:
            self._i2c.write_readinto(self._tx_mv[:2], self._rx_mv[:num_bytes])
        else:
            self._i2c.write_readinto(self._tx_mv[:2], buf)

    def __init__(self, indev, i2c):
        self._indev = indev
        self._i2c = i2c

        self._config_data = bytearray(_CONFIG_FRESH_REG - _CONFIG_START_REG + 1)
        self._config_mv = memoryview(self._config_data)

        print('=======init======')
        print(self._config_data.hex())

        print('=======_indev_read_reg=======')
        self._indev._read_reg(_CONFIG_START_REG, buf=self._config_mv[:-2])
        print(self._config_data.hex())

        print('=======_read_reg=======')
        self._tx_buf = bytearray(3)
        self._tx_mv = memoryview(self._tx_buf)
        self._rx_buf = bytearray(6)
        self._rx_mv = memoryview(self._rx_buf)
        self._read_reg(_CONFIG_START_REG, buf=self._config_mv[:-2])
        print(self._config_data.hex())
```

```bash
=======init======
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
=======_indev_read_reg=======
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000004400010000bb05c305f105ed05ef05ed05ec05ef05e705dd05e205e805b305f0062a062b062e062e06240631061f061a0622062405e105f50631062c0633062e062d0633062506270628062b0000
=======_read_reg=======
412003e001053d0001083c0578500305000000000000001a1c1e148b2a0c2a28eb0400000102032c0001000000000000000000143c94450208000004b716009f1b008b22007b2b0070360070000000000000000000000000000000000000000000000000000000000000000000000000020406080a0c0e1012141618ffffffffffffffffffffffffffffffffffff242221201f1e1d1c18161312100f0c0a0806040200ffffffffffffffffffffffffffffffffffffffffff0000
```

kdschlosser, 2024-09-13:
that is quite curious isn't it?

I will make the changes you have previously posted to get it working. it is strange as to why it's not working.

kdschlosser, 2024-09-13:
it should be good to go.

zjalen, 2024-09-13:
I have tested it, and it runs perfectly.Thank you.

closed 2024-09-13 by zjalen

gitcnd, 2025-02-05:
I made some instructions for getting this board working... slightly non-trivial because of the touch using the jtag pins...

https://github.com/lvgl-micropython/lvgl_micropython/discussions/287