lvgl-micropython/lvgl_micropython #117: ESP32-S3 compilation failed

217heidai, 2024-09-09:
```bash
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16 DISPLAY=gc9a01
```
erro information:
```
File "/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/make.py", line 201, in <module>
    extra_args, lv_cflags, board = mod.parse_args(extra_args, lv_cflags, board)
File "/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/builder/esp32.py", line 410, in parse_args
    extra_args = esp32_s3_args(mextra_args)
NameError: name 'mextra_args' is not defined. Did you mean: 'extra_args'?
```

kdschlosser, 2024-09-09:
this is all fixed...

closed 2024-09-09 by kdschlosser

217heidai, 2024-09-09:
it fixed, thank you.  
but I got another error:
```
. ./export.sh
Getting ESP-IDF build Environment
echo /usr/bin >> $GITHUB_PATH
echo /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/esp-idf >> $GITHUB_PATH
echo /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/esp-idf/tools >> $GITHUB_PATH
export "IDF_PATH=lib/esp-idf"
cd lib/esp-idf
. ./export.sh
printenv
Compiling mpy-cross....
make -C lib/micropython/mpy-cross
make: Entering directory '/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/micropython/mpy-cross'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
GEN build/genhdr/moduledefs.split
GEN build/genhdr/moduledefs.collected
Module registrations updated
GEN build/genhdr/moduledefs.h
GEN build/genhdr/root_pointers.split
GEN build/genhdr/root_pointers.collected
Root pointer registrations updated
GEN build/genhdr/root_pointers.h
mkdir -p build/py/
mkdir -p build/shared/runtime/
CC ../py/mpstate.c
CC ../py/nlr.c    
CC ../py/nlrx86.c
CC ../py/nlrx64.c
CC ../py/nlrthumb.c
CC ../py/nlraarch64.c
CC ../py/nlrmips.c   
CC ../py/nlrpowerpc.c
CC ../py/nlrxtensa.c 
CC ../py/nlrsetjmp.c
CC ../py/malloc.c   
CC ../py/gc.c    
CC ../py/pystack.c
CC ../py/qstr.c   
CC ../py/vstr.c
CC ../py/mpprint.c
CC ../py/unicode.c
CC ../py/mpz.c    
CC ../py/reader.c
CC ../py/lexer.c 
CC ../py/parse.c
CC ../py/scope.c
CC ../py/compile.c
CC ../py/emitcommon.c
CC ../py/emitbc.c    
CC ../py/asmbase.c
CC ../py/asmx64.c 
CC ../py/emitnx64.c
CC ../py/asmx86.c  
CC ../py/emitnx86.c
CC ../py/asmthumb.c
CC ../py/emitnthumb.c
CC ../py/emitinlinethumb.c
CC ../py/asmarm.c         
CC ../py/emitnarm.c
CC ../py/asmxtensa.c
CC ../py/emitnxtensa.c
[... 233 lines cut ...]
-- App "micropython" version: a61c446-dirty                        
-- Adding linker script /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- ESP_LCD_PANEL_IO_ADDITIONS: 1.0.1
Including User C Module(s) from ../../../../../ext_mod/micropython.cmake
-- Configuring incomplete, errors occurred!
See also "/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/CMakeFiles/CMakeOutput.log".
Found User C Module(s): usermod_lcd_bus, usermod_lvgl, lvgl_interface, usermod_lcd_utils, usermod_nvs
CMake Error at /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/micropython/extmod/extmod.cmake:114 (message):
MICROPY_PY_BTREE is enabled but the berkeley-db submodule is not initialised.
Run 'make BOARD=ESP32_GENERIC_S3 submodules'
Call Stack (most recent call first):
esp32_common.cmake:23 (include)
main_esp32s3/CMakeLists.txt:11 (include)
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
cmake failed with exit code 1, output of the command is in the /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_5127 and /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_5127
make: *** [Makefile:66: all] Error 1
make: Leaving directory '/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/micropython/ports/esp32'
```

commit ba24b256 (2024-09-09): fixes #117