lvgl-micropython/lvgl_micropython #119: LV_KEY_PREV cause system crash

jd3096-mpy, 2024-09-12:
I test lv.KEY.UP  lv.KEY.LEFT  lv.KEY.DOWN  lv.KEY.RIGHT  lv.KEY.NEXT  lv.KEY.PREV lv.KEY.BACKSPACE  lv.KEY.ENTER
When I press  lv.KEY.PREV ,system crashed
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420a799f
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x105c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbd8
load:0x403cc700,len:0x2e30
entry 0x403c989c
W (31) boot.esp32s3: PRO CPU has been reset by WDT.
W (31) boot.esp32s3: APP CPU has been reset by WDT.
```
All other keys work fine, but this one causes the system to crash. I've tried many widgets, and the result is the same.

jd3096-mpy, 2024-09-22:
GROUP should be create by 
`group = lv.group_create()`

not
`group = lv.group_t()`

closed 2024-09-22 by jd3096-mpy

kdschlosser, 2024-09-22:
it is being created correctly...

https://github.com/lvgl-micropython/lvgl_micropython/blob/927215d854a255ba00934f7bc2e18d3bdf063334/api_drivers/common_api_drivers/indev/sdl_keyboard.py#L165

jd3096-mpy, 2024-09-22:
> it is being created correctly...
> 
> https://github.com/lvgl-micropython/lvgl_micropython/blob/927215d854a255ba00934f7bc2e18d3bdf063334/api_drivers/common_api_drivers/indev/sdl_keyboard.py#L165

Yes, that's my mistake. I didn't use your framework but wrote a keyboard device from scratch. The strange thing is, when using group = lv.group_t(), it only crashes when running prev, but everything else works fine.

kdschlosser, 2024-09-23:
dunno why it would do that...