lvgl-micropython/lvgl_micropython #120: macOS SDL

dcmcshan, 2024-09-13:
Figured it might be useful to thread this for others.  

Built with this.

```
 % python3 make.py macOS clean submodules mpy_cross DISPLAY=sdl_display INDEV=sdl_pointer --heap-size=8388608
chmod +x build/lvgl_micropy_macos
```
Moved to this application area.

```
 % ls
MCT_SDL.py		SDLPointer_2		libSDL2-2.0.0.dylib	lvgl_micropy_macos
```

MCT_SDL.py look like this:

```
import sys
import lcd_bus
from micropython import const  # NOQA
import lvgl as lv

print ("Darwin")
_LCD_RST = const(-1)
_LCD_PWR = const(-1)
_LCD_BKL = const(-1)

# that flag will stop you from being able to resize it.
display_bus = lcd_bus.SDLBus(flags=lcd_bus.SDLBus.WINDOW_BORDERLESS)

import sdl_display

display_driver = sdl_display.SDLDisplay

import sdl_pointer

indev_driver = sdl_pointer.SDLPointer

display = display_driver(
    data_bus=display_bus,
    display_width=160,
    display_height=40,
    reset_pin=_LCD_RST,
    power_pin=_LCD_PWR,
    backlight_pin=_LCD_BKL,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display.set_power(True)
display.init()

indev = indev_driver()

display.set_rotation(lv.DISPLAY_ROTATION._90)
display.set_backlight(100)
```

Import occurs, and an application starts, but there is no window for the output?

```
 % ./lvgl_micropy_macos
LVGL MicroPython 1.23.0 on 2024-09-13; darwin [GCC 4.2.1] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import MCT_SDL
Darwin
>>> 
```

mark-hoy, 2024-09-24:
Dan, I tried what you did and also didn't get a window of output.  I also had to copy in two of the sdl python files. Maybe I built it incorrectly?  

```
$ ls -lt
total 6160
-rw-r--r--  1 mark_hoy  staff        0 Sep 24 09:58 SDLPointer_2
-rw-r--r--  1 mark_hoy  staff      849 Sep 24 09:55 MCT_SDL.py
-rw-r--r--  1 mark_hoy  staff     2495 Sep 24 09:54 sdl_pointer.py
-rw-r--r--  1 mark_hoy  staff    10295 Sep 24 09:53 sdl_display.py
-rwxr-xr-x  1 mark_hoy  staff  1565816 Sep 24 09:51 lvgl_micropy_macos
-rw-r--r--  1 mark_hoy  staff  1561776 Sep 24 09:51 libSDL2-2.0.0.dylib

$ ./lvgl_micropy_macos 
LVGL MicroPython 1.23.0 on 2024-09-22; darwin [GCC 4.2.1] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import MCT_SDL
Darwin
>>> 
```

kdschlosser, 2024-10-09:
this should now be fixed. give it a try and lemme know...

closed 2024-11-05 by kdschlosser