lvgl-micropython/lvgl_micropython #122: ESP32 Build no longer compiles a binary

androiddrew, 2024-09-15:
Using commit b0af377f52a1de02a728e828a3cb771334f46bcc I am running the build command that should have resulted in a ./build/ binary for my ESP32-S3. Instead it seems to fall short now. This was working on commit 058edd0 before I pulled down the latest changes from master.

Using MacOS. 

On a fresh clone:

```
git clone https://github.com/lvgl-micropython/lvgl_micropython.git
cd lvgl_micropython
python3 make.py esp32 submodules BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=ili9341 INDEV=xpt2046 --octal-flash --flash-size 32
```

```
...
Generating build files....
/Users/drewbednar/tmp/lvgl_micropython/api_drivers/py_api_drivers/frozen/display/display_driver_framework.py
/Users/drewbednar/tmp/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/touch_calibration/touch_cal_data.py
/Users/drewbednar/tmp/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/touch_calibration/touch_calibrate.py
/Users/drewbednar/tmp/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/_indev_base.py
/Users/drewbednar/tmp/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/button_framework.py
/Users/drewbednar/tmp/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/encoder_framework.py
/Users/drewbednar/tmp/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/keypad_framework.py
/Users/drewbednar/tmp/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/pointer_framework.py
/Users/drewbednar/tmp/lvgl_micropython/api_drivers/py_api_drivers/fs_driver.py
/Users/drewbednar/tmp/lvgl_micropython/api_drivers/common_api_drivers/frozen/other/i2c.py
/Users/drewbednar/tmp/lvgl_micropython/api_drivers/common_api_drivers/frozen/other/io_expander_framework.py
/Users/drewbednar/tmp/lvgl_micropython/api_drivers/common_api_drivers/frozen/other/task_handler.py
/Users/drewbednar/tmp/lvgl_micropython/api_drivers/common_api_drivers/indev/xpt2046.py
/Users/drewbednar/tmp/lvgl_micropython/api_drivers/common_api_drivers/display/ili9341.py
Compiling....
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Updating submodules: lib/berkeley-db-1.xx lib/micropython-lib
Submodule 'lib/berkeley-db-1.xx' (https://github.com/micropython/berkeley-db-1.xx) registered for path '../../lib/berkeley-db-1.xx'
Submodule 'lib/micropython-lib' (https://github.com/micropython/micropython-lib.git) registered for path '../../lib/micropython-lib'
Cloning into '/Users/drewbednar/tmp/lvgl_micropython/lib/micropython/lib/berkeley-db-1.xx'...
Cloning into '/Users/drewbednar/tmp/lvgl_micropython/lib/micropython/lib/micropython-lib'...
Submodule path '../../lib/berkeley-db-1.xx': checked out '85373b548f1fb0119a463582570b44189dfb09ae'
From https://github.com/micropython/micropython-lib
* branch            50ed36fbeb919753bcc26ce13a8cffd7691d06ef -> FETCH_HEAD
Submodule path '../../lib/micropython-lib': checked out '50ed36fbeb919753bcc26ce13a8cffd7691d06ef'
```

Full build output: [b0af377-esp32-s3-attempt1.txt](https://github.com/user-attachments/files/17004252/b0af377-esp32-s3-attempt1.txt)

androiddrew, 2024-09-15:
So I think the issue might be something to do with the `submodules` command because I reran immediately with 

```
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=ili9341 INDEV=xpt2046 --octal-flash --flash-size 32
```

And did get a binary built.

kdschlosser, 2024-09-15:
sometimes you have to let it sit there for a long while. It is because of the connection to github might be slow for some reason and also because MicroPython doesn't truncate their submodules so it pulls the entire git history which can take a while to do depending on how much a submodule or one of it's submodules has been updated.

I have had to let it sit for 5 minutes at least until it finished. Be patient it will go.

closed 2024-09-16 by androiddrew