lvgl-micropython/lvgl_micropython #123: ESP32 ILI9341 with xpt2046 freezes on creating second SPI.Bus

androiddrew, 2024-09-15:
Using commit b0af377 I built an ESP32-S3 binary and attempted to create a second `SPI.Device` for using with the XPT2046 touch device, but the program hangs when attempting to instantiate it.

```
import lvgl as lv
import lcd_bus

import ili9341
from machine import SPI

# DISPLAY SETUP

spi_bus = SPI.Bus(host=1, mosi=14, miso=41, sck=13)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=4,
    cs=6,
    freq=40000000,
)

display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=240,
    display_height=320,
    reset_pin=5,
    # comment the next line if you still don't get something on the display
    reset_state=ili9341.STATE_LOW,
    backlight_pin=40,  # NOT USED...YET
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=ili9341.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

display.set_power(True)
display.init()

# setting this to anything but 100 or 0 when you don't have the
# backlight state set to PWM isn't going to do anything
# display.set_backlight(100)

# TOUCH SETUP

print("Creating touch bus")
# T_IRQ = 36
# T_DO = 35
# T_DIN = 1
# T_CS = 2
# T_CLK = 42
touch_bus = SPI.Bus(host=2, mosi=1, miso=35, sck=42)

print("creating touch SPI Device")
touch_device = SPI.Device(
    spi_bus=touch_bus,
    freq=1000000,
    cs=2,
    polarity=0,
    phase=0,
    bits=8,
)
print("importing xpt2045")
import xpt2046

print("Instantiating touch Indev")
touch_indev = xpt2046.XPT2046(
    touch_device, touch_cal=None, startup_rotation=lv.DISPLAY_ROTATION._90, debug=False
)

# TOUCH CALIBRATE

print("Calibration")
if not touch_indev.is_calibrated:
    touch_indev.calibrate()

display.invert_colors()

# you want to rotate the display after the calibration has been done in order
# to keep the corners oriented properly.
display.set_rotation(lv.DISPLAY_ROTATION._90)

# SCREEN SETUP

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0xFFB6C1), 0)

slider = lv.slider(scrn)
slider.set_size(280, 20)
slider.center()

label = lv.label(scrn)
label.set_text("HELLO WORLD!")
label.align(lv.ALIGN.CENTER, 0, -50)

import task_handler

th = task_handler.TaskHandler()

```

```
(env) drewbednar@eisenhorn lvgl-upy-examples % mp cp ./project2/helloworld.py :
cp ./project2/helloworld.py :
(env) drewbednar@eisenhorn lvgl-upy-examples % mpremote                        
Connected to MicroPython at /dev/cu.usbserial-14110
Use Ctrl-] or Ctrl-x to exit this shell
>>> import helloworld
Creating touch bus
creating touch SPI Device
```

kdschlosser, 2024-09-15:
if memory serves host 2 you can only use hardware pins for and host 1 you can use whatever pins you want.

I have to double check on that.

kdschlosser, 2024-09-15:
Is this an ESP32 or another ESP32 variant? IE S2, S3, etc....

kdschlosser, 2024-09-15:
sorry I missed that you said it's an ESP32-S3
Let me check on the pins for that.

kdschlosser, 2024-09-15:
if you are using an ESP32 that has octal ram or flash you cannot use pin 35

androiddrew, 2024-09-15:
@kdschlosser That was it...sorry I didn't think to check that myself.

I do have it working now. I'll close out the issue, but two questions.

1.  Is there a good example for how to do calibration?  Do I have to write my own calibration function?
2. Also it seems like calibration should be saved for restarts is that what the NVS is for?

androiddrew, 2024-09-15:
This should be easier to understand the problem I am having.

https://github.com/user-attachments/assets/fa93992d-0844-48d6-a962-603949b71cf4

kdschlosser, 2024-09-15:
I still have to work on getting the calibration working properly. I haven't had the chance to do that yet. It will automatically save to the NVS when it is all finished up and working properly. I will mess around with it today to get it working properly. 

Glad to know it was something silly with the touch and the SPI. I have to look and see if there is a way for me to see if octal SPI is being used for flash or ram so I can add a guard for that. The fact that it just hangs is surprising I would have thought that the ESP-IDF would throw an error but I guess not.

androiddrew, 2024-09-15:
@kdschlosser if you get a chance please have a look at the attached video concerning the tabview active tab behavior. It's setting the index in reverse it seems. That is the only think keeping my UI back. I have tried adding a custom callback to each button to set the tabview.set_active to the correct idx if pressed. I've been debugging it for hours and am starting to think I am crazy here.

kdschlosser, 2024-09-15:
I don't do apple anything and .MOV files are apple and there for do not work for most people.

androiddrew, 2024-09-15:
> I don't do apple anything and .MOV files are apple and there for do not work for most people.

Does an .mp4 work?

https://github.com/user-attachments/assets/e675f2ff-459e-4183-9b6a-cf4b34c5f63b

kdschlosser, 2024-09-16:
HA!!! That is because your touchscreen is wired to the IC incorrectly. Gotta love stuff made in China.

The calibration once I fix it will correct this problem.

kdschlosser, 2024-09-16:
touch calibration should now be fixed...

closed 2024-11-05 by kdschlosser