lvgl-micropython/lvgl_micropython #124: Are there plans to add support for QSPI?

jd3096-mpy, 2024-09-18:
There are many AMOLED screens with vibrant colors that are suitable for use with LVGL, but they all use the QSPI protocol. Is QSPI currently supported? Do you plan to add support for QSPI in the future?

kdschlosser, 2024-09-19:
while the ESP-IDF docs state it is not yet supported....

https://docs.espressif.com/projects/esp-idf/en/v5.2.2/esp32/api-reference/peripherals/lcd.html#spi-interfaced-lcd

I have added it anyhow. I also opened an issue with Espressif asking them if it is supported or not. They don't always keep the documentation up to date properly.

https://github.com/espressif/esp-idf/issues/14605

kdschlosser, 2024-09-19:
to use quad SPI there are 2 new parameters when creating the bus. `data2` and `data3`. fill those 2 parameters with the pin numbers you are using for those additional lines. `data2` is "wp" and `data3` is "hd"

The hard part about QSPI is what exactly is QSPI. is the sending and receiving of  SPI commands QSPI? or is it just the data that is... I have it set so only the data is sent using QSPI and not the SPI commands.

jd3096-mpy, 2024-09-19:
> to use quad SPI there are 2 new parameters when creating the bus. `data2` and `data3`. fill those 2 parameters with the pin numbers you are using for those additional lines. `data2` is "wp" and [...]

I just thought about something regarding QSPI for the moment, and I'll conduct testing when I have a QSPI screen board in hand.

kdschlosser, 2024-09-19:
I was correct about the documentation not being updated for the IDF. It has been added. I just pushed the last commit that should enable the use of QSPI. It gets complicated if you have multiple devices on the same bus and only the display uses QSPI and the other devices do not. I am trying to figure out how to handle that scenario.

favnec5, 2024-09-19:
hello, you can found functionnal example here : https://github.com/jmgrosen/lcd_binding_micropython/tree/master/lcd/bus/qspi

straga, 2024-10-24:
https://github.com/lvgl-micropython/lvgl_micropython/discussions/161

closed 2024-11-05 by kdschlosser