lvgl-micropython/lvgl_micropython #125: The I8080 parallel interface screen cannot work

jd3096-mpy, 2024-09-19:
I test lilygo t-display-s3, it doesn't work.
Here is the pinmap:
[image]

and here is my code:
```
from machine import Pin
pr=Pin(15,Pin.OUT)
cs=Pin(6,Pin.OUT)
cs.off()
pr.on()
import lcd_bus

from micropython import const

# display settings
_WIDTH = const(320)
_HEIGHT = const(170)
_BL = const(0)
_RST = const(5)
_DC = const(7)
_WR = const(8)
_FREQ = const(20000000)
_DATA0 = const(39)
_DATA1 = const(40)
_DATA2 = const(41)
_DATA3 = const(42)
_DATA4 = const(45)
_DATA5 = const(46)
_DATA6 = const(47)
_DATA7 = const(47)
_BUFFER_SIZE = const(320*170*2)

display_bus = lcd_bus.I80Bus(
    dc=_DC,
    wr=_WR,
    freq=_FREQ,
    data0=_DATA0,
    data1=_DATA1,
    data2=_DATA2,
    data3=_DATA3,
    data4=_DATA4,
    data5=_DATA5,
    data6=_DATA6,
    data7=_DATA7
)

fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM)
fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM)

import st7789  # NOQA
import lvgl as lv  # NOQA

lv.init()

display = st7789.ST7789(
    data_bus=display_bus,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
#     reset=_RST,
#     reset_state=st7789.STATE_LOW,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

import task_handler  # NOQA
import time  # NOQA

display.init()
display.invert_colors()
display.set_backlight(100)

# you want to rotate the display after the calibration has been done in order
# to keep the corners oriented properly.
display.set_rotation(lv.DISPLAY_ROTATION._90)

display.set_backlight(100)

th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(100, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)

```
I found additional CS, RST, and RD pins but don't know how to use them correctly.

kdschlosser, 2024-09-19:
give this a shot....

```python
import lcd_bus

from micropython import const

import machine

# RD 9 - might have to be pulled high
rd = machine.Pin(9, machine.Pin.OUT)
rd(1)

# CS 6 - might have to be pulled low
cs = machine.Pin(6, machine.Pin.OUT)
cs(0)

# display settings
_WIDTH = const(320)
_HEIGHT = const(170)
_PWR = const(15)
_BL = const(38)
_RST = const(5)
_DC = const(7)
_WR = const(8)
_FREQ = const(40000000)
_DATA0 = const(39)
_DATA1 = const(40)
_DATA2 = const(41)
_DATA3 = const(42)
_DATA4 = const(45)
_DATA5 = const(46)
_DATA6 = const(47)
_DATA7 = const(47)
_BUFFER_SIZE = const(320*170*2)

display_bus = lcd_bus.I80Bus(
    dc=_DC,
    wr=_WR,
    freq=_FREQ,
    data0=_DATA0,
    data1=_DATA1,
    data2=_DATA2,
    data3=_DATA3,
    data4=_DATA4,
    data5=_DATA5,
    data6=_DATA6,
    data7=_DATA7
)

import st7789  # NOQA
import lvgl as lv  # NOQA

lv.init()

display = st7789.ST7789(
    data_bus=display_bus,
    # let the driver handle the frame buffers
    frame_buffer1=None,
    frame_buffer2=None,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    power_pin=_PWR,
    reset_pin=_RST,
    reset_state=st7789.STATE_LOW,
    color_space=lv.COLOR_FORMAT.RGB565,
    # color_byte_order=st7789.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

display.set_power(True)
display.init()
display.invert_colors()
display.set_backlight(100)

# Lets get it displaying before we worry about rotation
# display.set_rotation(lv.DISPLAY_ROTATION._90)

display.set_backlight(100)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(100, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)

import task_handler  # NOQA

th = task_handler.TaskHandler()
```

jd3096-mpy, 2024-09-19:
> give this a shot....
>
> ```python
> import lcd_bus
>
> from micropython import const
>
> import machine
>
>
> # RD 9 - might have to be pulled high
> rd = machine.Pin(9, machine.Pin.OUT)
> rd(1)
>
> # CS 6 - might have [...]

I've conducted some additional tests, and the current result is that the screen is displaying noise. I compared it with this library, and its firmware has been tested and works correctly. I strongly believe the issue lies with the CS pin. You cannot keep the CS pin low all the time; instead, you need to toggle the DC and CS pins accordingly before each data write, as per the timing requirements. Therefore, I think the solution is to add an additional CS pin parameter when initializing the I80 interface. Of course, this parameter can be set to None.

https://github.com/russhughes/st7789s3_mpy/blob/main/st7789/st7789.c

jd3096-mpy, 2024-09-19:
```
import lcd_bus
from micropython import const
import machine
import time

rd = machine.Pin(9, machine.Pin.OUT)
rd.value(1)
cs = machine.Pin(6, machine.Pin.OUT)
cs.value(0)

# display settings
_WIDTH = const(170)
_HEIGHT = const(320)
_PWR = const(15)
_BL = const(38)
_RST = const(5)
_DC = const(7)
_WR = const(8)
_CS = const(6)
_FREQ = const(20000000)
_DATA0 = const(39)
_DATA1 = const(40)
_DATA2 = const(41)
_DATA3 = const(42)
_DATA4 = const(45)
_DATA5 = const(46)
_DATA6 = const(47)
_DATA7 = const(48)
_BUFFER_SIZE = const(320*170*2)

display_bus = lcd_bus.I80Bus(
    dc=_DC,
    wr=_WR,
#     cs=_CS,
    freq=_FREQ,
    data0=_DATA0,
    data1=_DATA1,
    data2=_DATA2,
    data3=_DATA3,
    data4=_DATA4,
    data5=_DATA5,
    data6=_DATA6,
    data7=_DATA7
)

import st7789  # NOQA
import lvgl as lv  # NOQA

lv.init()

display = st7789.ST7789(
    data_bus=display_bus,
    # let the driver handle the frame buffers
    frame_buffer1=None,
    frame_buffer2=None,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    power_pin=_PWR,
    reset_pin=_RST,
    reset_state=st7789.STATE_LOW,
    color_space=lv.COLOR_FORMAT.RGB565,
    offset_y=35,
    color_byte_order=st7789.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

display.set_power(True)
display.init()
display.invert_colors()
display.set_backlight(100)

# Lets get it displaying before we worry about rotation
display.set_rotation(lv.DISPLAY_ROTATION._90)

display.set_backlight(100)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(100, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.set_pos(0,0)

import task_handler  # NOQA

th = task_handler.TaskHandler()
```
Thanks, I made some further adjustments based on your input, and now the screen is working properly. The issue mainly stemmed from the frequency settings. After continuous testing, I found that if the frequency is set above 20MHz, the screen displays noise. So, for now, the maximum frequency can only be set to 20,000,000.

closed 2024-09-19 by jd3096-mpy

kdschlosser, 2024-09-19:
OK cool.