lvgl-micropython/lvgl_micropython #133: NOT MODULE  ili9341_init

JuliRGomez, 2024-09-26:
Hello.
Using the most recent commit c7f9e3e and the following code. 
I got this error 

_**_File "display_driver_framework.py", line 487, in init
ImportError: no module named '_ili9341_init'_**_
```
import lvgl as lv
import lcd_bus
import ili9341
from machine import SPI, Pin

spi_bus = SPI.Bus(
    host=1,
    mosi=13,
    miso=12,
    sck=14
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=2,
    cs=15,
    freq=24000000,
)

display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=240,
    display_height=320,
    reset_pin=0,
    reset_state=ili9341.STATE_LOW,
    backlight_pin=27,
    color_space=lv.COLOR_FORMAT.RGB565,
    # color_byte_order=ili9341.BYTE_ORDER_BGR,
    rgb565_byte_swap=True
)

display.set_power(True)
display.set_rotation(lv.DISPLAY_ROTATION._90)
display.init()
```
here info about the ESP and display 
[image]
Cheap Yellow Display Board – CYD (ESP32-2432S032) with ili9341 and touch gt911

I have also tried with the documentation code, but i got this **_AttributeError: type object 'SPI' has no attribute 'MSB'_**
```
spi_device = SPI.Device(
    spi_bus=spi_bus,
    freq=10000000,
    cs=3,
    polarity=0,
    phase=0,
    bits=8,
    first_bit=SPI.MSB
)
```
btw, this is the command I used to compile 
```
python3 make.py  esp32 clean BOARD=ESP32_GENERIC  DISPLAY=ili9341 INDEV=gt911
```
I would really appreciate the help if I have made a mistake or if it is something else.

kdschlosser, 2024-09-27:
There are 2 types of the ili9341. so you need to pass a 1 or a 2 when you call init of the display driver.

```python
display.init(1)
```

JuliRGomez, 2024-09-27:
@kdschlosser Is there any other parameter I need to pass? It's just that I'm getting this error.
File "display_driver_framework.py", line 488, in init
TypeError: function takes 1 positional arguments but 0 were given

kdschlosser, 2024-09-27:
I just fixed the problem. Sorry about that. It's a new system that I just wrote to minimize memory use and I forgot to add the parameter top pass. Clone the repo again and give it a go. it should work now.

JuliRGomez, 2024-09-27:
Awsome, by far the easiest way to use LVGL with Micropython. Thank you!

closed 2024-09-27 by JuliRGomez

kdschlosser, 2024-09-27:
Sorry about the little hiccups you were having. There have been a lot of recent changes to this library and there are some bugs that are still needing to be ironed out.

> by far the easiest way to use LVGL with Micropython. Thank you!

That is the exact intention of this project. Easy to use for the newbie but also provides ways to change the behavior to suit the needs of advanced users. The official binding was difficult to get to compile and it didn't offer a whole lot in the way of supported displays and it only supported SPI displays. This project had the goal of providing a way to add additional displays simply and easily and it is done in a manner that is not wasteful of memory. It also allows users to customize init commands should they have a need to.