lvgl-micropython/lvgl_micropython #138: Building for ESP32 S3 Fails

Type-32, 2024-10-02:
Full Output of Terminal on MacOS Sequoia 15.1 Dev Beta
and logs as well

[idf_py_stderr_output_50151.log](https://github.com/user-attachments/files/17219557/idf_py_stderr_output_50151.log)
[idf_py_stderr_output_50618.log](https://github.com/user-attachments/files/17219558/idf_py_stderr_output_50618.log)
[idf_py_stdout_output_50151.log](https://github.com/user-attachments/files/17219559/idf_py_stdout_output_50151.log)
[idf_py_stdout_output_50618.log](https://github.com/user-attachments/files/17219560/idf_py_stdout_output_50618.log)
[terminal.log](https://github.com/user-attachments/files/17219561/terminal.log)

Type-32, 2024-10-03:
I thought I was using DISPLAY=ili9341 ?

kdschlosser, 2024-10-03:
line 477 in the terminal.log file.

> RuntimeError: Display not found "ili9433"

Type-32, 2024-10-03:
apologies for that, that was my fault. that was the output for the command when I ran it the first time, which I corrected in the subsequent times running the command, eventually coming down to `python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=ili9341 INDEV=gt911`.

This line of command basically did not work and I tried setting INDEV to xpt2046 as well but also didn't work and didn't find a firmware.bin or bootloader.bin in the output folders. By this point I'm pretty sure I'm looking somewhere wrong...

kdschlosser, 2024-10-03:
lets start out from square one.

Run the build command adding `> output.txt` to the end of it and when it is done/fails then attach that file to a post here. That will help me the most in determining what is going on.

Type-32, 2024-10-04:
[output.txt](https://github.com/user-attachments/files/17252730/output.txt)

I hope this helps. I'm running the same command from before
> `python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=ili9341 INDEV=gt911`

kdschlosser, 2024-10-04:
do me a favor. in the build directory there is a file called manifest.py could you attach that file?? to do that you need to add a .txt onto the end of the file name.

kdschlosser, 2024-10-04:
scratch that.

I found the problem and I fixed it.

Type-32, 2024-10-04:
[manifest.py.txt](https://github.com/user-attachments/files/17253558/manifest.py.txt)

(I'm still going to upload the file anyways)
is the issue fixed? If it's fixed then I can close this issue :)

kdschlosser, 2024-10-04:
it is fixed.

spienz, 2024-10-04:
Hy,
my Build for the S3 is also not working, i think the bin file is to small (2,40 MB (2.523.136 Bytes)). There is also a partion Warning in the log.
I need the bin for this device.
Maybe you can help.
My Git clone is up to date

THX

Sebastian

https://www.elecrow.com/esp32-display-5-inch-hmi-display-rgb-tft-lcd-touch-screen-support-lvgl.html
[output.txt](https://github.com/user-attachments/files/17255851/output.txt)

Type-32, 2024-10-04:
> Hy, my Build for the S3 is also not working, i think the bin file is to small (2,40 MB (2.523.136 Bytes)). There is also a partion Warning in the log. I need the bin for this device. Maybe you can [...]

The build process ran smoothly.
However, I am getting the same issues on the built firmware as @spienz . Flashing the firmware on to my board (using the command from the final output of the make.py command) and restarting & resetting the board doesn't switch it out of bootloader mode; it stays in the bootloader mode, so the built firmware is probably messed up or sth.

I'm not sure whether this is still related to this repo or not. My guess is it's probably something on micropython or lv_bindings library's side of stuff but I am not in a position to make this conjecture.

kdschlosser, 2024-10-04:
after you flash it you need to press the reset button on the display. that will take it out of boot loader mode.

Type-32, 2024-10-04:
I did press it after flashing it multiple times. It just stays in bootloader mode. I also checked that the firmware is flashed on to the thing correctly.

kdschlosser, 2024-10-04:
if you are using the USB port then you need to add `--usb-otg` to the build command. This will enable the REPL output to the USB CDC on the ESP32.

spienz, 2024-10-04:
Hy Reset button won't help, i can flash it with the esp windows tool.
But i will get some error regarding the partion table
How can i solve this

`ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cdd3d
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x105c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbd8
load:0x403cc700,len:0x2e30
SHA-256 comparison failed:
Calculated: e0725c018106b73308f8850f643e235629842a1db92caf46b93d064afc8dd5fa
Expected: f1a4a014e8fdd3480f4645b0ef5c8d91c9dfc54b46f452c5f9184f449e1583ed
Attempting to boot anyway...
entry 0x403c989c
E (50) flash_parts: partition 3 invalid - offset 0x268000 size 0x598000 exceeds flash chip size 0x400000
E (50) boot: Failed to verify partition table
E (54) boot: load partition table error!`

Greetings

Sebastian

kdschlosser, 2024-10-04:
your flash chip isn't the size that you passed in the build command,

spienz, 2024-10-05:
ok, thats was a little bit stupid from my side.
Works now. 
But i'm getting another error:

entry 0x403c989c
E (422) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (423) esp_psram: PSRAM enabled but initialization failed. Bailing out.

tried to play with BOARD_VARIANT=SPIRAM_OCT, BOARD_VARIANT=SPIRAM but wont help

my board should have 8mb PSRAM

any suggestions?

THX

kdschlosser, 2024-10-05:
but not octal ram. remove the board variant from the build command.

kdschlosser, 2024-10-05:
I have that exact same display sitting right here in front of me.

this is the build command...

and this is the proper way to use the driver...

```
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=8 DISPLAY=rgb_display INDEV=gt911
```

```python
from micropython import const  # NOQA
from i2c import I2C
import gt911
import lcd_bus
import task_handler
import lvgl as lv  # NOQA
import rgb_display

_WIDTH = const(800)
_HEIGHT = const(480)

_BUFFER_SIZE = const(76800)

_CTP_SCL = const(9)
_CTP_SDA = const(8)
_CTP_IRQ = const(4)

_SD_MOSI = const(11)
_SD_SCK = const(12)
_SD_MISO = const(13)

_LCD_FREQ = const(13000000)
_PCLK_ACTIVE_NEG = const(0)

_HSYNC_PULSE_WIDTH = const(10)
_HSYNC_BACK_PORCH = const(10)
_HSYNC_FRONT_PORCH = const(10)

_VSYNC_PULSE_WIDTH = const(10)
_VSYNC_BACK_PORCH = const(10)
_VSYNC_FRONT_PORCH = const(20)

_PCLK = const(7)
_HSYNC = const(46)
_VSYNC = const(3)
_DE = const(5)
_DISP = None
_BCKL = None
_DRST = None

I2C_BUS = I2C.Bus(
    host=1,
    scl=_CTP_SCL,
    sda=_CTP_SDA,
    freq=400000,
    use_locks=False
)

TOUCH_DEVICE = I2C.Device(
    I2C_BUS,
    dev_id=gt911.I2C_ADDR,
    reg_bits=gt911.BITS
)

_DATA15 = const(10)  # B7
_DATA14 = const(17)  # B6
_DATA13 = const(18)  # B5
_DATA12 = const(38)  # B4
_DATA11 = const(14)  # B3
_DATA10 = const(21)  # G7
_DATA9 = const(47)  # G6
_DATA8 = const(48)  # G5
_DATA7 = const(45)  # G4
_DATA6 = const(0)  # G3
_DATA5 = const(39)  # G2
_DATA4 = const(40)  # R7
_DATA3 = const(41)  # R6
_DATA2 = const(42)  # R5
_DATA1 = const(2)  # R4
_DATA0 = const(1)  # R3

bus = lcd_bus.RGBBus(
    hsync=_HSYNC,
    vsync=_VSYNC,
    de=_DE,
    pclk=_PCLK,
    data_pins=(
        _DATA0,
        _DATA1,
        _DATA2,
        _DATA3,
        _DATA4,
        _DATA5,
        _DATA6,
        _DATA7,
        _DATA8,
        _DATA9,
        _DATA10,
        _DATA11,
        _DATA12,
        _DATA13,
        _DATA14,
        _DATA15
    ),
    freq=_LCD_FREQ,
    hsync_front_porch=_HSYNC_FRONT_PORCH,
    hsync_back_porch=_HSYNC_BACK_PORCH,
    hsync_pulse_width=_HSYNC_PULSE_WIDTH,
    hsync_idle_low=False,
    vsync_front_porch=_VSYNC_FRONT_PORCH,
    vsync_back_porch=_VSYNC_BACK_PORCH,
    vsync_pulse_width=_VSYNC_PULSE_WIDTH,
    vsync_idle_low=False,
    de_idle_high=False,
    pclk_idle_high=False,
    pclk_active_low=_PCLK_ACTIVE_NEG,
)

buf = bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM)

display = rgb_display.RGBDisplay(
    data_bus=bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    frame_buffer1=buf,
    frame_buffer2=None,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display.set_power(True)
display.init()
display.set_backlight(100)
indev = gt911.GT911(TOUCH_DEVICE)

# display.set_rotation(lv.DISPLAY_ROTATION._90)  # NOQA

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.center()

task_handler.TaskHandler()
```

kdschlosser, 2024-10-05:
don't use the USB CDC port. use the other USB when flashing and connecting to the repl

spienz, 2024-10-05:
Nope, this was my fist try without board variant 

`Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x105c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbd8
load:0x403cc700,len:0x2e30
entry 0x403c989c
E (422) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (423) esp_psram: PSRAM enabled but initialization failed. Bailing out.
LVGL MicroPython 1.23.0 on 2024-10-05; Generic ESP32S3 module with ESP32S3
Type "help()" for more information.
`

kdschlosser, 2024-10-05:
did you try the build command I posted?

kdschlosser, 2024-10-05:
I am compiling it now to test on my display to see if it is working.

spienz, 2024-10-05:
Yes, almost same build command, flash Size 4. with 8 i become the Partition Table error

kdschlosser, 2024-10-05:
then the display you are referencing is not the same display. the display you are referencing has 8mb of flash. Maybe you got a bad display?

kdschlosser, 2024-10-05:
OK, I see what is happening. You need to open your own issue instead of hijacking one that was not made by you. You are getting me all kinds of confused.

spienz, 2024-10-05:
But from the same family, no problem, i open a seperate issue

kdschlosser, 2024-10-05:
2 different discussions at this point. I am getting mixed up between the 2 people.

Type-32, 2024-10-06:
I'm going to close this issue since the issue that the title suggests is fixed.

closed 2024-10-06 by Type-32

Type-32, 2024-10-06:
I opened the discussion at #142 specifically for talking about setting up the 4.3 touch board.