lvgl-micropython/lvgl_micropython #139: lcd_panel_io_tx_color error

dcmcshan, 2024-10-02:
Traceback (most recent call last):
  File "/lib/io/mct_thread.py", line 418, in scheduler
  File "/lib/io/mct_thread.py", line 199, in run
  File "/lib/io/MCT_DISPLAY.py", line 113, in update_display
  File "display_driver_framework.py", line 592, in _flush_cb
OSError: 257(lcd_panel_io_tx_color)

What is this error?

kdschlosser, 2024-10-02:
well lets find out what error 257 is in the esp-idf. Lets convert that to hex and do a search of the IDF error codes.

kdschlosser, 2024-10-02:
0x101 is the code.

kdschlosser, 2024-10-02:
ESP_ERR_NO_MEM.. out of memory

kdschlosser, 2024-10-02:
lets take a look and see what the code is for transmitting the color in the esp_lcd module. Last I knew you were using the SPI bus and that is what I am going to look at.

dcmcshan, 2024-10-02:
Good to know.  I have started to implement a BLE peripheral.  It's possible/likely that it has some sort of memory issue...

kdschlosser, 2024-10-02:
what I would do is add DEBUG to your build command and this will tell you exactly where the error is happening and how we can go about fixing it. 

I am going to tell you right now that you are going to have issues if you use the BT or even the WiFi due to issues with sharing the ADC. You need to check your pin configuration and make sure you are not using the ADC that conflicts with BT or WiFi.

kdschlosser, 2024-10-02:
what we don't know is if there is a stack memory issue or if it is an actual running out of RAM issue. That is what we need to know. Stack memory is easily handled. RAM issue could be because of not having enough RAM or it could be due to fragmentation of the RAM. 

There are things we can do to mitigate both of these problems.

You have seen my code a lot by now I am sure.

ever wonder why you see code like this??

```python
data = bytearray(100}
mv = memoryview(data)

data[:10] = bytearray([0x00, ..., 0x09]
send_data(mv[:10])

data[:5] = bytearray([0x00, ..., 0x04]
send_data(mv[:5])
```

This stop memory fragmentation if you create the buffer and the memoryview at a global level so it gets allocated when your program starts and it gets reused. It not only keeps the memory from getting fragmented it also speeds up the program execution times.

kdschlosser, 2024-10-02:
the BLE consumes a HUGE amount of memory. It consumes very specific type of memory as well. You could be having an issue because of running out of DMA memory that is in internal RAM.

How are you starting the display driver? are you creating the frame buffers yourself or are you letting the driver create the buffers?
You might want to allocate the buffers in SPIRAM instead of internal RAM. Internal RAM is going to be faster memory but you might not have enough internal DMA'able memory to satisfy both the display driver and BLE. This is the reason why I provided the ability for the user to allocate their own frame buffers. Providing that ability means they can choose what memory to use.

dcmcshan, 2024-10-02:
When you say add debug to command line do you mean like this?

`
python3 make.py esp32 clean submodules mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16 DISPLAY=ST7735 --usb-otg  DEBUG
`

I allocated the frame buffers explicitly like this: 
```

        display_memory_size = 80*160*lv.COLOR_FORMAT.RGB565
        self.fb1 = self.display_bus.allocate_framebuffer(display_memory_size, lcd_bus.MEMORY_SPIRAM)
        self.fb2 = self.display_bus.allocate_framebuffer(display_memory_size, lcd_bus.MEMORY_SPIRAM)	
        
        self.display = ST7735.ST7735R_Red(
            data_bus=self.display_bus,
            display_width=80,
            display_height=160,
            # we are going to let the driver handle the allocation of the frame buffers
            frame_buffer1=self.fb1,
            frame_buffer2=self.fb2,
            reset_pin=PIN_LCD_RESET,
            reset_state=ST7735.STATE_LOW,
            power_on_state=ST7735.STATE_HIGH,
            #backlight_pin=PIN_LCD_BACKLIGHT,
            #backlight_on_state=ST7735.STATE_LOW,
            offset_x=26,
            offset_y=1,
            color_space=lv.COLOR_FORMAT.RGB565,
            rgb565_byte_swap=True,
            color_byte_order=ST7735.BYTE_ORDER_BGR
        )

```

dcmcshan, 2024-10-03:
Looks like it takes about 7k, and there should be plenty RAM and SPIRAM free

```
Initial free memory: 7443328
Starting BLE service...
Services registered
BLE advertising started
BLE service started successfully
Free memory after BLE init: 7435600
stack: 688 out of 15360
GC: total: 255936, used: 160368, free: 95568, max new split: 7340032
 No. of 1-blocks: 1882, 2-blocks: 484, max blk sz: 562, max free sz: 581
```

dcmcshan, 2024-10-03:
I pulled most recent version of repo, that build command above results in: 

`make: *** No rule to make target `mpy_cross'.  Stop.`

Do I need to re clone from scratch?

kdschlosser, 2024-10-03:
That's your problem. You are allocating the buffers incorrectly. 

This is wrong...
```python
display_memory_size = 80*160*lv.COLOR_FORMAT.RGB565
self.fb1 = self.display_bus.allocate_framebuffer(display_memory_size, lcd_bus.MEMORY_SPIRAM)
self.fb2 = self.display_bus.allocate_framebuffer(display_memory_size, lcd_bus.MEMORY_SPIRAM)	
```

it should be

```python
display_memory_size = int(80 * 160 * 2 / 10)
self.fb1 = self.display_bus.allocate_framebuffer(display_memory_size, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
self.fb2 = self.display_bus.allocate_framebuffer(display_memory_size, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)	
```

kdschlosser, 2024-10-03:
don't worry about the DEBUG.. problem is already figured out...

You were making 2 frame buffers that were `80 * 160 * 18` in size.. That's 230400 bytes per frame buffer. The width and height get passed to the bus driver and it's math wasn't working properly for the buffer size being passed. Hence the memory error.

Since you are using such a small display and because you are using double buffering you need to be using DMA memory and because the buffers only need to be 1/10th of the size because the display has GRAM you can easily fir the buffers into internal memory which is a hell of a lot faster than the SPIRAM.

kdschlosser, 2024-10-04:
and if you want to see all of the back end debugging data you add `--debug` to your build command.

closed 2024-10-05 by dcmcshan

kdschlosser, 2024-10-05:
you closed this but you didn't tell me if changing the frame buffer size fixed your problem.

dcmcshan, 2024-10-05:
Oh, apologies.  It seems to have!

kdschlosser, 2024-10-05:
Oh Goodie!!!