lvgl-micropython/lvgl_micropython #141: Building for ESP32 ESP32-S3-WROOM-1-N4R8

spienz, 2024-10-05:
First sorry for the confusion.

Trying to bild for this Panel
(https://www.elecrow.com/esp32-display-5-inch-hmi-display-rgb-tft-lcd-touch-screen-support-lvgl.html)

with ESP32 ESP32-S3-WROOM-1-N4R8

Buil command:
[image]

But i'm getting this error

`Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x105c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbd8
load:0x403cc700,len:0x2e30
entry 0x403c989c
E (422) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (423) esp_psram: PSRAM enabled but initialization failed. Bailing out.
LVGL MicroPython 1.23.0 on 2024-10-05; Generic ESP32S3 module with ESP32S3
Type "help()" for more information.`

any idea ?

THX

Regards 

Sebastian

kdschlosser, 2024-10-05:
I am working on that now... Elecrow isn't exactly forth coming with the specifications for that board. There are configuration flags that can be set that will alter how the ram gets set up. Problem is I need to know what the ram is in order to set it up properly. That is what I am digging into.

according to Elecrow the MCU that is being used is the ESP32-S3-WROOM-1-N4R8

and according to Espressif that MCU has 4 MB of quad SPI flash and 8 MB of octal SPI RAM

looking at the error...

```
Build:Mar 27 2021 
rst:0x1 (POWERON), boot:0x8 (SPI_FAST_FLASH_BOOT) 
SPIWP:0xee 
mode:DIO, clock div:1 
load:0x3fce3820, len:0x105c 

load:0x403c9700,len:0x4 
load:0x403c9704,len:0xbd8 
load:0x403cc700,len:0x2e30 
entry 0x403c989c E (422) 
quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode E (423) esp_psram: PSRAM enabled but initialization failed. Bailing out. 

LVGL MicroPython 1.23.0 on 2024-10-05; Generic ESP32S3 module with ESP32S3 Type "help()" for more information.
```

I know what this is..

Do you get a REPL prompt? You should be...

at the REPL prompt type in the following...

```
import gc
gc.mem_free()
```

and tell me what the output is.

kdschlosser, 2024-10-05:
OH... You should also be compiling using `BOARD_VARIANT=SPIRAM_OCT`

spienz, 2024-10-05:
Hey,

thanks for reply....

the answer is 203152

I tried all variants of building

[image]

kdschlosser, 2024-10-05:
I need you to use this explicit build command

```
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=4
```

while the build is happening go into the `build` folder and copy the sdkconfig file and attach it to a post.

spienz, 2024-10-05:
i got the file

[sdkconfig.txt](https://github.com/user-attachments/files/17267377/sdkconfig.txt)

spienz, 2024-10-05:
i also had the original firmware from elecrow, its outdated, but maybe it could be helpful to get te right settings?

kdschlosser, 2024-10-05:
are you setting the BOARD_VARIANT correctly?

It is `BOARD_VARIANT=SPIRAM_OCT` It looks like you have set it as`BOARD_VARIANT=SPIRAM=OCT` which is incorrect

spienz, 2024-10-06:
OMG, your are right i tried all the time the wrong binary in my compiling vm i had booth.
Now it works 

THX

kdschlosser, 2024-10-06:
so it is up and running?

spienz, 2024-10-06:
hy, it looks like, no error, repl with no error, wifi is working, trying sd access but have to change the code because sdcard behavior is changed. 
Then i check the display.

kdschlosser, 2024-10-06:
the SD card has been changed so that you are now able to use the SD Card on the same bus as a touch panel or a display. With vanilla MicroPython you have to have a dedicated SPI bus for the SD Card. There is no way to share the bus with anything else. 

I corrected that issue. You need to create an spi.Bus instance and then create an SDCard instance passing the spi.Bus instance to the constructor of the SDCard.

so you can have a single SPI bus and on that bus you can have an SDCard, touch panel, display and any other devices you want. This was not possible with the original MicroPython design.

The best part is it's all thread safe too and you do not have to worry about the CS pins as those are all handled internally for you. Just pass the pin number to whatever it is you are adding to the bus and it is all managed internally. Currently the only thing that is non blocking when transmitting is the display. I still have to add that ability to the other types of devices.

spienz, 2024-10-07:
ok thanks for that, i will try to do it :-)

spienz, 2024-10-09:
Build is running THX

closed 2024-10-09 by spienz