lvgl-micropython/lvgl_micropython #143: How to add Modules like usqlite

spienz, 2024-10-06:
Hy,

how can i add modules like usqlite within the build process?
https://github.com/spatialdude/usqlite

Regards

Sebastian

kdschlosser, 2024-10-06:
```
cd ext_mod
git clone https://github.com/spatialdude/usqlite
echo "include(\${CMAKE_CURRENT_LIST_DIR}/usqlite/micropython.cmake)" >> micropython.cmake
```

pretty simple to do.

spienz, 2024-10-07:
Thx, i will test it this evening

spienz, 2024-10-07:
Hy,

it could be, but it isn't....
Getting a lot of errors....

Don't kno why, opend a Issue https://github.com/spatialdude/usqlite/issues/31

Regards

Sebastian

`from /home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:14678,
from /home/ubuntu/lvgl_micropython/ext_mod/usqlite/usqlite.c:50:
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c: In function 'sqlite3PcacheFetchFinish':
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:54760:11: error: implicit declaration of function 'sqlite3PcachePageSanity'; did you mean 'sqlite3PcachePagecount'? [-Werror=implicit-function-declaration]
54760 |   assert( sqlite3PcachePageSanity(pPgHdr) );
|           ^~~~~~~~~~~~~~~~~~~~~~~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c: In function 'pcache1TruncateUnsafe':
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:55844:11: error: 'nPage' undeclared (first use in this function); did you mean 'DbPage'?
55844 |   assert( nPage<0 || pCache->nPage==(unsigned)nPage );
|           ^~~~~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:55844:11: note: each undeclared identifier is reported only once for each function it appears in
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c: In function 'pcache1Fetch':
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:56213:11: error: 'pCache' undeclared (first use in this function); did you mean 'PCache'?
56213 |   assert( pCache->bPurgeable || createFlag!=1 );
|           ^~~~~~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c: In function 'pager_end_transaction':
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:59158:11: error: implicit declaration of function 'assert_pager_state' [-Werror=implicit-function-declaration]
59158 |   assert( assert_pager_state(pPager) );
|           ^~~~~~~~~~~~~~~~~~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c: In function 'subjournalPage':
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:61637:13: error: implicit declaration of function 'pageInJournal' [-Werror=implicit-function-declaration]
61637 |          || pageInJournal(pPager, pPg)
|             ^~~~~~~~~~~~~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c: In function 'sqlite3PagerUnrefNotNull':
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:62874:33: error: 'pPager' undeclared (first use in this function); did you mean 'Pager'?
62874 |   assert( sqlite3PcacheRefCount(pPager->pPCache)>0 ); /* tag-20230419-2 */
|                                 ^~~~~~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c: In function 'walIndexPage':
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:65633:39: error: 'Wal' has no member named 'nSehTry'
65633 | # define SEH_INJECT_FAULT assert( pWal->nSehTry>0 );
|                                       ^~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:65711:3: note: in expansion of macro 'SEH_INJECT_FAULT'
65711 |   SEH_INJECT_FAULT;
|   ^~~~~~~~~~~~~~~~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c: In function 'walCkptInfo':
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:65633:39: error: 'Wal' has no member named 'nSehTry'
65633 | # define SEH_INJECT_FAULT assert( pWal->nSehTry>0 );
|                                       ^~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:65723:3: note: in expansion of macro 'SEH_INJECT_FAULT'
65723 |   SEH_INJECT_FAULT;
|   ^~~~~~~~~~~~~~~~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c: In function 'walIndexHdr':
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:65633:39: error: 'Wal' has no member named 'nSehTry'
65633 | # define SEH_INJECT_FAULT assert( pWal->nSehTry>0 );
|                                       ^~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:65732:3: note: in expansion of macro 'SEH_INJECT_FAULT'
65732 |   SEH_INJECT_FAULT;
|   ^~~~~~~~~~~~~~~~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c: In function 'walFramePgno':
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:65633:39: error: 'Wal' has no member named 'nSehTry'
65633 | # define SEH_INJECT_FAULT assert( pWal->nSehTry>0 );
|                                       ^~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:66117:3: note: in expansion of macro 'SEH_INJECT_FAULT'
66117 |   SEH_INJECT_FAULT;
|   ^~~~~~~~~~~~~~~~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c: In function 'walIndexRecover':
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:65633:39: error: 'Wal' has no member named 'nSehTry'
65633 | # define SEH_INJECT_FAULT assert( pWal->nSehTry>0 );
|                                       ^~
[... 318 lines cut ...]
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c: In function 'sqlite3Pragma':
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:140564:17: error: implicit declaration of function 'sqlite3NoTempsInRange'; did you mean 'sqlite3GetTempRange'? [-Werror=implicit-function-declaration]
140564 |         assert( sqlite3NoTempsInRange(pParse,1,7+j) );
|                 ^~~~~~~~~~~~~~~~~~~~~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c: In function 'sqlite3InitCallback':
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:141949:25: error: 'rcp' undeclared (first use in this function); did you mean 'rc'?
141949 |     assert( (rc&0xFF)==(rcp&0xFF) );
|                         ^~~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c: In function 'sqlite3RunParser':
[... cut ...]
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:208868:18: warning: comparison of unsigned expression in '>= 0' is always true [-Wtype-limits]
208868 |   assert( iTarget>=0 && iTarget<pTarget->nBlob );
|                  ^~
/home/ubuntu/lvgl_micropython/ext_mod/usqlite/sqlite3.c:208869:17: warning: comparison of unsigned expression in '>= 0' is always true [-Wtype-limits]
208869 |   assert( iPatch>=0 && iPatch<pPatch->nBlob );
|                 ^~
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.`

spienz, 2024-10-09:
Get it to run, had to Change makefile of usql. the ESP32 query won't work. So i did it hard,  and the it works.

closed 2024-10-09 by spienz