lvgl-micropython/lvgl_micropython #145: Unix port : KO ; typo ?

picpic020960, 2024-10-08:
Bonjour ,
New build on raspberry pi with raspberry pi os ( 07/2024)
And error bellow
 ```GEN build-standard/genhdr/mpversion.h                                 GEN build-standard/genhdr/qstr.i.last
In file included from ../../extmod/modmachine.c:61:                   ../../ports/unix/modmachine.c:81:42: error: backslash-newline at end of file [-Werror]                                                      81 | #define MICROPY_PY_MACHINE_EXTRA_GLOBALS \
|                                                                     cc1: note: unrecognized command-line option ‘-Wno-unused-command-line-argument’ may have been intended to silence earlier diagnostics       cc1: all warnings being treated as errors
Command '['gcc', '-E', '-I/home/pi/lvgl_micropython/ext_mod/lcd_bus', '-I/home/pi/lvgl_micropython/ext_mod/lcd_bus/common_include', '-I/home/pi/lvgl_
```
Thorry for the *LF/CR*
```
../../ports/unix/modmachine.c:81:42: error: backslash-newline at end of file 
```

Thank's fort help
Regards

kdschlosser, 2024-10-08:
I am still working through some glitches with a new implementation of the task_handler. I am attempting to use SDL to handle the timer that is used to call `lv_tick_inc` and `lv_task_handler`. The reason for this change is because of macOS and the timer functionality not being available.

The build failure you are having I have fixed locally and I have not yet pushed a commit for it. There is another issue that comes up after that one is fixed which is the callback never gets called when the timer expires. It gets called in C code but the python function never gets called. I do not know why it is not getting called but it isn't. There is no error that is produced or a segfault occurs with different ways I have tried to make the callback happen. I am in the process of debugging to see what is going on. It is going to be another few days before I am able to get it running properly. 

I will let you know what I have gotten it working properly and I have updated the code.

picpic020960, 2024-10-08:
Thank's for this info
I have Time ,I'am retired

kdschlosser, 2024-10-08:
I am as well. or rather I should be. Seems my retirement is mostly spent working on my house and cars... LOL

kdschlosser, 2024-10-09:
OK this should be fixed. Give it a try and see if it is...

picpic020960, 2024-10-15:
Bonjour Kevin ,
Build unix OK (unable to find alsa ???)
But README.md script  ==> segment faut AT the end of script
```
>>> import task_handler
>>> th = task_handler.TaskHandler(duration=5)
>>> Erreur de segmentation
```
Raspberry pi zero headless , with VNC viewer from tablette for screen and ssh.
Thank's for help

kdschlosser, 2024-10-16:
i don't know what is special about the Pi's OS but there is something that is special about it because it works on Ubuntu. Maybe SDL doesn't work with RPI. IDK...

kdschlosser, 2024-10-16:
How are you compiling it? are you compiling it on the Pi? If you aren't then that is the problem. You need to compile it on the Pi for it to run on the Pi.

kdschlosser, 2024-10-16:
There are some dependencies you need to have installed in order for it to compile.

```
sudo apt-get install  libudev-dev libasound2-dev libdbus-1-dev libraspberrypi0 libraspberrypi-bin libraspberrypi-dev
```

There is also an optional flag that should be set depending on the version of the Pi you are using. I am going to do some digging to see if there is a way for me to be able to identify that from the build script so I can set some flags for the SDL2 build.

kdschlosser, 2024-10-16:
OK so I added support for compiling this on the raspberry pi.  It automatically detects if the pi supports neon and adds the needed switch to activate using neon in SDL2

compile using this command

```
python3 make.py raspberry_pi DISPLAY=sdl_display INDEV=sdl_pointer
```

That should do the trick for ya.

picpic020960, 2024-10-16:
Bonjour Kevin ,
[strace.txt](https://github.com/user-attachments/files/17391393/strace.txt)
Attached strace of README.md script.

No 'NEON' in /proc/cpuinfo

dependencies added.

Exists libSDL2.a in build tree. IS  possible to  test it ?

How to upgrade the repo without loose the ESP32 build ? (I'am not easy with *git*)

(rebuild unix on raspberry pi 0 = one Big hour !)

Thank's for help

picpic020960, 2024-10-20:
Bonjour Kevin,
Test make.py rasperry_pi ...
Some issues with directories actuals and raspberry_py , passed .
'parallel' option commented 
Now issue : 
```
[100%] Building C object CMakeFiles/SDL2_test.dir/src/test/SDL_test_ra
[100%] Linking C static library libSDL2_test.a                        
[100%] Built target SDL2_test                                                                                                               
make : on entre dans le répertoire « /home/pi/lvgl_micropython/lib/micropython/ports/unix »                                                 Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.                                                          Including User C Module from /home/pi/lvgl_micropython/ext_mod/lvgl   
Including User C Module from /home/pi/lvgl_micropython/ext_mod/lcd_bus
Including User C Module from /home/pi/lvgl_micropython/ext_mod/lcd_utils                                                                    LVGL-GEN build-standard/lv_mpy.c                                      
make: ***  Aucune règle pour fabriquer la cible « machine_timer.c », nécessaire pour « build-standard/genhdr/qstr.i.last ». Arrêt.
make: *** Attente des tâches non terminées....
python3: can't open file '/home/pi/lvgl_micropython/ext_mod/lcd_utils/gen/python_api_gen_mpy.py': [Errno 2] No such file or directory
/home/pi/lvgl_micropython/ext_mod/lvgl/micropython.mk:38 : la recette pour la cible « build-standard/lv_mpy.c » a échouée                   
make : on quitte le répertoire « /home/pi/lvgl_micropython/lib/micropython/ports/unix »                                                     make: *** [build-standard/lv_mpy.c] Erreur 2
```

Thank's for help

kdschlosser, 2024-10-20:
Je ne suis pas sûr que nous puissions effectuer une compilation croisée pour que vous puissiez compiler sur un ordinateur de bureau.

SI SDL2 a été compilé avec succès, il ne compilera pas SDL2 à nouveau à moins que vous ne nettoyiez la compilation.

picpic020960, 2024-10-20:
ESP32 ==> raspberry pi zero = OK
UNIX(raspberry_pi) ==> raspberry pi 3B+ = KO
One trick ?

kdschlosser, 2024-10-20:
KO? what is KO?

kdschlosser, 2024-10-20:
also have you tried pulling new updates down from the repo? I just did a sizeable update to the unix build system last night and new changes might correct any problems you are currently having.

picpic020960, 2024-10-20:
git pull : no update.
machine_timer.c : no rules...
(see the message above , 6 jours ago)

picpic020960, 2024-10-30:
Bonjour Kevin ,
After some tricks , lvgl works on termux/tablett Android with VNC
[image]
With shared library ( no success with SDL2 build : termux IS linux or android-linux ).
 But no success on raspberry_pi , with unix or raspberry_pi target. Seems issues with make variables : strange , the same repo works on termux.

Now esp32-s3 with qspi display ...

kdschlosser, 2024-10-30:
android-linux is not linux. It's like macOS which is linux based but once again it is not linux and it needs to have code added/changed so the binding will run on on it. MicroPython does not support running on android, only macOS and linux. That being said.. I could add android support I am sure. The time investment would be pretty substantial to do that as I would need to go through and locate all of the things in MicroPython that cause an issue with running on android.

kdschlosser, 2024-10-30:
I am not sure why there is a requirement for `libSDL2-2.0.so`. This is probably because clang is being used to compile and not gcc. clang doesn't allow me to link statically to SDL. GCC does. on macOS SDL2 is linked dynamically and the shared library for SDL2 MUST be in the same folder as the micropython executable. I have only written in support for using clang in the macOS port. If clang is what is needing to be used when compiling for android then I will have to make a completely new port option in the build system and handle things accordingly.

picpic020960, 2024-10-30:
For unix/termux how to enhanced the colors ?
Rotation : the display not rotate (and is smaller ?...) but the touche/indev yes

picpic020960, 2024-10-31:
```
Raspberry pi 3Bbuild-essential                12.3                      
libffi-dev                     3.2.1-6                   
pkg-config                     0.29-4                    
cmake                          3.7.2-1                   
ninja-build                    1.7.2-1                   
gnome-desktop-testing          2016.1-1                  
libasound2-dev                 1.1.3-5+rpt4              
libpulse-dev                   10.0-1+deb9u1             
libaudio-dev                   1.9.4-5                   
libjack-dev                    1:0.125.0-2               
libsndio-dev                   1.1.0-3                   
libx11-dev                     2:1.6.4-3+deb9u4          
libxext-dev                    2:1.3.3-1                 
libxrandr-dev                  2:1.5.1-1                 
libxcursor-dev                 1:1.1.14-1+deb9u2         
libxfixes-dev                  1:5.0.3-1                 
libxi-dev                      2:1.7.9-1                 
libxss-dev                     1:1.2.2-1                 
libxkbcommon-dev               0.7.1-2~deb9u1            
libdrm-dev                     2.4.74-1                  
libgbm-dev                     13.0.6-1+rpi2             
libgl1-mesa-dev                13.0.6-1+rpi2             
libgles2-mesa-dev              13.0.6-1+rpi2             
libegl1-mesa-dev               13.0.6-1+rpi2             
libdbus-1-dev                  1.10.32-0+deb9u1          
libibus-1.0-dev                1.5.14-3+deb9u2           
libudev-dev                    232-25+deb9u14            
fcitx-libs-dev                 1:4.2.9.1-6               
libpipewire-0.3-dev            Not installed	**not exists for RPI**             
libwayland-dev                 1.12.0-1+deb9u1           
libdecor-0-dev                 Not installed	**not exists for RPI**

```

picpic020960, 2024-12-01:
Bonjour Kevin,
OK on raspberry pi 4 with raspberry pi OS (64 bits) with static library SDL2

closed 2024-12-01 by picpic020960