lvgl-micropython/lvgl_micropython #146: ESP_ERR_FLASH_NOT_INITIALISED

dcmcshan, 2024-10-09:
With this build command 

`python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16 DISPLAY=ST7735 --usb-otg --dual-core-threads`

I get this error on micropython load:

```
Traceback (most recent call last):
  File "_boot.py", line 11, in <module>
  File "inisetup.py", line 38, in setup
  File "inisetup.py", line 7, in check_bootsec
OSError: (-24579, 'ESP_ERR_FLASH_NOT_INITIALISED')
```

Running with fresh checkout of main + that some core_id hack
I did not see this before.

kdschlosser, 2024-10-09:
You need to read the beginning of the README. it tells you the recent changes I have made and especially the ones that are going to break API.

`clean` and `submodules` are no longer used in the build command. It is done automatically for you. The only time you should use the `clean` command is to force the directory to be deleted. otherwise the make clean command is run each and every time you run the build script. `submodules` has been removed completely, It automatically detects what is and is not downloaded and it will download what it needs to when it needs to.

kdschlosser, 2024-10-09:
`ESP_ERR_FLASH_NOT_INITIALISED` usually happens if the flash size is being set incorrectly or if you have octal flash and you don't use the `--octal-flash` build parameter. seeing as how you have 16mb of flash your MCU may or may not have octal flash anything 8mb and less  is going to have quad flash.

if you have WROOM-1 it is going to be quad flash. If it is a WROOM-2 it will have octal flash.. see [WROOM-1](https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf#page=3) and [WROOM-2](https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-2_datasheet_en.pdf#page=3)

dcmcshan, 2024-10-09:
Yes, the issue was that I actually switched processors to an M5 ATOM S3, and didn't change the command line.  Thank you.

closed 2024-10-09 by dcmcshan

kdschlosser, 2024-10-09:
no worries m8. I like simple problems... easy to give you the solution...