lvgl-micropython/lvgl_micropython #147: Mounting SD CARD with ELECROW 5" Display

spienz, 2024-10-09:
Hy,

sorry but i'm hanging with the code. I'm not able to get it run.

`#SDCARD` initialisieren
from machine import SDCard, SPI
import vfs,os

spi_bus = SPI.Bus(
    host=1,
    mosi=11,
    miso=13,
    sck=12
)
      
spi_device = SPI.Device(
    spi_bus=spi_bus,
    freq=100000,
    cs=10,
    polarity=0,
    phase=0,
    bits=8
    )

     #first_bit=SPI.MSB
# initalise sdcard
spi_device.init()
vfs.mount(machine.SDCard(), `"/sd1")`

spienz, 2024-10-09:
Hy,

what ever i try i'm running into errors:

with the micropython bin i'm using bevor it worked like this

# Assign chip select (CS) pin (and start it high)
#cs = machine.Pin(10, machine.Pin.OUT)

#spi = machine.SPI(2,
                  baudrate=100000,
                  polarity=0,
                  phase=0,
                  bits=8,
                  firstbit=machine.SPI.MSB,
                  sck=machine.Pin(12),
                  mosi=machine.Pin(11),
                  miso=machine.Pin(13))
# Initialize SD card
#sd = sdcard.SDCard(spi, cs)

**but know i'm not able to get it running**

from machine import SDCard, SPI
import os

# Initialize SPI bus with shared devices
spi_bus = SPI.Bus(
    host=2,          
    mosi=11,         
    miso=13,
    sck=12
)

# Initialize SD card using the SPI bus
sd = SDCard(spi=spi_bus) #, cs=10)

# Mount the SD card to the filesystem
os.mount(sd, "/sd")
     

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

LVGL MPY version : 1.23.0 on 2024-10-08
IDF version : v5.2
Machine     : Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42199f99  PS      : 0x00060e30  A0      : 0x821bde35  A1      : 0x3fcbf630  
A2      : 0x00000000  A3      : 0x00000004  A4      : 0x3fcbf7dc  A5      : 0x00000006  
A6      : 0x3c335a6c  A7      : 0x3fcbf650  A8      : 0x3c329cb8  A9      : 0x3fcbf5f0  
A10     : 0x00000006  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000006  
A14     : 0x3fcbf668  A15     : 0x3fcbf7f4  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x42199f96:0x3fcbf630 0x421bde32:0x3fcbf650 0x421a36f9:0x3fcbf6c0 0x421a4ce9:0x3fcbf6e0 0x421a4db5:0x3fcbf700 0x40379a92:0x3fcbf720 0x4219ce4b:0x3fcbf7c0 0x421a4ce9:0x3fcbf830 0x421a4cfe:0x3fcbf850 0x421b1e4b:0x3fcbf870 0x421b1fb2:0x3fcbf900 0x421b982a:0x3fcbf950

ELF file SHA256: fd95e5ee5

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382ce6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x105c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbd8
load:0x403cc700,len:0x2e30
entry 0x403c989c
Traceback (most recent call last):
  File "main.py", line 7
IndentationError: unexpected indent
LVGL MicroPython 1.23.0 on 2024-10-08; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.

maybe you have an idea 

THX

Greetings 

Sebastian

kdschlosser, 2024-10-09:
can you decode that backtrace for me?

Here is an example of how to do that.

```
C:\Users\USERNAME\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-addr2line.exe -fe C:\MYPROJECT\build\MYELF.elf 0x4000c269:0x3ffb7810 0x400d5327:0x3ffb7830 0x400d59a2:0x3ffb7850 0x400d600f:0x3ffb7b20 0x400d5039:0x3ffb7b40 0x400d3f17:0x3ffb7b80
```

You will need to adjust the paths and the elf file is named `micropython.elf` and it's located in `lib/micropython/ports/esp32/build-{BUILDPARAMS}/micropython.elf` where `{BUILDPARAMS}` is a combination of the `BOARD` and `BOARD_VARIANT` you supplied to compile.

spienz, 2024-10-09:
i will try to do it, need to install xtensa-esp32-elf-addr2line first and export the elf file form the linux.

kdschlosser, 2024-10-10:
both of them should already exist since you built the firmware. You don't have to export anything or install anything. Do exactly like the command states. Go and loomk in your home directory for .espressif. it will be there. navigate the path like what you see in that example. the addr2line utility is already installed.,m

the elf should also exist since you compiled it.

spienz, 2024-10-10:
Good Mornng form Germany,

shure i have because on this Laptop i haven't ESP-IDF installed under Windows :-) I had Build it on the LINUX VM.
But no problem, there it is, i hope it helps:

mp_obj_get_int_maybe
/home/ubuntu/lvgl_micropython/lib/micropython/py/obj.c:335 (discriminator 1)
machine_hw_spi_bus_make_new
/home/ubuntu/lvgl_micropython/lib/micropython/ports/esp32/machine_hw_spi.c:514
type_call
/home/ubuntu/lvgl_micropython/lib/micropython/py/objtype.c:1003
mp_call_function_n_kw
/home/ubuntu/lvgl_micropython/lib/micropython/py/runtime.c:717
mp_call_method_n_kw
/home/ubuntu/lvgl_micropython/lib/micropython/py/runtime.c:733
mp_execute_bytecode
/home/ubuntu/lvgl_micropython/lib/micropython/py/vm.c:1042
fun_bc_call
/home/ubuntu/lvgl_micropython/lib/micropython/py/objfun.c:267
mp_call_function_n_kw
/home/ubuntu/lvgl_micropython/lib/micropython/py/runtime.c:717
mp_call_function_0
/home/ubuntu/lvgl_micropython/lib/micropython/py/runtime.c:691
parse_compile_execute
/home/ubuntu/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:123
do_reader_stdin
/home/ubuntu/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:280
mp_task
/home/ubuntu/lvgl_micropython/lib/micropython/ports/esp32/main.c:157

THX for your Support

Greetings

Sebastian

spienz, 2024-10-15:
Hy,

hope it's okay to ask, did you find some time looking at it ?

Regards 

Sebastian

spienz, 2024-10-24:
Hy, got it working, some more gray hairs, but know it works.
Reading a lot also old Issues.

Maybe this Code Example helps someone.

THX

```python
from machine import SPI, SDCard
import os, vfs

# Initialize the SPI bus with appropriate pins
spi_bus = SPI.Bus(
    host=2,        # SPI bus host number (adjust based on your board)
    mosi=11,       # Master Out Slave In (adjust based on your hardware)
    miso=13,       # Master In Slave Out (adjust based on your hardware)
    sck=12         # Clock pin (adjust based on your hardware)
)

# Mount SD Card
try:
	sd = SDCard(
        spi_bus=spi_bus,
        cs=10,
        #freq=100000
        freq=1000000
        )
	#os.mount(sd, '/sd')
	vfs.mount(sd, "/sd")
	print('Flash Mounted')
	print(os.listdir('/sd'))
except Exception as er:
	print(f'Flash Error\n{er}\nrebooting')
	time.sleep(5)
	reset()

```

closed 2024-10-24 by spienz

kdschlosser, 2024-10-24:
I was pretty sure the SD Card code works.  If you want you can post the code in  a new discussion so people will be able to find it easier.

kdschlosser, 2024-10-24:
Instead of having the ESP32 do boot loops over and over again this code would probably be a better way to go about it...

```python
import os
import vfs
import machine
import time

# Initialize the SPI bus with appropriate pins
spi_bus = machine.SPI.Bus(
    host=2,        # SPI bus host number (adjust based on your board)
    mosi=11,       # Master Out Slave In (adjust based on your hardware)
    miso=13,       # Master In Slave Out (adjust based on your hardware)
    sck=12         # Clock pin (adjust based on your hardware)
)

sd = None

while sd is None:
    try:
        sd = machine.SDCard(
            spi_bus=spi_bus,
            cs=10,
            freq=1000000
        )
        vfs.mount(sd, "/sd")
        print('Flash Mounted')
        print(os.listdir('/sd'))
    except Exception as er:
        sd = None
        print('sd card failure')
        time.sleep(5)
```

the reason why this is better to do is depending on the reset that is done only the memory could get cleared. The hardware doesn't get reset and that could cause issues. It's going to be more stable like this as it will not allow your code to continue running until the SD is set and you don't have the constant reading of large amounts of data from the internal flash memory.