lvgl-micropython/lvgl_micropython #149: no module named 'ST7735'

dcmcshan, 2024-10-12:
I looked in README, didn't see any mention of this changing?

```
LVGL MicroPython 1.23.0 on 2024-10-12; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
>>> import ST7735
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'ST7735'
```

kdschlosser, 2024-10-12:
it's st7735 not ST7735

dcmcshan, 2024-10-14:
Strange.  But this also doesn't work.  Did the API change entirely?  This has worked for many months...

```
Traceback (most recent call last):
  File "<stdin>", line 123, in <module>
  File "<stdin>", line 69, in __init__
AttributeError: 'module' object has no attribute 'ST7735R_Red'
```

kdschlosser, 2024-10-14:
HAHAHA... I changed the API so you were leading all kinds of code that wasn't going to get used. 

There is no `ST7735R_Red` class anymore. 

```python
import st7735

display = st7735.ST7735(
    ...
)

display.init(st7735.TYPE_R_RED)
```

That will get ya going.

closed 2024-11-05 by kdschlosser

dcmcshan, 2025-01-03:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "MCT_DISPLAY.py", line 49, in <module>
ImportError: no module named 'st7735'

Did this change back?