lvgl-micropython/lvgl_micropython #150: WT32-SC01 (not PLUS) with ESP32 (not S3)

picpic020960, 2024-10-13:
Bonjour Kevin ,
This vidéo for three issues : 
1 - 'hello word' IS inversed
2 - the scroll IS inverted
3 - how to control the backlight
Thanks for help
(After this , return to unix port)
https://github.com/user-attachments/assets/c1869a39-ac58-43c0-bb15-db4584edc13e

kdschlosser, 2024-10-13:
damned 7796 issues. This single display IC is going to be the bane of my existence.

I can fix the reversed text, that's not too big of an issue.

there are 2 possible things that can fix the touch.. The first is trying to set the startup rotation of the display.

```python
import ft6x06

touch = ft6x06.FT6x06(
    ...
    startup_rotation=pointer_framework.lv.DISPLAY_ROTATION._180
)
```

If that doesn't correct the problem you can do a display calibration and that should fix it. 

```python
import ft6x06

touch = ft6x06.FT6x06(
    ...
)

if not touch.is_calibrated:
    touch.calibrate()

```

for the time being to correct the problem with the text being backwards... point the display at a mirror and look at the mirror... :smile:, I'm joking...

Until I am able to get around to updating the driver I am going to have you override part of the display driver...

```python

import st7796

st7796.ST7796._ORIENTATION_TABLE = (0x40, 0xE0, 0x80, 0x20)

# create the driver as usual.
```

picpic020960, 2024-10-14:
Bonjour Kevin,
Calibrate and _ORIENTATION_TABLE works fine
Thank's

On the video, are the colors rights ?
Only this lines about colors in script :
```
color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7796.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,

```
And 
```
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)
```

picpic020960, 2024-10-14:
Kevin ,
Tryed this examples (lvgl v8)
[https://github.com/lvgl/lvgl/blob/4d96c27ce35dd2ea6b34926f24a647e7ea7c4b0c/examples/widgets/slider/lv_example_slider_1.py]
But the slider not respond 
You confirm ?

kdschlosser, 2024-10-14:
you cannot use lvgl8 code it's not going to work..

kdschlosser, 2024-10-14:
and for the display you are using you should have the color byte order set to RGB and set the rgb565 byte swap to True

picpic020960, 2024-10-14:
Thank's Kevin

closed 2024-10-14 by picpic020960