lvgl-micropython/lvgl_micropython #153: Library ntptime With OSError -202

Type-32, 2024-10-15:
Library ntptime gives `OSError -202` when invoking the methods in the library, such as `settime()` and `time()`.

Full REPL Log:
```log
>>> import ntptime
>>> ntptime.settime()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ntptime.py", line 1, in settime
  File "ntptime.py", line 1, in time
OSError: -202
>>> import time
>>> ntptime.settime()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ntptime.py", line 1, in settime
  File "ntptime.py", line 1, in time
OSError: -202
>>> time.localtime()
(2000, 1, 1, 0, 0, 43, 5, 1)
>>> ntptime.time()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ntptime.py", line 1, in time
OSError: -202
>>> help ntptime
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> help(ntptime)
object <module 'ntptime' from 'ntptime.py'> is of type module
  socket -- <module 'socket'>
  utime -- <module 'time'>
  __name__ -- ntptime
  struct -- <module 'struct'>
  __version__ -- 0.1.1
  host -- pool.ntp.org
  timeout -- 1
  __file__ -- ntptime.py
  time -- <function time at 0x3c25e7a0>
  settime -- <function settime at 0x3c25e7d0>
>>> ntptime.settime()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ntptime.py", line 1, in settime
  File "ntptime.py", line 1, in time
OSError: -202
>>> help(ntptime.settime)
object <function settime at 0x3c25e7d0> is of type function
>>> help(ntptime.settime())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ntptime.py", line 1, in settime
  File "ntptime.py", line 1, in time
OSError: -202
>>> 
```

kdschlosser, 2024-10-15:
OK so a few things...

Did you read the MicroPython documentation?? I am guessing not. That module seems to only be referenced in the documentation for the ESP8266. Not sure if it would apply to the ESP32 or not. The other thing is if we follow the directions for the ESP8266 you are using it incorrectly.

```python

from machine import RTC

    rtc = RTC()
    rtc.datetime((2017, 8, 23, 1, 12, 48, 0, 0)) # set a specific date and time
    rtc.datetime() # get date and time

    # synchronize with ntp
    # need to be connected to wifi              <<<< IMPORTANT
    import ntptime
    ntptime.settime() # set the rtc datetime from the remote server
    rtc.datetime()    # get the date and time in UTC

```

Type-32, 2024-10-15:
Oh shoot, my bad 🥶

closed 2024-10-15 by Type-32