lvgl-micropython/lvgl_micropython #154: ELECROW DIS07050H ESP32-S3-WROOM-1-N4R2

spienz, 2024-10-15:
Hy,

i'm getting crazy with that board, trying to getting the Display working with the RGB Driver.
Double checked almost everything, but won't find the Problem. In the txt_config is the Manufacturer PIN Out

So i need help again, please.

THX

Sebastian

```python
>>> from micropython import const  # NOQA
from i2c import I2C
import gt911
import lcd_bus
import task_handler
import lvgl as lv  # NOQA
import rgb_display

_WIDTH = const(800)
_HEIGHT = const(480)

_BUFFER_SIZE = const(76800)

_CTP_SCL = const(20)
_CTP_SDA = const(19)
_CTP_IRQ = None

_SD_MOSI = None
_SD_SCK = None
_SD_MISO = None

_LCD_FREQ = const(14000000)
_PCLK_ACTIVE_NEG = const(0)

_HSYNC_PULSE_WIDTH = const(10)
_HSYNC_BACK_PORCH = const(10)
_HSYNC_FRONT_PORCH = const(10)

_VSYNC_PULSE_WIDTH = const(10)
_VSYNC_BACK_PORCH = const(10)
_VSYNC_FRONT_PORCH = const(20)

_PCLK = const(0)
_HSYNC = const(39)
_VSYNC = const(41)
_DE = const(40)
_DISP = None
_BCKL = const(2)
_DRST = None

I2C_BUS = I2C.Bus(
    host=1,
    scl=_CTP_SCL,
    sda=_CTP_SDA,
    freq=400000,
    use_locks=False
)

TOUCH_DEVICE = I2C.Device(
    I2C_BUS,
    dev_id=gt911.I2C_ADDR,
    reg_bits=gt911.BITS
)

_DATA15 = const(1)  # B7
_DATA14 = const(9)  # B6
_DATA13 = const(46)  # B5
_DATA12 = const(3)  # B4
_DATA11 = const(8)  # B3
_DATA10 = const(4)  # G7
_DATA9 = const(16)  # G6
_DATA8 = const(15)  # G5
_DATA7 = const(7)  # G4
_DATA6 = const(6)  # G3
_DATA5 = const(5)  # G2
_DATA4 = const(14)  # R7
_DATA3 = const(21)  # R6
_DATA2 = const(47)  # R5
_DATA1 = const(48)  # R4
_DATA0 = const(45)  # R3

bus = lcd_bus.RGBBus(
    hsync=_HSYNC,
    vsync=_VSYNC,
    de=_DE,
    pclk=_PCLK,
    data0=_DATA0,
    data1=_DATA1,
    data2=_DATA2,
    data3=_DATA3,
    data4=_DATA4,
    data5=_DATA5,
    data6=_DATA6,
    data7=_DATA7,
    data8=_DATA8,
    data9=_DATA9,
    data10=_DATA10,
    data11=_DATA11,
    data12=_DATA12,
    data13=_DATA13,
    data14=_DATA14,
    data15=_DATA15,
    freq=_LCD_FREQ,
    hsync_front_porch=_HSYNC_FRONT_PORCH,
    hsync_back_porch=_HSYNC_BACK_PORCH,
    hsync_pulse_width=_HSYNC_PULSE_WIDTH,
    hsync_idle_low=False,
    vsync_front_porch=_VSYNC_FRONT_PORCH,
    vsync_back_porch=_VSYNC_BACK_PORCH,
    vsync_pulse_width=_VSYNC_PULSE_WIDTH,
    vsync_idle_low=False,
    de_idle_high=False,
    pclk_idle_high=False,
    pclk_active_low=_PCLK_ACTIVE_NEG,
)

print(bus)
buf = bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM)

display = rgb_display.RGBDisplay(
    data_bus=bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    frame_buffer1=buf,
    frame_buffer2=None,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display.set_power(True)
display.init()
display.set_backlight(100)
indev = gt911.GT911(TOUCH_DEVICE)

# display.set_rotation(lv.DISPLAY_ROTATION._90)  # NOQA

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.center()

task_handler.TaskHandler()

Traceback (most recent call last):
  File "<stdin>", line 74, in <module>
TypeError: 'disp' argument required
>>> 
``` 

https://www.elecrow.com/wiki/esp32-display-502727-intelligent-touch-screen-wi-fi26ble-800480-hmi-display.html#micropython

[image]
[tft_config.txt](https://github.com/user-attachments/files/17382307/tft_config.txt)

spienz, 2024-10-24:
Hy,
get it working. but still have some issues.

Could it only run once, without error. After a reset it works again. See fatal error occurred message.

Touch is working, but the Slider is moving on every klick to the right of the screen until it's completely gone. 

Why will this happen?

Regards

Sebastian

```
A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

LVGL MPY version : 1.23.0 on 2024-10-15
IDF version : v5.2
Machine     : Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42072eb4  PS      : 0x00060030  A0      : 0x82072a9e  A1      : 0x3fcbf200  
A2      : 0x3c3647d4  A3      : 0x00000001  A4      : 0x3c364800  A5      : 0x00009470  
A6      : 0x00000001  A7      : 0x3fcbf200  A8      : 0x00000003  A9      : 0x3c364800  
A10     : 0x3c364800  A11     : 0x3c331dec  A12     : 0x3c33be94  A13     : 0x3c33cd19  
A14     : 0x3c364704  A15     : 0x3fcbf1d0  SAR     : 0x00000020  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000003  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x42072eb1:0x3fcbf200 0x42072a9b:0x3fcbf240 0x4203a5dd:0x3fcbf280 0x42029672:0x3fcbf2c0 0x42004f45:0x3fcbf300 0x421a4ce9:0x3fcbf340 0x421a4db5:0x3fcbf360 0x40379a92:0x3fcbf380 0x4219ce4b:0x3fcbf420 0x421a4ce9:0x3fcbf450 0x421a4db5:0x3fcbf470 0x40379a92:0x3fcbf490 0x4219ce4b:0x3fcbf530 0x421a4ce9:0x3fcbf560 0x421d2b8c:0x3fcbf580 0x421a4ce9:0x3fcbf5d0 0x421a4db5:0x3fcbf5f0 0x40379a92:0x3fcbf610 0x4219ce4b:0x3fcbf6b0 0x421a4ce9:0x3fcbf6e0 0x421d2b8c:0x3fcbf700 0x421a4ce9:0x3fcbf750 0x421a4db5:0x3fcbf770 0x421a38cc:0x3fcbf790 0x421a36f9:0x3fcbf7f0 0x421a4ce9:0x3fcbf810 0x421a4db5:0x3fcbf830 0x40379a92:0x3fcbf850 0x4219ce4b:0x3fcbf8f0 0x421a4ce9:0x3fcbf920 0x421a4cfe:0x3fcbf940 0x421b1e4b:0x3fcbf960 0x421b1fb2:0x3fcbf9f0 0x421b982a:0x3fcbfa40

ELF file SHA256: a09732e80

Rebooting...
���ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382ce6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x105c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbd8
load:0x403cc700,len:0x2e30
entry 0x403c989c
LVGL MicroPython 1.23.0 on 2024-10-15; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
``` 

Touch Product id: 911
Touch Firmware version: 0x1060
Touch Vendor id: 0x0
Touch resolution: width=800, height=480

```python
from i2c import I2C
import gt911
import lcd_bus
import task_handler
import lvgl as lv  # NOQA
import rgb_display
#import lv_utils
import time

_WIDTH = const(800)
_HEIGHT = const(480)

_BUFFER_SIZE = const(768000)  # width * height * 2 1107651508 768000

_CTP_SCL = const(20)
_CTP_SDA = const(19)
_CTP_IRQ = None

_SD_MOSI = None
_SD_SCK = None
_SD_MISO = None

_LCD_FREQ = const(15000000)
_PCLK_ACTIVE_NEG = const(1)#0

_HSYNC_PULSE_WIDTH = const(20)#10,4
_HSYNC_BACK_PORCH = const(46)#10,43
_HSYNC_FRONT_PORCH = const(210)#10,8

_VSYNC_PULSE_WIDTH = const(10)#10,4
_VSYNC_BACK_PORCH = const(23)#10,12
_VSYNC_FRONT_PORCH = const(22)#20,8

_PCLK = const(0)
_HSYNC = const(39)
_VSYNC = const(41)
_DE = const(40)
_DISP = None
_BCKL = const(2) # backlight
_DRST = None

I2C_BUS = I2C.Bus(
    host=1,
    scl=_CTP_SCL,
    sda=_CTP_SDA,
    freq=400000,
    use_locks=False
)

TOUCH_DEVICE = I2C.Device(
    I2C_BUS,
    dev_id=gt911.I2C_ADDR,
    reg_bits=gt911.BITS
)

_DATA15 = const(1)  # B7
_DATA14 = const(9)  # B6
_DATA13 = const(46)  # B5
_DATA12 = const(3)  # B4
_DATA11 = const(8)  # B3
_DATA10 = const(4)  # G7
_DATA9 = const(16)  # G6
_DATA8 = const(15)  # G5
_DATA7 = const(7)  # G4
_DATA6 = const(6)  # G3
_DATA5 = const(5)  # G2
_DATA4 = const(14)  # R7
_DATA3 = const(21)  # R6
_DATA2 = const(47)  # R5
_DATA1 = const(48)  # R4
_DATA0 = const(45)  # R3

bus = lcd_bus.RGBBus(
    hsync=_HSYNC,
    vsync=_VSYNC,
    de=_DE,
    pclk=_PCLK,
    data0=_DATA0,
    data1=_DATA1,
    data2=_DATA2,
    data3=_DATA3,
    data4=_DATA4,
    data5=_DATA5,
    data6=_DATA6,
    data7=_DATA7,
    data8=_DATA8,
    data9=_DATA9,
    data10=_DATA10,
    data11=_DATA11,
    data12=_DATA12,
    data13=_DATA13,
    data14=_DATA14,
    data15=_DATA15,
    freq=_LCD_FREQ,
    hsync_front_porch=_HSYNC_FRONT_PORCH,
    hsync_back_porch=_HSYNC_BACK_PORCH,
    hsync_pulse_width=_HSYNC_PULSE_WIDTH,
    hsync_idle_low=False,
    vsync_front_porch=_VSYNC_FRONT_PORCH,
    vsync_back_porch=_VSYNC_BACK_PORCH,
    vsync_pulse_width=_VSYNC_PULSE_WIDTH,
    vsync_idle_low=False,
    de_idle_high=False,
    pclk_idle_high=False,
    pclk_active_low=_PCLK_ACTIVE_NEG,
    disp=1,
)

buf1 = bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM)
buf2 = bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM)

display = rgb_display.RGBDisplay(
    data_bus=bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    frame_buffer1=buf1,
    frame_buffer2=buf2,
    backlight_pin=_BCKL,
    backlight_on_state=rgb_display.STATE_PWM,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=False
)

display.set_power(True)
display.init()
display.set_backlight(100)
indev = gt911.GT911(TOUCH_DEVICE)

# display.set_rotation(lv.DISPLAY_ROTATION._90)  # NOQA
th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)

```

kdschlosser, 2024-10-24:
are you using thonny? If you are don't. Thonny doesn't perform a hard reset of the ESP32 so hardware states don't get reset only the memory does. This will cause issues when restarting the ESP32.

kdschlosser, 2024-10-24:
If the touch is working incorrectly you might need to tell the touch driver to rotate. Sometimes the touchscreen is attached to the display so the orientation is incorrect. to change this you add the `startup_rotation` parameter when creating the touch driver...

```python
indev = gt911.GT911(TOUCH_DEVICE, startup_rotation=lv.DISPLAY_ROTATION._0)
```

`lv.DISPLAY_ROTATION._0`, `lv.DISPLAY_ROTATION._90`, `lv.DISPLAY_ROTATION._180` and `lv.DISPLAY_ROTATION._270` are the allowed values.

closed 2024-11-05 by kdschlosser