lvgl-micropython/lvgl_micropython #159: Typo in builder/__init__.py

mark-hoy, 2024-10-22:
Typo in builder/__init__.py 

'''
--- a/builder/__init__.py
+++ b/builder/__init__.py
@@ -332,10 +332,10 @@ def get_lvgl():
 def get_micropython():

     cmd_ = [
-        'git submodule updatem --init --depth=1 -- lib/micropython',
+        'git submodule update --init --depth=1 -- lib/micropython',
     ]
     print()
'''

mark-hoy, 2024-10-22:
This allows the build to go further (on MacOS) but there is something wrong in 
```
def copy_micropy_updates(port):
    src_path = f'micropy_updates/{port}'
    org_path = f'micropy_updates/originals/{port}'

    if port in ('raspberry_pi', 'macOS'):
        port = 'unix'

    dst_path = f'lib/micropython/ports/{port}'

```

The files for macOS should be coming from the unix directory (I think)
```
bash-5.2$ ls lvgl_micropython/micropy_updates/unix
machine_sdl.c   machine_sdl.h   machine_timer.c machine_timer.h
```

mark-hoy, 2024-10-22:
A check to see if there is an original file allows the build to get a lot further.

```
    def iter_files(s_path, d_path, o_path):
        for file in os.listdir(s_path):
            src_file = os.path.join(s_path, file)
            dst_file = os.path.join(d_path, file)
            org_file = os.path.join(o_path, file)

            if os.path.isdir(src_file):
                if not os.path.exists(org_file):
                    os.makedirs(org_file)

                iter_files(src_file, dst_file, org_file)
            else:
                # backup the original file, if it exits                                                             
                if os.path.isfile(dst_file):
                    print("copyfile ", dst_file, " orig", org_file)
                    shutil.copyfile(dst_file, org_file)

                print("copyfile ", src_file, " dst", dst_file)
                shutil.copyfile(src_file, dst_file)
                
```

kdschlosser, 2024-10-23:
Oh shit. I forgot to correct that typo.
Uhhh yes I had forgotten about adding new files to the build I will make these changes.

mark-hoy, 2024-10-28:
Thanks for fixing it.

closed 2024-10-28 by mark-hoy