lvgl-micropython/lvgl_micropython #160: Remote Viewer Feature Request

dcmcshan, 2024-10-24:
https://github.com/CubeCoders/LVGLRemoteServer

I think this is pretty cool and would be a good feature to have in lvgl-micropython.

Or, at least a basic screenshot capability.

picpic020960, 2024-10-24:
as https://github.com/harbaum/ftDuino-controller

kdschlosser, 2024-10-24:
It is a good idea except... A large portion of GUI development on a micro controller is going to be resource management, specifically speed/performance. Because of all of the different connection typologies that exist for displays the passing of the frame buffer data to a PC over say a network connection would need to be done so it would duplicate the speed in which the frame buffer data would transfer to a real attached display. I don't think this would be able to be done given how fast data is able to be sent to a display. If you have a 16 lane I80 display running at 80Mhz that's 160,000,000 bytes per second. This would not be possible over a WiFi connection nor would it be possible over a serial or CDC connection either. Serial the most you are going to get is 125,000 bytes a second, CDC tops out at 525,000 bytes a second (user reported) and the WiFi tops out at 3,750,000 bytes a second is using UDP. 

A single lane 80mhz SPI connection would be able to send the data faster than what is able to be achieved. (10,000,000 bytes a second). It would be painfully slow when rendering the display data to the PC. Small displays are less of an issue but once you get into the higher resolution RGB displays it would be awful to use.

picpic020960, 2024-10-28:
Lvgl send only little area where pixels change, so the bandwidth is small. Some time all the screen change but not always.

kdschlosser, 2024-10-28:
I have been looking into this and unfortunately the code has been written for the Arduino IDE which makes it unusable for anything written outside of the Arduino IDE

kdschlosser, 2024-10-28:
It also requires using an application written specifically for use with it and that application ONLY runs on Windows.

I have started to write a VNC server driver which will work with a VNC client making the client cross platform and something that is able to be used with a plethora of different clients that are made. It is also going to be able to work over any kind of connection that is made, whether is be over BlueTeeth, Serial, WiFi, TCP, UDP, Ethernet... doesn't matter what the type of connection is because the RFB protocol is a high level protocol and only has specification for the way the data is being formatted to go over the wire.

The only thing I don't know if whether or not clients are written to deal with partial frame buffers. it kind of sounds like it might be but then again it might not be and that is the big question as this is not mentioned in the driver specification. I don't know if the clients hold the frame buffer in memory and only update the sections of the frame buffer that need updating or does the entire frame buffer need to be sent each and every single time an update to the display is done. If partial data is not able to be sent I will have to write a client but that is not something that would be hard to do as I would be able to do it using LVGL and that would make it cross platform.

dcmcshan, 2025-03-25:
Any update on this?  Would be very cool if I could view display in browser...

asherevan, 2025-03-25:
A screenshot function would be very nice.

kdschlosser, 2025-03-25:
> A screenshot function would be very nice.

It's called `lv.snapshot` and it is already built into it.

asherevan, 2025-03-25:
Oh! Thank you!

kdschlosser, 2025-03-25:
> Any update on this? Would be very cool if I could view display in browser...

I would have to write a VNC driver from the ground up because everything I find wants to manage the network connection and VNC is a high level protocol so it should be able to work over any kind of a connection. doesn't matter if it's serial, BT, WiFi, etc.... It's a lot of work to do that. I don't really want to write something custom and then have to write a cross platform viewer for it.

bhupiister, 2025-05-08:
Hi. I was looking for the possibility to have GUI over VNC. Were you able to finish up some work?