lvgl-micropython/lvgl_micropython #166: how to properly initialize spi screen?

Bigmaxmafia, 2024-10-30:
esp32 s3

A lot has changed, I was using a very old build. I compiled a fresh one and took the working code from the discussions, but I get an error

Traceback (most recent call last):
  File "boot.py", line 27, in <module>
  File "display_driver_framework.py", line 215, in __init__
  File "display_driver_framework.py", line 280, in _init_bus
SyntaxError: Cannot convert 'bound_method' to pointer!
LVGL MicroPython 1.23.0 on 2024-10-29; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
```
import lvgl as lv
import lcd_bus

import ili9341
from machine import SPI

# DISPLAY SETUP

spi_bus = SPI.Bus(host=1, mosi=14, miso=41, sck=13)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=4,
    cs=6,
    freq=40000000,
)

display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=240,
    display_height=320,
    reset_pin=5,
    # comment the next line if you still don't get something on the display
    reset_state=ili9341.STATE_LOW,
    backlight_pin=40,  # NOT USED...YET
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=ili9341.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

display.set_power(True)
display.init()

# setting this to anything but 100 or 0 when you don't have the
# backlight state set to PWM isn't going to do anything
# display.set_backlight(100)

# TOUCH SETUP

print("Creating touch bus")
# T_IRQ = 36   48
# T_DO = 35   47
# T_DIN = 1
# T_CS = 2
# T_CLK = 42
touch_bus = SPI.Bus(host=2, mosi=1, miso=47, sck=42)

print("creating touch SPI Device")
touch_device = SPI.Device(
    spi_bus=touch_bus,
    freq=1000000,
    cs=2,
    polarity=0,
    phase=0,
    bits=8,
)
print("importing xpt2045")
import xpt2046

print("Instantiating touch Indev")
touch_indev = xpt2046.XPT2046(
    touch_device, touch_cal=None, startup_rotation=lv.DISPLAY_ROTATION._90, debug=False
)

# TOUCH CALIBRATE

print("Calibration")
if not touch_indev.is_calibrated:
    touch_indev.calibrate()

display.invert_colors()

# you want to rotate the display after the calibration has been done in order
# to keep the corners oriented properly.
display.set_rotation(lv.DISPLAY_ROTATION._90)

# SCREEN SETUP

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0xFFB6C1), 0)

slider = lv.slider(scrn)
slider.set_size(280, 20)
slider.center()

label = lv.label(scrn)
label.set_text("HELLO WORLD!")
label.align(lv.ALIGN.CENTER, 0, -50)

import task_handler

th = task_handler.TaskHandler()
```

Bigmaxmafia, 2024-10-30:
I updated to 0ee7f2c, but the error remained. I deleted the line with the error, now it gives a different error
File "boot.py", line 9, in <module>
TypeError: can't convert module to int

kdschlosser, 2024-10-30:
don't update. You need to clone the repo again. That is the only way the fix is going to get populated. The issue was with the LVGL submodule pointing to the wrong commit in LVGL. This only gets corrected by cloning the repo again.

Bigmaxmafia, 2024-10-31:
> don't update. You need to clone the repo again. That is the only way the fix is going to get populated. The issue was with the LVGL submodule pointing to the wrong commit in LVGL. This only gets corrected by cloning the repo again.

The image started to work, but when I try to calibrate the touchscreen, the screen just turns white and nothing happens
console
```
Creating touch bus
creating touch SPI Device
importing xpt2046
Instantiating touch Indev
```

Bigmaxmafia, 2024-11-03:
@kdschlosser Thanks for the help with the initialization, but it looks like I need a little more help. After touch_indev.calibrate() is used nothing happens on the screen, the screen itself is white with very thin stripes. If I remove the calibration, the touchscreen and screen work, but I still need calibration, any ideas? I noticed that when stopping the program via ctrl-c, the output always occurs on line 70 pointer_framework.py

kdschlosser, 2024-11-03:
what is the exact code you are using?

kdschlosser, 2024-11-03:
OH I just noticed something...

you need to have these lines of code..

```python
import task_handler

th = task_handler.TaskHandler()
```

placed before the calibration code but after the display has been initialized.

Bigmaxmafia, 2024-11-04:
> OH I just noticed something...
> 
> you need to have these lines of code..
> 
> ```python
> import task_handler
> 
> th = task_handler.TaskHandler()
> ```
> 
> placed before the calibration code but after the display has been initialized.

@kdschlosser This helped, now the calibration is enabled, but a new problem has been added. When we click on the second calibration point we have it in the console
```
Calibration
1: Tpcal_point(touch_x=31, touch_y=201, screen_x=24, screen_y=24)
2: Tpcal_point(touch_x=31, touch_y=201, screen_x=24, screen_y=296)
3: Tpcal_point(touch_x=158, touch_y=198, screen_x=216, screen_y=24)
Traceback (most recent call last):
  File "task_handler.py", line 132, in _task_handler
  File "touch_calibrate.py", line 498, in on_press
ZeroDivisionError: divide by zero
```

Bigmaxmafia, 2024-11-04:
I noticed that I held the first point for a long time. When you quickly click on the first point, the second point is actually not on the screen, and on the side you can see a scroll bar of the screen.

kdschlosser, 2024-11-04:
I made some updates to the touch driver. If you want to test the changes and see if they work properly you can do that by cloning the repo using this command...

```
git clone -b touch-calibration https://github.com/lvgl-micropython/lvgl_micropython
```

Bigmaxmafia, 2024-11-04:
> I made some updates to the touch driver. If you want to test the changes and see if they work properly you can do that by cloning the repo using this command...
> 
> ```
> git clone -b touch-calibration https://github.com/lvgl-micropython/lvgl_micropython
> ```

@kdschlosser with calibration branch this

```
Traceback (most recent call last):
  File "boot.py", line 80, in <module>
  File "pointer_framework.py", line 64, in calibrate
TypeError: function takes 4 positional arguments but 3 were given
```

kdschlosser, 2024-11-05:
ok give it a try again. I fixed that error.

Bigmaxmafia, 2024-11-05:
> ok give it a try again. I fixed that error.

@kdschlosser Now a new calibration interface is opening, but now the touchscreen does not respond to my touches. There are no errors. Here is the key interrupt log, I looked at it several times, we come out on the same line.

```
Calibration
Traceback (most recent call last):
  File "task_handler.py", line 132, in _task_handler
  File "touch_calibrate.py", line 438, in on_touch
KeyboardInterrupt:
```

kdschlosser, 2024-11-05:
OK, I am not going to be able to check it out until later on today. I am tied up with having to deal with my vehicle that got damaged when it was at the dealership for service. Have to get photos and videos sent over to them with descriptions of what is going on. It's going to take me a while to finish doing that.

Bigmaxmafia, 2024-11-10:
@kdschlosser What's up?

kdschlosser, 2024-11-10:
Oh man, we ended up getting over 3' of snow and I have been shoveling that crap from my 125' long driveway for the past few days, snowblower is inop. I have been spent and my back has been sore (back surgery earlier this year) so I haven't spent a whole heck of a lot of time at the keyboard.

I promise I will look at it this evening and get it all sorted out.

Bigmaxmafia, 2024-11-10:
> Oh man, we ended up getting over 3' of snow and I have been shoveling that crap from my 125' long driveway for the past few days, snowblower is inop. I have been spent and my back has been sore [...]

No problem! Just wanted to make sure I didn't miss anything.

kdschlosser, 2024-11-10:
no you didn't miss anything at all. Life got in the way.. got stuff that I have to take care of at the homestead ya know?

kdschlosser, 2024-12-17:
The touch calibration is now 100% working correctly. I am going to close this issue now.