lvgl-micropython/lvgl_micropython #168: add a build option to format vfs as littlefs rather than fat.

dcmcshan, 2024-10-31:
I am having a difficult time working with FAT, and I know that littlefs works as expected.

dcmcshan, 2024-10-31:
Incidentally, I think the issue here is that FAT fs are case insensitive, but micropython's implementation is case-sensitive...

kdschlosser, 2024-10-31:
I have never messed with the partition types so you will have to provide me examples of what you were doing and how you were doing it. Then I can tie it into this code to make it work.

kdschlosser, 2024-10-31:
check the LVGL forum, I sent you a message.

wfclhuhu, 2025-01-11:
Could you please help me? I found that there are no implementation methods for VfsSpiFFS and VfsLttleFS in the os class. What functions should be used to use the file system?