lvgl-micropython/lvgl_micropython #170: esp32-s3-wroom N4R8 don't boot

favnec5, 2024-11-02:
Hello,
I have a fresh S3 N4R8 (JC4827W543), i compile a lot of time, But it not boot :-/
Nothing after burning. 
I've tried many versions:
- python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 --flash-size=4
- python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM --flash-size=4
- python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=4
- python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=FLASH_4M

Worst of all, the straga's firmware boot (https://github.com/lvgl-micropython/lvgl_micropython/discussions/161, and https://github.com/straga/micropython_lcd/tree/master)
With this 16Mb version, i see the micropython start message, with flash not innitialised error.

I don't understand what I'm missing.
could you help me ?

regards,
Thomas

picpic020960, 2024-11-02:
Bonjour ,
Same process , same issue.
Used https://github.com/orgs/micropython/discussions/14333 and the display works (qspi) but no lvgl in thé firmware.

favnec5, 2024-11-02:
hello,
thanks. my problem is not (at this moment) from the display.
The board don't start. If this is 4MB octal , it would be work with :
python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=4
But generated firmware not boot. 
t'm tring  python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=4 CONFIG_ESPTOOLPY_FLASHSIZE_DETECT=y CONFIG_COMPILER_OPTIMIZATION_SIZE=y CONFIG_ESPTOOLPY_OCT_FLASH=y CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG=y CONFIG_BOOTLOADER_LOG_LEVEL=4 CONFIG_COMPILER_OPTIMIZATION_SIZE=y
(...)

kdschlosser, 2024-11-03:
I need to know who makes your board and the model number of the board.

picpic020960, 2024-11-03:
[image]
From aliexpress

favnec5, 2024-11-03:
lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-4.bin created this morning. Same result. the board is staying in boot mode.

kdschlosser, 2024-11-05:
OK so you have the following ESP32. 

ESP32-S3-WROOM-1-N4R8 4 MB (Quad SPI) 8 MB (Octal SPI)

The Board is wired to use the USB-CDC/OTG pins on the ESP32 and not the typical UART pins. So you need to add an additional command to enable that. 

you should be using the following as the build command

```
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=4 --usb-otg

```

I have not added display drivers for the display this board uses. I have been working on them with another user and once we get everything all hammered out then I will add them. The touch is also going to be a special driver as will unless it is a resistive touch panel. In that case it will be the xpt2046 driver.

picpic020960, 2024-11-05:
Bonjour Kevin,
With the micropython above , the gt911 works but maybe lvgl has this own rules.

kdschlosser, 2024-11-05:
??? ya lost me on that...

favnec5, 2024-11-05:
"python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=4 --usb-otg"
Ok thanks Kevin, i go to try that and back to you. Sorry compiling takes a long time on my PC, I won't know until this evening, or this morning for you ;-)

@picpic020960
If i find the good cooking to the display and touchpanel, i'm back to you.
I thing solution is here https://github.com/straga/micropython_lcd/tree/master/device/JC3248W535/driver/axs15231b

picpic020960, 2024-11-05:
Kevin
```
MPY: soft reboot                                         
LVGL MicroPython 1.23.0 on 2024-11-05; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3                      Type "help()" for more information.                      
>>>
```

OK with --usb-otg and no display and no indev

kdschlosser, 2024-11-05:
it only takes me about 30 seconds to compile it so if you want me to compile and post the binaries for ya I can do that.

picpic020960, 2024-11-05:
Kevin ,
One driver gt911.py named works with the board. So Can I build with 'indev=gt911' ?
But what for the 'display=...' ?
(OK for pre-build binaires)

kdschlosser, 2024-11-05:
the display I don't have a driver for just yet. It is being worked on. and if the GT911 driver works then yes you can use it.

Give me a day to get a rough draft of the display driver together.

favnec5, 2024-11-05:
Cool, just to confirm : it works on my board, of course. I ´m trying to write ic driver so

favnec5, 2024-11-09:
Sorry i'm little busy on an other board, but don't forget to do this.
I'm pretty sure I can make a driver for this board because I managed to run it without LVGL.

picpic020960, 2024-11-10:
Bonjour ,
First test with adapted https://github.com/straga/micropython_lcd , from discussion #161

Some pixels lose and Bad colors. No touch configured.

picpic020960, 2024-11-10:
[image]

picpic020960, 2024-11-10:
With Gt911, no closed pixels
Always bad colors

https://github.com/user-attachments/assets/a44edf1e-01a2-4187-b6b8-eab814f9a88a

Finger to right, slider to left
One trick?

kdschlosser, 2024-11-10:
if you are moving your finger right and the slider is moving to the left then you need to pass a rotation of 180 to the indev driver when initializing it. 

```python
gt911.GT911(i2c_device, start, startup_rotation=lv.DISPLAY_ROTATION._180)
```

The bad colors is going to be the pin definitions that you are using and them not being correct. It's pretty easy to figure out what the pin definitions should be. set the background to be `lv.color_hex(0x000000)` and then tell me what the color is that you are seeing. I can hammer out the pin definitions from that.

kdschlosser, 2024-11-10:
I also am not sure what you mean by this... 

> no closed pixels

I know that you had reported having the pixel anomalies on the one side of the display. are those all gone now?

picpic020960, 2024-11-11:
Bonjour Kevin,

OK for touch.

For color , background color in the script : bottom left --> yellow , bottom right --> black , top left --> blue , top rights -->  ?  : as in the screenshot above.
So what pins ?

other question : 
scr_act  ==> screen_active
get_disp ==>  ?????
(In  "lv.screen_active().get_disp().driver.ver_res")
Where found the translation between lvgl V8.1 to V8.3 or V9 ?

For pixel , the first display , pixel lose ; then when slider ... No pixel lose .

Thank's for help

kdschlosser, 2024-11-11:
what do you mean by "pixel lose"

> For color , background color in the script : bottom left --> yellow , bottom right --> black , top left --> blue , top rights --> ? : as in the screenshot above.

OK you said the colors are not correct. are the colors you are seeing in that screen shot the colors you are supposed to be seeing? If they are then the pins are correct and the colors are correct. If they are not the colors you have set them to be then what colors are they supposed to be?

After you compile your firmware for eh ESP32 there is a stub file made for the entire API of LVGL. That stub file gets created in the root of the clone of this repository. the filename is lvgl.pyi. If you are using an IDE when writing your code then it should be able to read a stub file. By placing that file into the root of you project the IDE will be able to read it and it will know all of the functions, classes, methods, variables, etc... that are located in the lvgl module that is flashed to your ESP32.

If you want to know the specific API changes from 8.3.x to 9.x.x then open the `lib/lvgl/src/lv_api_map_v8.h` file and you can see the "translations" between the 2 versions. This is not an all inclusive list of changes as some of the parameters for functions may have changed it just shows you the change in function names and structure names. That's all.

kdschlosser, 2024-11-11:
ok so with the colors. Your colors are flipped. meaning that FF0000 for red is getting interpreted by the display as 00FFFF instead. Not sure what would cause that. maybe a setting in the display init is going it.

kdschlosser, 2024-11-11:
I know what it is. set your color inversion to on.

picpic020960, 2024-11-11:
Kevin ,
OK for invert_colors()
The colors are right but not flash.
In https://github.com/echo-lalia/beach-portal/blob/main/display/tft_config.py 
values send to the display , without know why , but After the colors are better , author saids.
Is that in datasheet ?

OK for translation lvgl v8 ==> V9
But how to translate lvgl ***C*** to lvgl ***micropython*** versus v8 / V9

Example : 
lv_scr_act ==> lv.screen_active
lv_get_disp ==> ?????
Many users facing this issue

kdschlosser, 2024-11-11:
lv.get_display

Open the lvgl.pyi file it will show you the entire LVGL API

favnec5, 2024-11-11:
I'm back to the game ;-)

[nv3041a.zip](https://github.com/user-attachments/files/17706922/nv3041a.zip)
[image]

Noise was to big sqpi bus freq for me

kdschlosser, 2024-11-11:
post the code and submit a PR for the driver. Would be nice to add it so other will be able to use it.

kdschlosser, 2024-11-11:
adding drivers is not all that hard to do is it? I tried to make it a pretty easy thing to do. It offers the ability to really extend the driver framework to alter how it works and it also handles the really simple additions that require only init commands to be sent.

kdschlosser, 2024-11-11:
ok so I have some questions regarding the driver. I just saw the zip file above the image.

how come you are sending the buffer 10 bytes at a time??

picpic020960, 2024-11-11:
OK Kevin , fine.
Now I test keyboard widget with password (from lvgl Doc). Diff between C lvgl V8 and V9 , seems me null.
So the micropython example must go on.
But when <CR> , error blob object has no attribut 'get_text'
Not easy...
No success with __dereference__
One idea ?

favnec5, 2024-11-11:
This is a good question.  Because the only example of qspi driver with your repo is https://github.com/lvgl-micropython/lvgl_micropython/discussions/161 and it's what it did.
I haven't tested it yet without or with different settings.

favnec5, 2024-11-11:
[image]

The hardware rotation is magic on Qspi screen

kdschlosser, 2024-11-11:
> The hardware rotation is magic on Qspi screen

not sure what you mean by this,,,

> Now I test keyboard widget with password (from lvgl Doc). Diff between C lvgl V8 and V9 , seems me null.
> So the micropython example must go on.
> But when 'enter' , error blob object has no attribut 'get_text'
> Not easy...
> No success with 'dereference'

Need to see the code you are using to be able to help.

favnec5, 2024-11-11:
Sorry this is for an other discussion here : https://github.com/lvgl-micropython/lvgl_micropython/discussions/176
This (old) board is Qspi with hardware rotation, it works perfectly like you said, whereas for my board with rgbBus it's a different story...
I've pushed the NV3041A driver into the directory. 
I'll come back if I find a solution for rotate the rgb display without killing perfomances

picpic020960, 2024-11-12:
Thorry 
(Test keyboard widget with password)
```
def ta_event_cb(e):
    code = e.get_code()
    ta = e.get_target()
    if code == lv.EVENT.CLICKED or code == lv.EVENT.FOCUSED:
        # Focus on the clicked text area
        if kb != None:
            kb.set_textarea(ta)

    elif code == lv.EVENT.READY:
        print("Ready, current text: " + ta.get_text())

# Create the password box
LV_HOR_RES = lv.scr_act().get_disp().driver.hor_res
LV_VER_RES = lv.scr_act().get_disp().driver.ver_res

pwd_ta = lv.textarea(lv.scr_act())
pwd_ta.set_text("")
pwd_ta.set_password_mode(True)
pwd_ta.set_one_line(True)
pwd_ta.set_width(LV_HOR_RES // 2 - 20)
pwd_ta.set_pos(5, 20)
pwd_ta.add_event_cb(ta_event_cb, lv.EVENT.ALL, None)

# Create a label and position it above the text box
pwd_label = lv.label(lv.scr_act())
pwd_label.set_text("Password:")
pwd_label.align_to(pwd_ta, lv.ALIGN.OUT_TOP_LEFT, 0, 0)

# Create the one-line mode text area
text_ta = lv.textarea(lv.scr_act())
text_ta.set_width(LV_HOR_RES // 2 - 20)
text_ta.set_one_line(True)
text_ta.add_event_cb(ta_event_cb, lv.EVENT.ALL, None)
text_ta.set_password_mode(False)

text_ta.align(lv.ALIGN.TOP_RIGHT, -5, 20)

# Create a label and position it above the text box
oneline_label = lv.label(lv.scr_act())
oneline_label.set_text("Text:")
oneline_label.align_to(text_ta, lv.ALIGN.OUT_TOP_LEFT, 0, 0)

# Create a keyboard
kb = lv.keyboard(lv.scr_act())
kb.set_size(LV_HOR_RES, LV_VER_RES // 2)

kb.set_textarea(pwd_ta)  # Focus it on one of the text areas to start

```

picpic020960, 2024-11-12:
Bus error

favnec5, 2024-11-12:
> There is a couple more things I have to change in the driver to make sure it is going to run properly. I am going to do that now.

No problem, it's a good thing
thanks

closed 2024-11-21 by favnec5