lvgl-micropython/lvgl_micropython #171: A warning during ili9341 initialization

Lenergy1982-droid, 2024-11-03:
Hello!
I'm trying to get working this combination of hardware: esp32-S3-Devkit-1N16R8 with PSRAM + ili9341 display + xpt2046 touchscreen.
My build string: python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash_size 16 DISPLAY=ili9341 INDEV=xpt2046

Using code, published here: https://github.com/fabse-hack/temp_humidity_micropython_lvgl?tab=readme-ov-file#code
and during start this error occurs (while the display shoes only white screen):

Traceback (most recent call last):
  File "main.py", line 34, in <module>
  File "display_driver_framework.py", line 487, in init
ImportError: no module named '_ili9341_init'

Exact code:
```
from machine import SPI, Pin
import sys
import lcd_bus
import ili9341
import lvgl as lv

lv.init()

spi_bus = SPI.Bus(host=2, mosi=11, miso=13, sck=12)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=4,
    cs=10,
    freq=80000000,
)
	
display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=240,
    display_height=320,
    reset_pin=5,
    # comment the next line if you still don't get something on the display
    # reset_state=ili9341.STATE_LOW,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=ili9341.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

pin3 = Pin(3, Pin.OUT, value=1) # A pin for managing the power to 
display.set_power(True)
display.init()

display.set_rotation(lv.DISPLAY_ROTATION._90)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0xFFB6C1), 0)

label = lv.label(scrn)
label.set_text("HELLO WORLD!")
label.align(lv.ALIGN.CENTER, 0, -50)

```
I tried to download a previous version of ili9341.py and display_driver_framework.py (sorry, can't say what exact version) and put them into the project dir on the flash, but no luck - the error message doesn't appear, and the screen is still blank.

I also changed host number to 1 and changed the pins to: 

spi_bus = SPI.Bus(host=1, mosi=14, miso=41, sck=13)
```
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=4,
    cs=6,
    freq=40000000,
)
```
, but it didn't help. 

I'm asking for assistance from those, who managed to get this setup work properly, cause there are no more alternatives in the code to try - I've tried to change almost all parameters.

Big thanks in advance!!

picpic020960, 2024-11-03:
Maybe display.init(1)?

Lenergy1982-droid, 2024-11-03:
Thanks!! Strange how could I miss one of the previous issues, where kdschlosser describes it.. But it was late night, actually morning)
However ili9341 still doesn't work) Whether I use host 1or 2, display.init(1) or display.init(2), comment or uncomment reset_state=ili9341.STATE_LOW, it is still white or gray. 
Maybe the problem is the pins chosen? I took them from here: http://wiki.fluidnc.com/en/hardware/ESP32-S3_Pin_Reference
This is the extract:
---
The SPI2 (VSPI) default pins are:
gpio.12 SCK Defined as SPI0_SCK
gpio.11 MOSI Defined as SPI0_MOSI
gpio.13 MISO Defined as SPI0_MISO
gpio.10 CSO Defined as SPI_CS0
---

kdschlosser, 2024-11-03:
it could be the pins being used are not in alignment with the host being used. You should be using host=1

Lenergy1982-droid, 2024-11-03:
Yeah, with ili9341 I will change pins and add the task handler, I think everything is going to be okay.
Thanks one more time!!

kdschlosser, 2024-11-03:
no worries m8.  easy to diagnose  problem with a super simple fix.

closed 2024-11-05 by kdschlosser