lvgl-micropython/lvgl_micropython #175: build option to easily add fonts

dcmcshan, 2024-11-09:
I have some fonts (as .font files) that I'd like to bake into mp kernel.  I'd like to be able to use your build script to point to a directory where they live and have it include them.  Loading dynamically is not fast.

kdschlosser, 2024-11-09:
You cannot bake anything into the firmware that isn't C code. When baking python source files they are actually compiled beforehand.

What I recommend doing is writing a python script that will read your font directory and have it read each file as binary data and then use `binascii.hexlify` to convert it into a hex string of sorts. It does it using only the last 2 characters of the hex string for each byte so `0x00` becomes `"00"` It cuts down on the number of characters stored in the source file. That same script then writes that as a variable called `font` into a python source file. One font per file. The filenames are then written to a manifest.py file so you can pass that filename to the build script using the `FROZEN_MANIFEST` parameter.

kdschlosser, 2024-11-09:
What I would actually do is write a dynamic loading routine. something along the lines of...

```
import sys

def get_font(font_name):
    mod = __import__(font_name)
    del sys.modules[font_name']  # frees up the memory used by the module
    return mod.font
```

The downside of doing this is it is going to load all of the font data into memory. I have not done much looking into the fonts and how that all works but I would imagine that there is some kind of a header that is an index of sorts that contains the start byte number of a character so one character is able to be loaded into memory instead of the whole character set. I could be wrong in that, I would have to check to see how it works.

kdschlosser, 2024-11-09:
Here is a script for ya that will convert a font file that is generated using the [LVGL Font Converter](https://lvgl.io/tools/fontconverter)

This will only read a font that has been converted into C code. so make sure you check that option off when doing the conversion.

<details>
  <summary>CODE HERE</summary>

```python
import os
import sys

if len(sys.argv) == 1:
    raise RuntimeError('you must provide the path to the generated font file')
    
font_path = sys.argv[1] 

if not font_path.endswith('.c'):
    raise RuntimeError('the generated font file must be in C code')

with open(font_path, 'rb') as f:
    data = f.read().decode('utf-8')

characters = []
character = None
byte_count = 0

glyph_dsc = False
cmaps = False
brace_count = 0
cmap_params = ''

font_dsc = False
font_dsc_params = ''

font = False
font_params = ''
font_name = ''

blank_line_count = 0

print('import lvgl as lv')
print()

for line in data.split('\n'):
    line = line.strip()
    if not line:
        blank_line_count += 1
        if blank_line_count < 2:
            print()

        if character is not None:
            characters.append(character)
            character['data'] = eval(f'bytearray([{character["data"]}])')
            character['data_len'] = len(character['data'])
            byte_count += character['data_len']
            character = None
        continue

    if 'uint8_t glyph_bitmap' in line:
        blank_line_count = 0
        print('glyph_bitmap = bytearray([')

    if line.startswith('/* U+'):
        character = dict(
            data='',
            name=line.split('"', 1)[-1].rsplit('"', 1)[0],
            u_code=line[2:].split('"', 1)[0].strip(),
            data_len=0,
            index=byte_count,
        )
[... 110 lines cut ...]
    if line.startswith('const lv_font_t '):
        blank_line_count = 0
        line = line.replace('const lv_font_t ', '')
        line = line.split(' =', 1)[0].strip()
        font_name = line
        print(line + ' = lv.font_t(dict(')
        font = True
    elif font:
        line = line.replace('lv_font_', 'lv.font_')
        if line.startswith('};'):
            blank_line_count = 0
            font = False
            print('))')
            print()

            print(f'{font_name}.get_glyph_dsc = {font_name}.get_glyph_dsc_fmt_txt')
            print(f'{font_name}.get_glyph_bitmap = {font_name}.get_bitmap_fmt_txt')

        elif line.startswith('.') and not line.startswith('.subpx') and 'NULL' not in line:
            if not line.startswith('.get_glyph_dsc') and not line.startswith('.get_glyph_bitmap'):
                blank_line_count = 0
                line = line[1:].replace('&', '').replace(' ', '')
                print('    ' + line)
```
</details>

kdschlosser, 2024-11-16:
Hey I wanted to let you know the macOS port has been tested and it's fully functional. The timer problem is fixed. There is one change in the build tho. You need to install SDL2 using homebrew prior to compiling

```
brew install SDL2
```

I don't know if the person that tested it was using Apple silicon. I do know it does compile properly for it. Just don't know for sure if it runs or not. I will ask the person that tested it if they are using an Intel processor.

closed 2024-12-17 by kdschlosser