lvgl-micropython/lvgl_micropython #185: will display.invert_colors() support param or not?

xiaohuboer, 2024-11-15:
This is a great project !

I met color invert issue when using st7789. I changed "display.invert_colors() " to "display.set_params(0x20)" to fix this issue.

but could you pls confirm the content in the picture?
[image]

xiaohuboer, 2024-11-15:
btw, the `rgb565_byte_swap` param worked not as expected. 
In below code, when changed `lv.color_make(0, 0, 255)` as (255, 0, 0)->(0, 255, 0)->(0, 0, 255)， the bg of lcd was R->G->B;
But if I set `rgb565_byte_swap` to False, and used same sequence like (255, 0, 0)->(0, 255, 0)->(0, 0, 255), the bg of lcd was B->R->G (I expected B->G->R)

```_BUFFER_SIZE = const(30720)
fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

display = st7789.ST7789(
    data_bus=display_bus,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    display_width=240,
    display_height=320,
    backlight_pin=14,
    color_byte_order=st7789.BYTE_ORDER_BGR,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display.init()
display.set_params(0x20)
display.set_rotation(lv.DISPLAY_ROTATION._90)
display.set_backlight(100)

th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_make(0, 0, 255), 0)
```

kdschlosser, 2024-11-15:
invert colors doesn't change anything that LVGL does it only flip flops things in the displays GRAM.

so if you have a color of 255, 0, 0 it becomes 0, 255, 255. or if you have a color of 127, 146, 89 it becomes 128, 106, 166

255, 0, 0 in binary is
```
11111111 00000000 00000000
```

inverted it is 
```
00000000 11111111 11111111
```

127, 146, 89 in binary is 
```
01111111 10010010 01011001
```

and when it is inverted it becomes
```
10000000 01101101 10100110
```

it flips the bits is what it does. 

`rgb565_byte_swap` changes the order of the bytes at the pixel level.

an RGB value of 0, 255, 0 looks like this in binary
```
00000000 11111111 00000000
```

and in RGB565 it look like this
```
00000111 11100000
```
The lowest 3 bits of red and blue get truncated and the lowest 2 bits of blue get truncated.

so when you have the color 15, 63, 127 in RGB565 that color is actually 8, 60, 120
binary for the RGB888 
```
00001111 00111111 01111111
```

binary for the RGB565
```
00001000 00111100 01111000
```

so if we have 0, 255, 0 as the color when using the RGB565 color format and setting the buye swap the pits change from 
```
00000111 11100000
```

to 

```
11100000 00000111
```

This is simply how the data gets sent to the display.  It has NOTHING to do with LVGL at all.

kdschlosser, 2024-11-15:
The color byte order is the order of the RGB888 bytes. This DOES effect how LVGL handles the colors.

so instead of LVGL writing the bytes as RGB RGB RGB to the frame buffer it writes them as BGR BGR BGR is you change it to BGR.

So if you set the byte order to BGR and the color format to RGB565 when set a color in LVGL to 255, 0, 0 it gets written to the frame buffer as
```
00000000 00011111
```

where as if it left as RGB it gets written to the frame buffer as 

```
11111000 00000000
```

It is fairly easy to determine what needs to be done.

If you set the background color to 0, 255, 0 and leave the byte order set to RGB and use RGB565 and rgb565 set to False. Paste a photo of what the screen is displaying and I can tell you want needs to be done to get the colors correct.  You have to look at things in binary to know what to do.

xiaohuboer, 2024-11-16:
Thanks for reply @kdschlosser . 

I apologize that my issue came from:
1. my wrong understanding on ```rgb565_byte_swap```. 
2. I have checked "api_drivers/**lvgl_api_drivers**/frozen/display/display_driver_framework.py" for the implementation of invert_colors; but it should be "api_drivers/**py_api_drivers**/frozen/display/display_driver_framework.py"

Now, it works well. 
The only thing to note is that ```display.invert_colors()``` need be called twice in my case  （I added comment in code for reason）

Below is my code and result (In case others encounter the same question for st7789).
btw, I generated my firmware through:
```bath
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16 DISPLAY=st7796 DISPLAY=st7789 INDEV=xpt2046 INDEV=cst816s LV_CFLAGS="-DLV_COLOR_DEPTH=16"
```

source code:
```python
import lcd_bus
from machine import SPI, Pin
import st7789
import lvgl as lv
import task_handler  # NOQA

spi_bus = SPI.Bus(
    host=2,
    mosi=47,
    miso=48,
    sck=21
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=41,
    cs=42,
    freq=20000000,
)

_BUFFER_SIZE = const(30720)
fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

display = st7789.ST7789(
    data_bus=display_bus,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    display_width=240,
    display_height=320,
    backlight_pin=14,
    color_byte_order=st7789.BYTE_ORDER_BGR,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display.init()

display.set_rotation(lv.DISPLAY_ROTATION._90)
display.set_backlight(100)

# self.set_params(_INVON) is called in _st7789_init.py and the 1st call of invert_colort will do self.set_params(self._INVON)
# so have to call it twice here
display.invert_colors()
display.invert_colors()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0) 

btn1 = lv.button(scrn)
btn1.set_size(80, 50)
btn1.align(lv.ALIGN.CENTER, -100, 0)
btn1.set_style_bg_color(lv.color_hex(0xFF0000), 0)
label1 = lv.label(btn1)
label1.set_text("Red")

btn2 = lv.button(scrn)
btn2.set_size(80, 50)
btn2.align(lv.ALIGN.CENTER, 0, 0)
btn2.set_style_bg_color(lv.color_hex(0x00FF00), 0)
label2 = lv.label(btn2)
label2.set_text("Green")

btn3 = lv.button(scrn)
btn3.set_size(80, 50)
btn3.align(lv.ALIGN.CENTER, 100, 0)
btn3.set_style_bg_color(lv.color_hex(0x0000FF), 0) 
label3 = lv.label(btn3)
label3.set_text("Blue")

th = task_handler.TaskHandler()
```

result screen(it is correct):
[image]

closed 2024-11-16 by xiaohuboer

kdschlosser, 2024-11-16:
you don't need to have this in the build command

`LV_CFLAGS="-DLV_COLOR_DEPTH=16"`  

It doesn't do anything that effects the operation of LVGL. The number of bits used for the color is controlled by you setting the RGB565 when you create the driver.

kdschlosser, 2024-11-16:
I just pushed a commit that changes the color inversion so it is a function you pass a boolean value to. 

```
driver.set_color_inversion(True/False)
```