lvgl-micropython/lvgl_micropython #186: Invalidate area is not allowed during rendering.

dcmcshan, 2024-11-17:
I get this while setting up some objects.  I am running task_handler in a thread.

` 79.31 [LVGL   ]                       [Error]	(4.400, +100)	 _lv_inv_area: Asserted at expression: !disp->rendering_in_progress (Invalidate area is not allowed during rendering.) lv_refr.c:257`

hI am protecting with thread locks, but apparently that's insufficient.  Wondering if there is some method I can query to know if  lvgl is rendering.  At least for debugging, it would be helpful.  And, it would be helpful to not fail hard as this currently does.

dcmcshan, 2024-11-17:
What's causing it is calling lv methods in another thread.  I know that lvgl is not thread safe.  Was just wondering if there is some internal lvgl way to know what it's doing (like whether it's calling task_handler).  I think I have sorted it out with thread locks, but that's an external thing.

kdschlosser, 2024-11-17:
I did provide you with a thread worker example yes?

dcmcshan, 2024-11-17:
Yes.  I'm using a rate monotonic task scheduler that is similar.  The issue is that I'm setting up the UI and then adding the update task to the scheduler.  The setup code happens while the display task is running.  An easy fix is just to stop the update while we setup, and/or thread locks.  Was just thinking for debug of all this it would be cooo if there was a simple LVGL Boolean set while it's updating

kdschlosser, 2024-11-18:
no easy way to do that.

If you use a a thread handler it should be taking care of calling lv_task_handler at the same time as setting up the UI. It's actually really easy to do because it takes care of calling the functions inside the single thread. since you are doing everything from the same thread at that point all that is needed are 2 locks. one to lock the task handler when it's not in use and you unlock the lock when something needs to get done. the second lock is for adding and removing the items that have been added to the queue to be run.

kdschlosser, 2024-11-18:
You can have all of the LVGL related code handled by one thread and then everything else by the other.

dcmcshan, 2024-11-18:
In concept, I'm hoping to have tiles that dynamically load.  It's not necessarily a requirement to create/destroy them, but I thought it would be better use of memory than to create them all up front.

kdschlosser, 2024-11-18:
if you are making tiles then the best bet is to load the images into memory and then change the images for what is being displayed. It's not that hard to do actually. I do not recall offhand if I saved the code or not. You can check in this repo and see what I have in there...

https://github.com/kdschlosser/mp_lvgl_widgets

The widgets in there are written for 9.0 and not 9.1 but it wouldn't be hard to make the needed changes.

closed 2024-12-17 by kdschlosser