lvgl-micropython/lvgl_micropython #188: OverflowError: overflow converting long int to machine word in File "touch_cal_data.py", line 105, in __init__

xiaohuboer, 2024-11-18:
I have  done calibration like this :
```
...
display = st7789.ST7789(
        data_bus=display_bus, 
        display_width=240,
        display_height=320,
        backlight_pin=14,
        )
touch_dev = machine.SPI.Device(
    spi_bus=spi_bus,
    freq=2500000,
    cs=46
    )
display.init()
th = task_handler.TaskHandler()
indev = xpt2046.XPT2046(touch_dev)
if not indev.is_calibrated:        
        indev.calibrate()
...
```
after the calibration, if I re-run my code, it raised error like below:
```
Traceback (most recent call last):
  File "<stdin>", line 51, in <module>
  File "<stdin>", line 40, in create_indev
  File "xpt2046.py", line 57, in __init__
  File "pointer_framework.py", line 21, in __init__
  File "touch_cal_data.py", line 105, in __init__
OverflowError: overflow converting long int to machine word
```

kdschlosser, 2024-11-18:
you need to update your clone. This issue has been fixed.

xiaohuboer, 2024-11-18:
I have alreay update the latest one. I recorded the steps below:
## behaviour 1 (click "ok" after calibration)
[Video: Behaviour 1](https://github.com/user-attachments/assets/bea6b71f-0caf-4ed2-bead-97f1d712fa8c)

the output in Thony:
[image]

## behaviour 2 (click "Try again" after calibration)
[Video: Behaviour 2](https://github.com/user-attachments/assets/94b151bd-3e7a-4b1e-afd9-f846b8e8a288)

After "reset" through "reset button" on board, I ran the same code again, the output in Thonny:
[image]

## source code
```python
import machine
import lcd_bus
import lvgl as lv
import task_handler  # NOQA

spi_bus = machine.SPI.Bus(
    host=2,
    mosi=47,
    miso=48,
    sck=21
    )

def create_display():
    display_bus = lcd_bus.SPIBus(
        spi_bus=spi_bus,
        dc=41,
        cs=42,
        freq=80000000,
        )
    
    import st7789
    display = st7789.ST7789(
        data_bus=display_bus, 
        display_width=240,
        display_height=320,
        backlight_pin=14,
        )
    
    return display

def create_indev():
    import xpt2046  # NOQA

    touch_dev = machine.SPI.Device(
        spi_bus=spi_bus,
        freq=2500000,
        cs=46
        )

    indev = xpt2046.XPT2046(touch_dev)
    
    return indev

if __name__ == '__main__':
    click_count = 0
    
    def button_event_handler(event):       
        global click_count 
        click_count += 1	
        label1.set_text(f"{click_count}")
    
    display = create_display()
    indev = create_indev()
    
    display.init()
    th = task_handler.TaskHandler()
    
    display.set_backlight(100) 
    if not indev.is_calibrated:        
        indev.calibrate()
       
    display.set_rotation(lv.DISPLAY_ROTATION._90)
    display.set_color_inversion(False)
    
    scrn = lv.screen_active()
    scrn.set_style_bg_color(lv.color_hex(0x000000), 0)
    
    btn1 = lv.button(scrn)
    btn1.set_size(80, 50)
    btn1.align(lv.ALIGN.CENTER, 0, 0)
    btn1.set_style_bg_color(lv.color_hex(0xBB5E00), 0)
    label1 = lv.label(btn1)
    label1.set_text("Browse")
    label1.align(lv.ALIGN.CENTER, 0, 0)
    
    btn1.add_event_cb(button_event_handler, lv.EVENT.CLICKED, None)
```

kdschlosser, 2024-11-18:
i am confused as to how you have 2 of the touch points having the exact same user input.

the first touch point has a screen location of 24, 24 and the second has a screen location of 24, 296 but both of those have a user input of 200, 282. 

I do not understand what would cause that kind of an issue.

Bigmaxmafia, 2024-11-19:
> i am confused as to how you have 2 of the touch points having the exact same user input.
>
> the first touch point has a screen location of 24, 24 and the second has a screen location of 24, 296 but [...]

It writes the same thing with the coordinates for me. ili9341+xpt

kdschlosser, 2024-12-01:
The touch calibration should now be fixed.

closed 2024-12-01 by kdschlosser