lvgl-micropython/lvgl_micropython #189: Is “USER_C_MODULES” parameter supported using make.py?

xiaohuboer, 2024-11-19:
Since I want to use ov2640 camera. I used below command to build firmware. (followed most step as https://github.com/lemariva/micropython-camera-driver)
The firmware can be generated , but its size is smaller (1.6m) ，and both `lvgl` and `camera` package could be found.
```bash
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT USER_C_MODULES=/home/a/gitlab/micropython-camera-driver/src/micropython.cmake --flash-size=16 CONFIG_OV2640_SUPPORT=y DISPLAY=st7796 DISPLAY=st7789 INDEV=xpt2046 INDEV=cst816s
```
Is there any error in my command ?

kdschlosser, 2024-11-19:
in order to use the USER_C_MODULES parameter what you need to do is place whatever module you want to add into the ext_mod folder. Then edit the micropython.cmake file in the est_mod folder and and add the include to point to the cmake file for the new user module you added.

xiaohuboer, 2024-11-20:
Thanks. It did work.

There were some compile issue happened during compiling camera module. I know these issues are NOT related to thsi repro, but can you give me some suggetion about how to solve it? I am not familar with the 'MP_QSTR_xx' strategy and not able to locate the corresponding source codes to check.

thanks in advance.

## steps
```
cd /home/a/gitlab/lvgl_micropython/lib/esp-idf/components
git clone https://github.com/espressif/esp32-camera

cd /home/a/gitlab/lvgl_micropython/ext_mod
git clone https://github.com/lemariva/micropython-camera-driver.git

gedit micropython.cmake
```
add the include to point to the cmake file for the new user module 
```
include(${CMAKE_CURRENT_LIST_DIR}/lcd_bus/micropython.cmake)
include(${CMAKE_CURRENT_LIST_DIR}/lvgl/micropython.cmake)
include(${CMAKE_CURRENT_LIST_DIR}/lcd_utils/micropython.cmake)
include(${CMAKE_CURRENT_LIST_DIR}/micropython-camera-driver/src/micropython.cmake)
```
update  `micropython-camera-driver/src/micropython.cmake`
```
...
# Define MODULE_CAMERA_ENABLED macro
target_compile_definitions(usermod_esp32camera INTERFACE MODULE_CAMERA_ENABLED=1)
...
```
update  `micropython-camera-driver/src/modcamera.c`
```
change all uppercase `STATIC` to lowercase `static` 
```
Compile the firmware...
```
cd /home/a/gitlab/lvgl_micropython
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16 ONFIG_OV2640_SUPPORT=y CONFIG_IDF_TARGET_ESP32S3=y DISPLAY=st7796 DISPLAY=st7789 INDEV=xpt2046 INDEV=cst816s
```

I got below error :
[image]

kdschlosser, 2024-11-20:
OK so I got it to compile.

This is what you need to do.

clone the [micropython-camera-driver](https://github.com/lemariva/micropython-camera-driver) repo into the ext_mod folder

edit `ext_mod/micropython.cmake` and add the following line

```cmake
include(${CMAKE_CURRENT_LIST_DIR}/micropython-camera-driver/src/micropython.cmake)
```

edit `ext_mod/micropython-camera-driver/src/micropython.cmake` and replace the entire file contents with this code

```cmake
# Create an INTERFACE library for our C module.
add_library(usermod_esp32camera INTERFACE)

list(APPEND IDF_COMPONENTS esp32-camera)

# Add our source files to the lib
target_sources(usermod_esp32camera INTERFACE ${CMAKE_CURRENT_LIST_DIR}/modcamera.c)

# Add include directories.
target_include_directories(usermod_esp32camera INTERFACE
    ${CMAKE_CURRENT_LIST_DIR}
    ${MICROPY_PORT_DIR}/managed_components/espressif__esp32-camera/driver/include
    ${MICROPY_PORT_DIR}/managed_components/espressif__esp32-camera/driver/private_include
    ${MICROPY_PORT_DIR}/managed_components/espressif__esp32-camera/conversions/include
    ${MICROPY_PORT_DIR}/managed_components/espressif__esp32-camera/conversions/private_include
    ${MICROPY_PORT_DIR}/managed_components/espressif__esp32-camera/sensors/private_include
)

# Link our INTERFACE library to the usermod target.
target_link_libraries(usermod INTERFACE usermod_esp32camera)
```

Edit `ext_mod/micropython-camera-driver/src/modcamera.c` and change `STATIC` to `static`

you are going to want to initialize the MicroPython submodule and to do that use these commands

```shell
git submodule update --init --jobs 4 -- lib/micropython
```

then edit `lib/micropython/ports/esp32/main_{ESP32_VARAINT}/idf_component.yml` where `{ESP32_VARAINT}` is `esp32`, `esp32c3`, `esp32s2` or `esp32s3`

Add `  espressif/esp32-camera: "~2.0.15"` on a new line right after `  espressif/esp_tinyusb: "~1.0.0"`

don't miss the 2 spaces at the beginning of the added line.

That should be all you need to do to get it up and running. I wish there was a way to do this programmatically but it is something I have not added the ability to do. I will update lvgl_micropython so it will be easier to add components from the esp component registry. I will also have the build script add the component to the build but unfortunately because of the order in which things take place things like the components includes cannot be collected in the cmake files. You can see that there are hard coded paths to the esp32-camera component in your user modules cmake file and those paths needed to be different because of how the build script runs.

I can however create a new cmake file that will set the path to the component so it will be easier to locate instead of having to prefix paths with `${MICROPY_PORT_DIR}/managed_components/espressif__esp32-camera/`

This is something I will mess about with over the next week or two.

xiaohuboer, 2024-11-21:
Hi @kdschlosser , thanks for your efforts.

You missed to enable `MODULE_CAMERA_ENABLED`, see below code in `modcamera.c`, which is the key to make compile error happen. Or else, even the firmware was built succesfully, `camera` package was NOT able to be imported in micropython.
[image]

I used your way and get the same issue. 
In your way, I have to do :
```
cp -r /home/a/gitlab/lvgl_micropython/lib/esp-idf/components/esp32-camera/ /home/a/gitlab/lvgl_micropython/lib/micropython/ports/esp32/managed_components/espressif__esp32-camera
```
not sure why `espressif__esp32-camera` was not be copied automatically even I added it into `lib/micropython/ports/esp32/main_esp32s3/idf_component.yml`

kdschlosser, 2024-11-21:
OK I am working on updating the build script so it will make it easier for you to get things up and running properly.

You shouldn't have to copy anything anywhere. When the script runs it should automatically download and install the esp32-camera component. So delete the one you have in the components folder. and delete the copy you made. I am going to add that macro and see what happens with the compilation.

It's probably something simple like a missing include.

kdschlosser, 2024-12-01:
I added the ability to supply a path to additional user_c_modules. You can supply the `USER_C_MODULE={path to module cmake}` multiple times if you want to add more than one additional user c module. You want to use the absolute path to avoid any possible issues with the mechanics in cmake. cmake and relative paths can get a little funky with the MicroPython build system,

I also added the means to tell the idf to add a component to the build as well. You do this using`COMPONENT={idf_compomnent}` where `{idf_compomnent}` would be the component registry nomenclature/syntax, IE: `COMPONENT="espressif/esp32-camera^2.0.15" ` You can also supply this build command multiple times if there is more than one component that needs to be downloaded and installed from the ESP32 component registry.

xiaohuboer, 2024-12-04:
when I ran below commad:
```bash
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16 DISPLAY=st7789 INDEV=xpt2046 COMPONENT="espressif/esp32-camera^2.0.15" USER_C_MODULE="/home/a/gitlab/micropython-camera-API/src/micropython.cmake"
```

I got error:
```bash
...
Compiling....
Executing action: add-dependency
ERROR: Directory "/home/a/gitlab/lvgl_micropython/main" does not exist! Please specify a valid component under /home/a/gitlab/lvgl_micropython or try to use --path
```

Btw, did I need put user module to `lvgl_micropython/ext_mod` now ?

Note: I have created a new enviroment to make sure all codes are latest.

kdschlosser, 2024-12-04:
> Btw, did I need put user module to lvgl_micropython/ext_mod now ?

no you do not. that is done automatically for you.

OK so what I need to do is I have to change the directory to where "main" is to get it to download the component. I am not sure as to why is need to be in that folder but apparently it does. I should be able to download a component and have it installed into the esp-idf components directory. I am going to poke about the idf.py file and see if there is a way to instruct it to do that.

kdschlosser, 2024-12-06:
I am still working on this issue. It is rather complicated because of the order in which things are done in the MicroPython build scripts. once I get it fully working I am going to let the folks know that there is an issue with the build script so they will be able to fix it upstream where the problem should be corrected. I really don't have much hope of them fixing it due to the track record of having issues fixed that I tell them about.

actually I can open the issue now seeing as how I know where the problem is I just am unsure of how to go about correcting it. Maybe they can shed some light on it. Dunno...

kdschlosser, 2024-12-17:
I am still working on this problem and I believe I have it figured out. Give me a day to get it running.

straga, 2024-12-19:
It will be nice if possible just build, lvgl for esp32 as USER_C_MODULES in the own env, without automatically adding freeze or other stuff.

kdschlosser, 2024-12-20:
the frozen modules are the display drivers. without them you are going to have to completely roll your own drivers and that would be a very large amount of work that doesn't need to be done. If you don't use the DISPLAY or INDEV build parameters the only thing that gets frozen into the firmware are the frameworks for the drivers and the task handler. But freezing is a necessity and is not something that is going to be removed.

It should be working with the last commit I pushed to the repo. Hopefully it should be anyhow. When supplying a user c module you can also optionally supply any esp-idf components your user c module depends on. The script will automatically import the new user C module and add any dependent components to the IDF component manager so it will be downloaded during the build routing. You should be able to include the headers from the component as you would if the component was apart of the ESP-IDF. The build script also adds the components to the list of dependencies for micropython which is what adds the include paths to the components to the compiler command when cmake calls the compiler.

straga, 2024-12-20:
Ok. Thanks. 
It is Not a problem to add py or mpy files to the board by yourself.

hcwang, 2025-03-19:
I still faced the same issue, is it fixed now? Thanks.
> ...
> Compiling....
> Executing action: add-dependency
> ERROR: Directory "/home/a/gitlab/lvgl_micropython/main" does not exist! Please specify a valid component under /home/a/gitlab/lvgl_micropython or try to use --path

ThomasFarstrike, 2025-04-25:
Forgot to tag @hcwang in my comment above :-)

1847123212, 2025-04-26:
does USER_C_MODULES  ov2640_cam support now?

kdschlosser, 2025-04-26:
If you submit a PR that corrects the problem I will merge it. I am not sure what is not working properly with it. If someone could explain in more detail what the exact problem is I would be more then happy to correct it.