lvgl-micropython/lvgl_micropython #190: ValueError: Frame buffer size is not correct for the display size, no matter what I try

MarkMakies, 2024-11-23:
Using a Waveshare ESP32-S3-Touch-LCD-7 board, 800 x 480,  8MB PSRAM, 8MB Flash, ST7262 display driver,  GT911 touch driver.  (https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-7#Overview)

I followed these build instructions:
```
sudo apt install build-essential
sudo apt install cmake
sudo apt install ninja-build
sudo apt install libusb-1.0-0-dev
sudo apt install git
sudo apt install python3-venv

git clone --recursive https://github.com/lvgl-micropython/lvgl_micropython.git
cd lvgl_micropython
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=rgb_display INDEV=gt911
/home/mark/.espressif/python_env/idf5.2_py3.12_env/bin/python -m esptool --chip esp32s3 -p /dev/ttyACM0 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 8MB --flash_freq 80m --erase-all 0x0 /home/mark/lvgl_micropython/build/lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-8.bin
```

My code seems ok, it's basically a copy of the sample with pin numbers allocated appropriately.
```

from i2c import I2C
import gt911
import lcd_bus
import lvgl as lv  # NOQA
import rgb_display

_WIDTH = 800
_HEIGHT = 480

I2C_BUS = I2C.Bus(
    host=1,
    scl=9,
    sda=8,
    freq=400000,
    use_locks=False
)

TOUCH_DEVICE = I2C.Device(
    I2C_BUS,
    dev_id=gt911.I2C_ADDR,
    reg_bits=gt911.BITS
)

bus = lcd_bus.RGBBus(
    hsync=46,
    vsync=3,
    de=5,
    pclk=7,
    data0=1,
    data1=2,
    data2=42,
    data3=41,
    data4=40,
    data5=39,
    data6=0,
    data7=45,
    data8=48,
    data9=47,
    data10=21,
    data11=14,
    data12=38,
    data13=18,
    data14=17,
    data15=10,
    freq=12000000,
    hsync_front_porch=10,
    hsync_back_porch=10,
    hsync_pulse_width=10,
    hsync_idle_low=False,
    vsync_front_porch=10,
    vsync_back_porch=10,
    vsync_pulse_width=10,
    vsync_idle_low=False,
    de_idle_high=False,
    pclk_idle_high=False,
    pclk_active_low=0,
)

display = rgb_display.RGBDisplay(
    data_bus=bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

```
**ValueError: Frame buffer size is not correct for the display size (768000)**
And no matter what I try over the course of a day, I cannot resolve what this error actually means.  I did ponder through the source code but I'm not proficient enough to work it out.

```
>>> ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x105c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbd8
load:0x403cc700,len:0x2e30
entry 0x403c989c
Traceback (most recent call last):
  File "main.py", line 64, in <module>
  File "rgb_display.py", line 38, in __init__
  File "rgb_display_framework.py", line 42, in __init__
  File "display_driver_framework.py", line 214, in __init__
  File "display_driver_framework.py", line 233, in _init_bus
ValueError: Frame buffer size is not correct for the display size (768000)
LVGL MicroPython 1.23.0 on 2024-11-23; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
>>> 
```
I would very much appreciate any help.

MarkMakies, 2024-11-23:
I also tried without BOARD_VARIANT rather than BOARD_VARIANT=SPIRAM_OCT, as I read in another issue that could be the problem, but I don't think so as that threw an error on initialization

> E (423) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode

So I must have SIPRAM 8 lane on my board.

favnec5, 2024-11-23:
The RGB bus is under developement by Kevin to improve performance and rotation.
If you need RGB driver quickly, you can use repo just before the last code update or here : https://github.com/favnec5/lvgl_micropython
It works but without rotation.

MarkMakies, 2024-11-23:
Hi favnec5,  I tried using the repo you suggested, but there appears to be coding errors:

```
/home/mark/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus.c: In function 'rgb_bus_trans_done_cb':
/home/mark/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus.c:61:9: error: expected ',' or ';' before 'if'
61 |         if (self->view2 != NULL && self->last_buf != curr_buf) {
|         ^~
/home/mark/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus.c:66:11: error: 'else' without a previous 'if'
66 |         } else if (self->view2 == NULL) {
|           ^~~~
/home/mark/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus.c:59:15: warning: unused variable 'curr_buf' [-Wunused-variable]
59 |         void *curr_buf = rgb_panel->fbs[rgb_panel->cur_fb_index]
|               ^~~~~~~~
[1364/1957] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/home/mark/lvgl_micropython/ext_mod/lcd_bus/esp32_src/i80_bus.c[1365/1957] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/home/mark/lvgl_micropython/lib/micropython/lib/littlefs/lfs2.c[1366/1957] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/home/mark/lvgl_micropython/lib/lvgl/src/core/lv_obj_class.c.ob[1367/1957] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/__/frozen_content.c.obj                                    [1372/1957] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/__/lv_mp.c.obj                                             
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/mark/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_610840 and /home/mark/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_610840
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: Leaving directory '/home/mark/lvgl_micropython/lib/micropython/ports/esp32'
make: *** [Makefile:66: all] Error 1
```

kdschlosser, 2024-11-23:
I believe I fixed the issue in the merge I just did. If you want to clone the repo again and give it a try to see if it is fixed.

favnec5, 2024-11-23:
my bad, this is fixed on my repo, lol we go back in time

favnec5, 2024-11-23:
@kevin
Sorry i'v black screen with last RVB display.
were you able to test it on a board of your own?

kdschlosser, 2024-11-23:
I have not had the time to. Too much going on with coming into the holiday season. Getting things ready for friends and family when they visit. The next month is going to be pretty hectic for me.

favnec5, 2024-11-23:
damn, Family is important, enjoy your loved ones :-)

MarkMakies, 2024-11-24:
Thanks guys, code runs now without error, but like you I've only got a blank screen

MarkMakies, 2024-11-24:
Sorry, new to this system.  Closing this issue as it is already progressing in the discussion forum.

closed 2024-11-24 by MarkMakies