lvgl-micropython/lvgl_micropython #194: Can this project be used on monochrome or 8-color screens?

jd3096-mpy, 2024-11-25:
If you have free time, you should try running it on classic monochrome OLEDs like the SSD1306 or on Sharp's 8-color Memory LCD. It should be quite interesting.

jd3096-mpy, 2024-12-11:
I have successfully implemented a pure Python driver for the most common monochrome OLED, the SSD1306. Below is the test code. Although there is still plenty of room for optimization, it works as expected. I hope this can serve as a starting point to inspire others to adapt more screens.

```
from machine import SoftSPI,Pin
from ssd1306 import SSD1306_SPI
import time
res=Pin(4)
dc=Pin(5)
cs=Pin(6)
spi = SoftSPI(baudrate=1000000,sck=Pin(1),mosi=Pin(2),miso=Pin(0))

oled = SSD1306_SPI(128, 64, spi,dc=dc,res=res,cs=cs)

import lcd_bus
import lvgl as lv
if not lv.is_initialized():
    lv.init()

disp = lv.display_create(128,64)
disp.set_color_format(lv.COLOR_FORMAT.RGB565)
b1 = bytearray(128*64*2)
b2 = bytearray(128*64*2)
buf1 = memoryview(b1)
buf2 = memoryview(b2)
disp.set_buffers(buf1,buf2,len(buf1),lv.DISPLAY_RENDER_MODE.FULL)
pixel_size = lv.color_format_get_size(lv.COLOR_FORMAT.RGB565)

def flush_cb(_, area, color_p):
    x1 = area.x1
    x2 = area.x2
    y1 = area.y1
    y2 = area.y2
    print(x1,x2,y1,y2)
    size = ((x2 - x1 + 1) *(y2 - y1 + 1) *2 )
    data_view = bytes(color_p.__dereference__(size))
    print(len(data_view))
    i=0
    hp=0
    print(data_view[0:10])
    for y in range(y1,y2+1):
        for x in range(x1,x2+1):
            if data_view[i]!=0:
                oled.pixel(x,y,1)
                hp+=1
            i+=2
            

    oled.show()
    disp.flush_ready()
    
disp.set_flush_cb(flush_cb)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(60, 10)
slider.center()

label = lv.label(scrn)
label.set_text('MPY LVGL!')
label.align(lv.ALIGN.CENTER, 0, -20)

import th
go = th.TaskHandler()
```

closed 2024-12-11 by jd3096-mpy

kdschlosser, 2024-12-11:
I added the display driver. it can be used with either the lcd_bus.SPIBus or lcd_bus.I2CBus drivers.