lvgl-micropython/lvgl_micropython #195: pclk_active_low - meaning?

MarkMakies, 2024-11-25:
I'm using a Waveshare ESP32-S3-Touch-LCD-7 board, 800 x 480, 8MB PSRAM, 8MB Flash, ST7262 display driver, GT911 touch driver. (https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-7#Overview)

Initially the display did not show fonts very well, they looked a bit blurry, so I tried playing around with the lcd_bus parameters.  Varying frequency and sync settings did little to improve the situation, however setting pclk_active_low to True (was False) made a very big improvement to the clarity - it looks like a different display now.  I haven't been able to find much information on how to best fine tune these settings, or what they actually do at a hardware/driver level.

It would be great if someone could point me in the right direction to best setup these parameters.  For my project clarity is more important than frames rate - I recon there would be a sweet spot somewhere.

```
from micropython import const
from i2c import I2C
import lcd_bus
import gt911
import rgb_display
import lvgl as lv
import task_handler

WIDTH = const(800)
HEIGHT = const(480)

i2c_bus = I2C.Bus(host=1, scl=9, sda=8, freq=400000, use_locks=False)
touch_device = I2C.Device(i2c_bus, dev_id=gt911.I2C_ADDR, reg_bits=gt911.BITS)

display_bus = lcd_bus.RGBBus(
    hsync=46, vsync=3, de=5, pclk=7,
    data0=14, data1=38, data2=18, data3=17, data4=10, 
    data5=39, data6=0, data7=45,data8=48, data9=47, data10=21, 
    data11=1, data12=2, data13=42, data14=41, data15=40,
    freq=12000000,
    hsync_front_porch=10, hsync_back_porch=10, hsync_pulse_width=10, hsync_idle_low=False,
    vsync_front_porch=10, vsync_back_porch=10, vsync_pulse_width=10, vsync_idle_low=False,
    de_idle_high=False, pclk_idle_high=False, pclk_active_low=True)

display = rgb_display.RGBDisplay(data_bus=display_bus, display_width=WIDTH, display_height=HEIGHT,
    color_space=lv.COLOR_FORMAT.RGB565,rgb565_byte_swap=False)

display.init()

indev = gt911.GT911(touch_device)
if indev.hw_size != (WIDTH, HEIGHT):
    fw_config = indev.firmware_config
    fw_config.width = WIDTH
    fw_config.height = HEIGHT
    fw_config.save()
    del fw_config

scrn = lv.screen_active()
th = task_handler.TaskHandler()

scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

dd = lv.dropdown(scrn)
dd.set_options("\n".join([
    "Apple",
    "Banana",
    "Orange",
    "Cherry",
    "Grape",
    "Raspberry",
    "Melon",
    "Orange",
    "Lemon",
    "Nuts"]))

dd.align(lv.ALIGN.TOP_MID, 0, 20)
```

kdschlosser, 2024-11-26:
the pclk is the pixel clock. The polarity is what is supposed to represent an on or an off in the clock signal. some click treat a 1 as on while others treat a 0 as on. It just depends on the display IC

MarkMakies, 2024-11-26:
Thanks Kevin,  I've checked the ST7262 datasheet and the DCLKP pin does clock on the falling edge.  Amazing it worked at all!

Looking at the datasheet, I'm assuming that the actual clock frequency is double what _freq_ parameter is set to.  Following on from this all the figures in this table which have the 'Unit measure' as DCLK or MHz should be halved to reflect the  values seen at the chip.

So for the fastest refresh,  whilst keeping in spec, 
```
freq=13500000,  hsync_front_porch=2, hsync_back_porch=2, hsync_pulse_width=1, 
vsync_front_porch=4, vsync_back_porch=4, vsync_pulse_width=2
```
Which appears to work well in practice.  Would you agree with my logic here, or am I missing something?

[image]

kdschlosser, 2024-11-26:
There are many things that will effect the pixel click you need to set. This is the "freq" parameter when starting the RGB bus driver.
The biggest thing that is going to effect it is going to be the sharing of the SPI bus for the PSRAM. because the RGB bus driver needs to have 2 frame buffers to operate properly and those frame buffers reside in DMA memory that allows the CPU to process other code when transferring buffer data. So while one buffer is sending the other is able to be filled. Due to the very large size of the frame buffers that are needed for an RGB display they are only going to fit into PSRAM (SPIRAM). This is a very slow RAM at only 80mhz at it's typical default max speed. When reading and writing to the RAM at the same time that cuts the speed in half. The number of lanes that are available also effects it.

How you dial in the best pixel clock is you are going to want to set up a test that runs for the maximum amount of things that is going to be updated at the same time in your UI. Then run the program. if it runs without any kind of visual anomalies then up the frequency by 500000hz. Once you get to a point where it's not going to be able to keep up you should see your UI start to walk horizontally. This might take some time to happen depending on how close the set pixel clock is to how fast the buffers are able to be updated.

Typically I have seen pretty good results setting the freq to 12500000hz sometimes as high as 13000000hz. It is something you are going to have to mess about with.

RGB displays are the hardest to work with because of the display not having a control IC or GRAM. That means those functions need to be supplied by the ESP32. I have been working on improving the performance but there is always a cost and it is usually memory. I have been working on sharing the load with the second core on the ESP32 but I have not gotten it dialed in 100%. It's damned close tho. I am pretty sure that this change is going to help a lot. once I get it up and running properly I will run some tests to see if the effort was worth it. and if the additional memory use makes it worth it. I am pretty sure it will. Plus it will also allow for the rotation of the RGB displays which currently is horribly slow because of the internal LVGL mechanics that handle rotation. 

I also did want to mention that depending on the ESP32 you are using and what the SPI connection is to the PSRAM and also the flash it is possible to overclock the PSRAM to get an effective transfer speed of 240mhz instead of the typical 80mhz. You would need to attach a heat sink to the ESP32 to improve cooling and there would also be a higher power consumption from the ESP32 but it is possible to achieve better transfer speeds with the PSRAM which is going to give better transfer speeds to the display.

closed 2024-11-30 by MarkMakies