lvgl-micropython/lvgl_micropython #201: Cannot access lcd_bus.SPI3Wire

grafail, 2024-12-02:
I am trying to get this running on a [Sensecap Indicator D1](https://wiki.seeedstudio.com/Sensor/SenseCAP/SenseCAP_Indicator/Get_started_with_SenseCAP_Indicator/) (ST7701S based) and I am having trouble importing `lcd_bus.SPI3Wire`.
Trying currently to go mostly off this issue https://github.com/lvgl-micropython/lvgl_micropython/issues/111. Not sure if I am missing some build argument or config to get this class. I can see it at the stub file on the repo though.

**Repl Output**:
```py
>>> import lcd_bus
>>> lcd_bus.__dict__
{'__name__': 'lcd_bus', 'RGBBus': <class 'RGBBus'>, 'SPIBus': <class 'SPI_Bus'>, 'I2CBus': <class 'I2CBus'>, 'I80Bus': <class 'I80Bus'>, '_pump_main_thread': <function>, 'DEBUG_ENABLED': 0, 'MEMORY_32BIT': 2, 'MEMORY_8BIT': 4, 'MEMORY_DMA': 8, 'MEMORY_SPIRAM': 1024, 'MEMORY_INTERNAL': 2048, 'MEMORY_DEFAULT': 4096}
>>> lcd_bus.SPI3Wire
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'SPI3Wire'
>>> 
```

kdschlosser, 2024-12-03:
It's not in the lcd_bus module. It's in the spi3wire module.

grafail, 2024-12-03:
@kdschlosser Oh I see, thank you for your reply. I was confused because I saw it typed differently here:
https://github.com/lvgl-micropython/lvgl_micropython/blob/main/lcd_bus.pyi

kdschlosser, 2024-12-03:
Thanks for pointing that out. I have to update the stub file for it. I forgot to do that..

closed 2024-12-17 by kdschlosser