lvgl-micropython/lvgl_micropython #202: SPI.Bus How should I use it

center789, 2024-12-04:
import st7789  # NOQA
import lvgl as lv  # NOQA
import lcd_bus
from micropython import const
import machine
import task_handler

_WIDTH = const(240)
_HEIGHT = const(240)
_BL = const(15)
_RST = const(9)
_DC = const(13)

_MOSI = const(11)
_MISO = const(0)
_SCK = const(12)
_HOST = const(1)  # SPI2

_LCD_CS = const(10)
_LCD_FREQ = const(80000000)

_TOUCH_CS = const(18)
_TOUCH_FREQ = const(10000000)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)

display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_RGB,
    rgb565_byte_swap=True,
)

lv.init()

th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)

AttributeError: type object 'SPI' has no attribute 'Bus'

I would like to ask, after compiling the firmware, the SPI Bus cannot be used. How should I use it? Where might the issue be? I am using an ST7789 screen.

kdschlosser, 2024-12-04:
what MCU are you using. Currently this library only works with the ESP32 line of MCU's. I should be able to get the SPI working fairly quickly for the STM MCU's and the RP2 MCU. Others will take a little bit of time, maybe a couple of days.

center789, 2024-12-04:
> 您使用的是什么 MCU。目前，此库仅适用于 ESP32 系列 MCU。我应该能够让 STM MCU 和 RP2 MCU 的 SPI 相当快地工作。其他 MCU 可能需要一点时间，也许几天。

I am using the ESP32-S3. I referred to someone else's code and now I know how to use this SPI. The display seems to be working, but the screen shows artifacts (garbled display). I'm not sure what the reason is.
[image]

center789, 2024-12-05:
> your MISO needs to be -1 not 0.
> 
> drop the LCD frequency to 40000000.
> 
> Other than that it looks good.
> 
> If you provide me with a link to the actual display I will be able to help you further.

import st7789  # NOQA
import lvgl as lv  # NOQA
import lcd_bus
from micropython import const
import machine

spi_bus = machine.SPI.Bus(host=1, mosi=11, miso=45 , sck=12)

display_bus = lcd_bus.SPIBus(spi_bus=spi_bus, dc=13, cs=10, freq=40000000)
display = st7789.ST7789(
    data_bus=display_bus,
    display_width=240,
    display_height=240,
    reset_pin=9,
    reset_state=st7789.STATE_LOW,
    backlight_pin=17,
    backlight_on_state=st7789.STATE_PWM,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_RGB,
    rgb565_byte_swap=True,
)

display.init()

style_base = lv.style_t()
style_base.init()
style_base.set_bg_color(lv.palette_main(lv.PALETTE.LIGHT_BLUE))
style_base.set_border_color(lv.palette_darken(lv.PALETTE.LIGHT_BLUE, 3))
style_base.set_border_width(2)
style_base.set_radius(10)
style_base.set_shadow_width(10)
style_base.set_shadow_offset_y(5)
style_base.set_shadow_opa(lv.OPA._50)
style_base.set_text_color(lv.color_white())
style_base.set_width(100)
style_base.set_height(lv.SIZE_CONTENT)

# Set only the properties that should be different
style_warning = lv.style_t()
style_warning.init()
style_warning.set_bg_color(lv.palette_main(lv.PALETTE.YELLOW))
style_warning.set_border_color(lv.palette_darken(lv.PALETTE.YELLOW, 3))
style_warning.set_text_color(lv.palette_darken(lv.PALETTE.YELLOW, 4))

# Create an object with the base style only
obj_base = lv.obj(lv.screen_active())
obj_base.add_style(style_base, 0)
obj_base.align(lv.ALIGN.LEFT_MID, 20, 0)

label = lv.label(obj_base)
label.set_text("Base")
label.center()

# Create another object with the base style and earnings style too
obj_warning = lv.obj(lv.screen_active())
obj_warning.add_style(style_base, 0)
obj_warning.add_style(style_warning, 0)
obj_warning.align(lv.ALIGN.RIGHT_MID, -20, 0)

label = lv.label(obj_warning)
label.set_text("Warning")
label.center()

python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7789 INDEV=gt911 --flash-size=16

center789, 2024-12-05:
My display does not have an MISO pin, so it is typically assigned to -1 or some other unused pin. With the firmware compiled using lv_micropython, these pin definitions work fine, but perhaps the issue is that the display content exceeds the range. Is there a way to define X and Y offset corrections for the screen? I'd like to try this to see if the issue is caused by content going out of range.

kdschlosser, 2024-12-05:
yes you can but those will not cause the display to act like what you are seeing unless the width and height you are providing to the display is way out of bounds.

If I had access to the datasheet for the display there are some thing I can check to make sure everything is being set right. The initialization commands that are sent to the display IC are not going to be specific to a panel that is attached to that IC. The ones that are sent is a best attempt to get the display up and running and it may not work for all display even tho the display IC is the same. There are settings that need to be set that can be specific to the actual panel. I need the data sheet to see what would be needed. also an example provided by then would be helpful as well.

kdschlosser, 2024-12-05:
also are you sure the display IC is the st7789?..

center789, 2024-12-05:
Yes, I am certain it is an ST7789, a 1.69-inch screen with a resolution of 240x280. The interface type is SPI, with a COG connection method. The driver IC is ST7789P3. The visible area is 29.37x34.33 mm, and the active area is 29.97x32.63 mm. The FPC interface pins are labeled as follows: GND, LEDK, LEDA+VDD, VDDIO_1.8, GND, GND, DC, CS, SCL, SDA, RESET, TE.

kdschlosser, 2024-12-05:
I bet you have to drive the TE pin either high or low. I don't remember off hand.

kdschlosser, 2024-12-05:
> My display does not have an MISO pin, so it is typically assigned to -1 or some other unused pin.

When using SPI HOST2 (1) that host can only be used with the hardware pins. so if you are not using the MISO pin it MUST be set to -1. If it is set to any other pin it will give you undefined behavior. Now if you used SPI HOST3 (2) then you can set whatever pins you want to as that uses the GPIO matrix instead if the IOMUX. Using the IOMUX is going to allow for faster data transfer speeds to the display. You should be able to get to at least 40mhz and if the wires are short enough and the connections are solid you can should be able to get to 80mhz.

Try using this code to see if it works...

```python
import machine
import lcd_bus

spi_bus = machine.SPI.Bus(
    host=1,
    mosi=11,
    miso=-1,
    sck=12
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=13,
    cs=10,
    freq=40000000
)

buf1 = display_bus.allocate_framebuffer(17280, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
buf2 = display_bus.allocate_framebuffer(17280, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import st7789 # NOQA
import lvgl as lv # NOQA

display = st7789.ST7789(
    data_bus=display_bus,
    display_width=240,
    display_height=240,
    frame_buffer1=buf1,
    frame_buffer2=buf2,
    reset_pin=9,
    reset_state=st7789.STATE_LOW,
    backlight_pin=17,
    backlight_on_state=st7789.STATE_PWM,
)

display.set_power(True)
display.init()
display.set_backlight(100)

import task_handler

task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.colorhex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(150, 50)
slider.center()
```

Leave the script exactly how it is do not change anything. The imports are done like they are for a reason.\ There is only so much internal DMA memory available so we don't want to fill it up with other things. So only import the necessities to get the lcd bus running so the buffers can then be allocated. This will ensure that there is enough DMA memory available to do that.

I am also setting the color depth to RGB888. the display IC supports it and with the double buffering and DMA memory being used coupled with the display having a low pixel count there is not going to be any noticeable performance impact. I also want to remove the code used for RGB565 from being a possible cause.

lets see if that works for ya.,.

center789, 2024-12-05:
[image]

Yes, this code works perfectly to make the display function properly. Thank you so much for your patient guidance! ☺

kdschlosser, 2024-12-05:
from your picture it doesn't look like it's perfect. is that what it is looking like with the area that is all static?

center789, 2024-12-06:
[image]
It seems like the manufacturer labeled the resolution size incorrectly. The actual resolution should be 240x300, but this isn't a big issue.

kdschlosser, 2024-12-06:
and what are you getting for a frames per second on that. It looks like it is running pretty good. You can try adjusting the bus frequency up to 80Mhz that should double the speed of sending the buffer data.

kdschlosser, 2024-12-06:
you should also change the frame buffer size from 17280 to 21600 with the increase in resolution. The math for the frame buffer size is width * height * 3 / 10

center789, 2024-12-06:
`
fps_label = lv.label(scrn)
fps_label.set_text("FPS: 0.00")
fps_label.align(lv.ALIGN.TOP_MID, 50, 50)

last_frame_time = time.ticks_us()  

def update_fps():
    global last_frame_time
    current_time = time.ticks_us()
    elapsed_time = time.ticks_diff(current_time, last_frame_time) / 1000000
    if elapsed_time > 0:
        fps = 1 / elapsed_time
        fps_label.set_text(f"FPS: {fps:.2f}")
    last_frame_time = current_time
`
The average frame rate is around 18, but I'm not sure if my calculation is accurate. As for the frequency, I watched someone else's test video, and it actually tops out at 60 MHz anything higher doesn’t make a difference.

kdschlosser, 2024-12-06:
>  it actually tops out at 60 MHz anything higher doesn’t make a difference.

That might be because of the frame buffer sizes. You really need to adjust those. The other thing that is going to be limiting you is how frequent LVGL is refreshing the display. the default time is 33 milliseconds. and the default time for the task handler is 33 milliseconds as well.

You should something along the lines of what is seen in the code below to get a real frames per second time.
This is pseudo code and it has not been tested. There are likely things that will need to be changed and it to only be used as a reference.
```
scrn = lv.screen_active()

disp = scrn.get_display()

timer = disp.get_refr_timer()
# 1 millisecond between refreshes of the display.
timer.set_period(1)

# times and amounts used in the calculations
flush_start_time = time.ticks_us()
frame_count = 0
total_elapsed_time = 0
print_start_time = time.ticks_us()

# we are going to override the flush callback in the python display driver 
# replacing it with the one seen below. 
def flush_cb(disp_, area, color_p):
    global flush_start_time
    global frame_count
    global total_elapsed_time
    global print_start_time

    is_last = disp_.flush_is_last()

    # call the original flush function in the python display driver
    # we do this before doing the calculations because this piece of the buffer 
    # is the last to be flushed for the entire frame. 
    display._flush_cb(disp_, area, color_p)
    
    # Because the frame buffer is cut into 1/10th the size of the frame it may take more than one flush to update the entire frame.
    # so we check to se if the flush is the last in the run to update the whole frame.
    if is_last:
        flush_end_time = time.ticks_us()
        # add the .0 on the end of the divisor in order to have a float returned so it is fractional.
        frame_count += 1
        # add the time to the total for later use when calculating the FPS
        
        total_elapsed_time += time.ticks_diff(flush_end_time, flush_start_time)

        print_diff = time.ticks_diff(flush_end_time, print_start_time) / 1000000.0
        if print_diff >= 1000:  # 1 second
            total_time = total_elapsed_time / 1000000.0
            # get the average per frame.
            avg_time = total_time / frame_count
            
            # use that everage to get the frame per second. rounded to 2 decimal positions
            # the reason why we are using a print statement instead of having a widget 
            # display it is because any change to the widget that is displaying the FPS
            # is going to trigger the flush to run and that is going to skew the measurement
            print('FPS:', round(1.0 / avg_time, 2))

            # reset the variables to get another reading. 
            total_elapsed_time = 0
            frame_count = 0
            print_start_time = time.ticks_us()
        
        # do code here to move the object and set it's color

        # reset the start time after the code to change the object as it 
        # shouldn't be in the FPS Count.
        flush_start_time = time.ticks_us()

disp.set_flush_cb(flush_cb)

import task_handler
# set the task handler so it will check lvgl for any data that needs to be rendered every 1 millisecond.
task_handler.TaskHandler(1)

```

kdschlosser, 2024-12-06:
also to let you know. If you edit the `lib/lv_conf.h` file and change the following settings

```c
#define LV_DEF_REFR_PERIOD  33

#define LV_USE_OBSERVER 1

#define LV_USE_SYSMON   0

#define LV_USE_PERF_MONITOR 0
```

to the following

```c
#define LV_DEF_REFR_PERIOD  1

#define LV_USE_OBSERVER 0

#define LV_USE_SYSMON   1

#define LV_USE_PERF_MONITOR 1
```

and recompile the firmware and then in the python code change

```python
task_handler.TaskHandler()
```

to 

```python
task_handler.TaskHandler(1)
```

it will display the FPS for you as well. LVGL has built in mechanics to display the frame rate.

center789, 2024-12-06:
`Including User C Module(s) from ../../../../../ext_mod/micropython.cmake
Traceback (most recent call last):
File "/home/suadmin/lvgl_micropython/gen/python_api_gen_mpy.py", line 623, in <module>
ast = cparser.parse(pp_data, input_headers[0])
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/suadmin/lvgl_micropython/lib/pycparser/pycparser/c_parser.py", line 147, in parse
return self.cparser.parse(
^^^^^^^^^^^^^^^^^^^
File "/home/suadmin/lvgl_micropython/lib/pycparser/pycparser/ply/yacc.py", line 331, in parse
return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/suadmin/lvgl_micropython/lib/pycparser/pycparser/ply/yacc.py", line 1199, in parseopt_notrack
tok = call_errorfunc(self.errorfunc, errtoken, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/suadmin/lvgl_micropython/lib/pycparser/pycparser/ply/yacc.py", line 193, in call_errorfunc
r = errorfunc(token)
^^^^^^^^^^^^^^^^
File "/home/suadmin/lvgl_micropython/lib/pycparser/pycparser/c_parser.py", line 1931, in p_error
self._parse_error(
File "/home/suadmin/lvgl_micropython/lib/pycparser/pycparser/plyparser.py", line 67, in _parse_error
raise ParseError("%s: %s" % (coord, msg))
pycparser.plyparser.ParseError: /home/suadmin/lvgl_micropython/lib/lvgl/src/others/sysmon/lv_sysmon.h:39:5: before: lv_subject_t
-- Configuring incomplete, errors occurred!
OUTPUT:
RESULT: 1
CMake Error at /home/suadmin/lvgl_micropython/ext_mod/lvgl/micropython.cmake:41 (message):
Failed to generate
`

#define LV_USE_SYSMON cannot be changed from the default 0 to 1 because I tried modifying it and encountered errors when compiling the firmware. The other settings can be modified without errors, but I won’t try to investigate this further, as it’s beyond my knowledge and I can’t solve the other issues I’ve encountered. Anyway, I just want to say thank you, I really appreciate your positive responses and the support you've given me.

kdschlosser, 2024-12-06:
I will investigate why that is not working. 

I am going to close this issue because it is technically solved.
I am happy to have been able to assist you in getting your display up and running.

closed 2024-12-06 by kdschlosser