lvgl-micropython/lvgl_micropython #203: many, many, many issues when using thonny

alseenwalnoot, 2024-12-07:
Okay, so i just switched from arduino + tft_espi to this, it is all set up and working now, however there are lots of issues:

- Random panics
- Random error when trying: spi_bus = SPI.Bus
It does work however it just gives me a typerror when it feels like it, and then i have to replug and restart the esp32
- Crashes/ not working WHENEVER IT FEELS LIKE IT!!

all these problems get fixed when i just run my program via boot.py, and not when i run it via thonny.

Is this problem recognised yet or not? 

Device: ESP32 (generic) ili9488, and XPT2046.
code: from machine import SPI, Pin
import lvgl as lv
import lcd_bus
import ili9488
import xpt2046
import task_handler

# Button callback
def btn_cb(event):
    if event.get_code() == lv.EVENT.CLICKED:
        print('Button clicked!')
        msg_txt.set_text('Button clicked!')
        
spi_bus = SPI.Bus(
    host=1,
    mosi=23,
    miso=19,
    sck=18
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=2,
    cs=5,
    freq=40000000,
)

display = ili9488.ILI9488(
    data_bus=display_bus,
    display_width=320,
    display_height=480,
    reset_pin=4,
    reset_state=ili9488.STATE_LOW,
    color_space=lv.COLOR_FORMAT.RGB888,
    rgb565_byte_swap=True
)
display.init()

# Initialize the XPT2046 touch controller

touch_dev = SPI.Device(spi_bus=spi_bus, freq=1_000_000, cs=15)
touch = xpt2046.XPT2046(touch_dev, startup_rotation=lv.DISPLAY_ROTATION._0, debug=True)
# Create the main screen
if not touch.is_calibrated: touch.calibrate()
main = lv.screen_active()

# Add a label
msg_txt = lv.label(main)
msg_txt.align(lv.ALIGN.CENTER, 0, 80)
msg_txt.set_text('Hello World!')

# Add a button with a label
btn = lv.button(main)
btn.set_size(90, 40)
btn.align(lv.ALIGN.CENTER, 0, 0)
btn.add_event_cb(btn_cb, lv.EVENT.CLICKED, None)

btn_lbl = lv.label(btn)
btn_lbl.set_text("Click Me!")
btn_lbl.align(lv.ALIGN.LEFT_MID, 0, 0)

# Set backlight for the display
display.set_backlight(100)

# Start the task handler
th = task_handler.TaskHandler()

kdschlosser, 2024-12-08:
Thonny does some odd things when handling the ESP32. One of the biggest things is the fact that it doesn't hard reset the device. It only does a soft reset which only clears the MicroPython heap. All hardware settings persist between the restarts and when MicroPython starts back up and the display bus code runs you can end up with undefined behavior. There is no way to hook the restart to know that the hardware needs to be reset.

kdschlosser, 2024-12-08:
If you press the reset button on the ESP you will see that everything will end up working as normal until Thonny does a reset of the ESP.

h-milz, 2024-12-08:
I can only confirm what @kdschlosser says. I'm using Thonny as well with an ESP32-S3, and from what I learned, Thonny uses the raw REPL to do certain things. Apparently, the raw REPL is sometimes screwed up and Thonny fails. It turned out if I reset the ESP and klicked the stop sign in Thonny, things get back to normal again. 

An alternative would be to use mpremote for accessing the REPL if you happen to be on Linux.

alseenwalnoot, 2024-12-08:
alright, i will definetly look into mpremote. currently im using uPyCraft which runs a bit more stable, but still is nothing more then thonny with a bit better ide. i was also looking into PyMakr, but that wouldnt even connect. @kdschlosser, isnt there a option to make it always hard reset?. I also found that indeed, thonny's behaviour is just inpredictable when running a script via Thonny. geuss i'l have to use uPyCraft for now

kdschlosser, 2024-12-08:
I used a modified version of a python script called ampy and I use a terminal program to access the REPL. both the terminal program and ampy both send the correct signals up the hardware flow control lines to perform a hard reset when connecting to the ESP32.

closed 2024-12-17 by kdschlosser