lvgl-micropython/lvgl_micropython #204: WT32-SC01 Plus

sasakihin, 2024-12-08:
I am using WT32-SC01 Plus and compiling lvgl_micropython with the following command:
`python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 DISPLAY=st7796 INDEV=ft6x36 --flash-size=16
`
However, when testing with the examples provided in the README.md, the following error occurs:

indev = ft6x36.FT6x36(touch_dev)

MPY: soft reboot
Traceback (most recent call last):
  File "<stdin>", line 71, in <module>
  File "ft6x36.py", line 26, in __init__
  File "focaltech_touch.py", line 58, in __init__
  File "focaltech_touch.py", line 117, in _read_reg
  File "i2c.py", line 91, in write_readinto
  File "i2c.py", line 107, in read_mem
  File "i2c.py", line 73, in readfrom_mem_into
OSError: [Errno 19] ENODEV

I tried commenting out the line:

# indev = ft6x36.FT6x36(touch_dev)

However, I encountered the following issue:

MPY: soft reboot
Traceback (most recent call last):
  File "<stdin>", line 73, in <module>
AttributeError: 'ST7796' object has no attribute 'invert_colors'

kdschlosser, 2024-12-08:
My suggestion to you is to only use Thonny for uploading files to the display. DO NOT use it to flash the firmware. After you compile the firmware the last couple of lines in the output is the instruction on how to flash the ESP32..

Here is some code you can use to get the display up and running. The build command you need to use is located at the top of the script.

```python
# build command
# python3 esp32 BOARD=ESP32_GENERIC_S3 DISPLAY=st7796 INDEV=ft6x36 --flash-size=16

from micropython import const

_WIDTH = const(320)
_HEIGHT = const(480)

_FREQ = const(40000000)
_D7 = const(15)
_D6 = const(16)
_D5 = const(17)
_D4 = const(18)
_D3 = const(8)
_D2 = const(3)
_D1 = const(46)
_D0 = const(9)

_WR = const(47)
_RS = const(4)
_DC = const(0)
_CS = const(6)
_BL = const(45)
_TE = const(47)

_TP_SDA = const(6)
_TP_SDL = const(5)
_TP_RST = const(4)
_TP_HST = const(0)
_TP_INT = const(7)
_TP_FREQ = const(400000)

_SD_MOSI = const(40)
_SD_MISO = const(38)
_SD_SCK = const(39)
_SD_CS = const(41)

_EXT_IO1 = const(10)
_EXT_IO2 = const(11)
_EXT_IO3 = const(12)
_EXT_IO4 = const(13)
_EXT_IO5 = const(14)
_EXT_IO6 = const(21)

import lcd_bus

display_bus = lcd_bus.I80Bus(
    wr=_WR,
    dc=_DC,
    cs=_CS,
    data0=_D0,
    data1=_D1,
    data2=_D2,
    data3=_D3,
    data4=_D4,
    data5=_D5,
    data6=_D6,
    data7=_D7,
    dc_data_high=True,
    dc_cmd_high=False,
    dc_idle_high=False,
    dc_dummy_high=False
)

# Use DMA transfers with double buffering for best performance. 
# The buffers should fit into the internal memory as this is 
# going to be the fastest memory available 
buf1 = display_bus.allocate_framebuffer(int(_WIDTH * _HEIGHT * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
buf2 = display_bus.allocate_framebuffer(int(_WIDTH * _HEIGHT * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import st7796
import lvgl as lv

display_driver = st7796.ST7796(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    frame_buffer1=buf1,
    frame_buffer2=buf2,
    reset_pin=_RS,
    
    # You may have to set this to STATE_LOW if you get a black screen
    reset_state=st7796.STATE_HIGH,
    
    backlight_pin=_BL,
    backlight_on_state=st7796.STATE_PWM,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display_driver.set_power(True)
display_driver.init()
display_driver.send_params(0x34)  # TEOFF
display_driver.set_color_inversion(True)

import i2c

import ft6x36

touch_bus = i2c.I2C.Bus(
    host=_TP_HST,
    sda=_TP_SDA,
    scl=_TP_SDL,
    freq=_TP_FREQ
)

touch_device = i2c.I2C.Device(
    bus=touch_bus,
    dev_id=ft6x36.I2C_ADDR,
    reg_bits=ft6x36.BITS
)

indev = ft6x36.FT6X36(device=touch_device)

import time
import task_handler

display_driver.set_rotation(lv.DISPLAY_ROTATION._90)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

lv.refr_now(scrn.get_display())

for i in range(1, 101, 1):
    display_driver.set_backlight(i)
    time.sleep_ms(3)

task_handler.TaskHandler()

```

sasakihin, 2024-12-08:
I generally use flash_download_tool_3.9.7 to flash the *.bin files.

I used the above code but still encountered errors

```
MPY: soft reboot
Traceback (most recent call last):
  File "<stdin>", line 95, in <module>
AttributeError: 'ST7796' object has no attribute 'send_params'

```

> My suggestion to you is to only use Thonny for uploading files to the display. DO NOT use it to flash the firmware. After you compile the firmware the last couple of lines in the output is the [...]

I generally use flash_download_tool_3.9.7 to flash the *.bin files.

I used the above code but still encountered errors

```
MPY: soft reboot
Traceback (most recent call last):
  File "<stdin>", line 95, in <module>
AttributeError: 'ST7796' object has no attribute 'send_params'

```

h-milz, 2024-12-08:
Please note that your runtime error is unrelated to how you flash your chip. 

In order to find out if there may be a typo in the example you were given, you can always on the Micropython REPL (e.g. in Thonny) do: 

`dir(display_driver)`

after program end or crash, and then it will tell you which attributes this object has. And troubleshoot from there.

sasakihin, 2024-12-10:
I tried it again with my laptop in the morning. After correcting two things, it works normally. I will post the corrected code for reference.
```
# build command
# python3 esp32 BOARD=ESP32_GENERIC_S3 DISPLAY=st7796 INDEV=ft6x36 --flash-size=16
# LVGL MicroPython 1.23.0 on 2024-12-06
# WT32-SC01 PLUS

from micropython import const

_WIDTH = const(320)
_HEIGHT = const(480)

_FREQ = const(40000000)
_D7 = const(15)
_D6 = const(16)
_D5 = const(17)
_D4 = const(18)
_D3 = const(8)
_D2 = const(3)
_D1 = const(46)
_D0 = const(9)

_WR = const(47)
_RS = const(4)
_DC = const(0)
_CS = const(6)
_BL = const(45)
_TE = const(47)

_TP_SDA = const(6)
_TP_SDL = const(5)
_TP_RST = const(4)
_TP_HST = const(0)
_TP_INT = const(7)
_TP_FREQ = const(400000)

_SD_MOSI = const(40)
_SD_MISO = const(38)
_SD_SCK = const(39)
_SD_CS = const(41)

_EXT_IO1 = const(10)
_EXT_IO2 = const(11)
_EXT_IO3 = const(12)
_EXT_IO4 = const(13)
_EXT_IO5 = const(14)
_EXT_IO6 = const(21)

import lcd_bus

display_bus = lcd_bus.I80Bus(
    wr=_WR,
    dc=_DC,
    cs=_CS,
    data0=_D0,
    data1=_D1,
    data2=_D2,
    data3=_D3,
    data4=_D4,
    data5=_D5,
    data6=_D6,
    data7=_D7,
    dc_data_high=True,
    dc_cmd_high=False,
    dc_idle_high=False,
    dc_dummy_high=False
)

# Use DMA transfers with double buffering for best performance. 
# The buffers should fit into the internal memory as this is 
# going to be the fastest memory available 
buf1 = display_bus.allocate_framebuffer(int(_WIDTH * _HEIGHT * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
buf2 = display_bus.allocate_framebuffer(int(_WIDTH * _HEIGHT * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import st7796
import lvgl as lv

display_driver = st7796.ST7796(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    frame_buffer1=buf1,
    frame_buffer2=buf2,
    reset_pin=_RS,
    
    # You may have to set this to STATE_LOW if you get a black screen
    reset_state=st7796.STATE_LOW, # WT32-SC01 PLUS use st7796.STATE_LOW
    
    backlight_pin=_BL,
    backlight_on_state=st7796.STATE_PWM,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display_driver.set_power(True)
display_driver.init()
display_driver.set_params(0x34)  #display_driver.send_params(0x34)  # TEOFF 
display_driver.set_color_inversion(True)

import i2c

import ft6x36

touch_bus = i2c.I2C.Bus(
    host=_TP_HST,
    sda=_TP_SDA,
    scl=_TP_SDL,
    freq=_TP_FREQ
)

touch_device = i2c.I2C.Device(
    bus=touch_bus,
    dev_id=ft6x36.I2C_ADDR,
    reg_bits=ft6x36.BITS
)

indev = ft6x36.FT6x36(device=touch_device)

import time
import task_handler

display_driver.set_rotation(lv.DISPLAY_ROTATION._90)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

lv.refr_now(scrn.get_display())

for i in range(1, 101, 1):
    display_driver.set_backlight(i)
    time.sleep_ms(3)

task_handler.TaskHandler()

```

closed 2024-12-10 by sasakihin