lvgl-micropython/lvgl_micropython #206: Failed to compile ESP32-S3 with hx8369 and gt911

217heidai, 2024-12-10:
```bash
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=8 DISPLAY=hx8369 INDEV=gt911
```
log:
```
2024-12-10T02:20:52.4505780Z Generating build files....
2024-12-10T02:20:52.4507100Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/py_api_drivers/frozen/display/display_driver_framework.py
2024-12-10T02:20:52.4509123Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/py_api_drivers/frozen/display/rgb_display_framework.py
2024-12-10T02:20:52.4511467Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/touch_calibration/touch_cal_data.py
2024-12-10T02:20:52.4513394Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/touch_calibration/touch_calibrate.py
2024-12-10T02:20:52.4515178Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/_indev_base.py
2024-12-10T02:20:52.4516832Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/button_framework.py
2024-12-10T02:20:52.4518532Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/encoder_framework.py
2024-12-10T02:20:52.4520190Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/keypad_framework.py
2024-12-10T02:20:52.4521824Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/pointer_framework.py
2024-12-10T02:20:52.4523370Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/py_api_drivers/fs_driver.py
2024-12-10T02:20:52.4524874Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/common_api_drivers/frozen/other/i2c.py
2024-12-10T02:20:52.4526564Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/common_api_drivers/frozen/other/io_expander_framework.py
2024-12-10T02:20:52.4528318Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/common_api_drivers/frozen/other/task_handler.py
2024-12-10T02:20:52.4529982Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/common_api_drivers/frozen/other/spi3wire.py
2024-12-10T02:20:52.4531629Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/common_api_drivers/indev/gt911_extension.py
2024-12-10T02:20:52.4533205Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/common_api_drivers/indev/gt911.py
2024-12-10T02:20:52.4534809Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/common_api_drivers/display/hx8369/_hx8369_init_type1.py
2024-12-10T02:20:52.4536442Z /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/api_drivers/common_api_drivers/display/hx8369/hx8369.py
2024-12-10T02:20:52.4537591Z Compiling....
2024-12-10T02:20:52.5450048Z Traceback (most recent call last):
2024-12-10T02:20:52.5451017Z   File "/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/make.py", line 225, in <module>
2024-12-10T02:20:52.5451750Z     mod.compile(*extra_args)
2024-12-10T02:20:52.5452941Z   File "/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/builder/esp32.py", line 1226, in compile
2024-12-10T02:20:52.5453748Z     build_sdkconfig(*args)
2024-12-10T02:20:52.5454568Z   File "/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/builder/esp32.py", line 1210, in build_sdkconfig
2024-12-10T02:20:52.5455360Z     base_config.extend(read_file(display_path).split('\n'))
2024-12-10T02:20:52.5455825Z TypeError: read_file() missing 1 required positional argument: 'file'
2024-12-10T02:20:52.5566444Z ##[error]Process completed with exit code 1.
```

kdschlosser, 2024-12-10:
I fixed this issue so it should compile properly now. You will need to delete your current clone of the repo and clone it again to make sure you get all of the changes made.

217heidai, 2024-12-11:
It looks like some error in `ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c` with latest commit.  
log:
```
/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c
/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c:120:36: error: expected ',' or ';' before 'static'
120 |     __attribute__((always_inline)) static inline void rotate_8bpp(const uint8_t *src, uint8_t *dst, uint32_t x_start, uint32_t y_start, uint32_t x_end, uint32_t y_end, uint32_t dst_width, uint32_t dst_height, uint8_t rotate)
|                                    ^~~~~~
/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c: In function 'rgb_bus_copy_task':
/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c:191:9: error: unknown type name 'copy_func_cb_t'
191 |         copy_func_cb_t func;
|         ^~~~~~~~~~~~~~
/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c:214:13: error: implicit declaration of function 'copy_pixels' [-Werror=implicit-function-declaration]
214 |             copy_pixels(
|             ^~~~~~~~~~~
/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c:191:24: warning: unused variable 'func' [-Wunused-variable]
191 |         copy_func_cb_t func;
|                        ^~~~
/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c: At top level:
/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c:324:10: warning: conflicting types for 'copy_pixels'; have 'void(uint8_t *, uint8_t *, uint32_t,  uint32_t,  uint32_t,  uint32_t,  uint32_t,  uint32_t,  uint32_t,  uint8_t)' {aka 'void(unsigned char *, unsigned char *, long unsigned int,  long unsigned int,  long unsigned int,  long unsigned int,  long unsigned int,  long unsigned int,  long unsigned int,  unsigned char)'}
324 |     void copy_pixels(uint8_t *dst, uint8_t *src, uint32_t x_start, uint32_t y_start,
|          ^~~~~~~~~~~
/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c:214:13: note: previous implicit declaration of 'copy_pixels' with type 'void(uint8_t *, uint8_t *, uint32_t,  uint32_t,  uint32_t,  uint32_t,  uint32_t,  uint32_t,  uint32_t,  uint8_t)' {aka 'void(unsigned char *, unsigned char *, long unsigned int,  long unsigned int,  long unsigned int,  long unsigned int,  long unsigned int,  long unsigned int,  long unsigned int,  unsigned char)'}
214 |             copy_pixels(
|             ^~~~~~~~~~~
/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c: In function 'copy_pixels':
/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c:347:17: error: implicit declaration of function 'rotate_8bpp' [-Werror=implicit-function-declaration]
347 |                 rotate_8bpp(src, dst, x_start, y_start, x_end, y_end, d_width, d_height, rotate);
|                 ^~~~~~~~~~~
/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c:349:17: error: implicit declaration of function 'rotate_16bpp' [-Werror=implicit-function-declaration]
349 |                 rotate_16bpp(src, dst, x_start, y_start, x_end, y_end, d_width, d_height, rotate);
|                 ^~~~~~~~~~~~
/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c:351:17: error: implicit declaration of function 'rotate_24bpp' [-Werror=implicit-function-declaration]
351 |                 rotate_24bpp(src, dst, x_start, y_start, x_end, y_end, d_width, d_height, rotate);
|                 ^~~~~~~~~~~~
/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c:353:17: error: implicit declaration of function 'rotate_32bpp' [-Werror=implicit-function-declaration]
353 |                 rotate_32bpp(src, dst, x_start, y_start, x_end, y_end, d_width, d_height, rotate);
|                 ^~~~~~~~~~~~
/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c: At top level:
/home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c:377:10: warning: conflicting types for 'rotate_8bpp'; have 'void(const uint8_t *, uint8_t *, uint32_t,  uint32_t,  uint32_t,  uint32_t,  uint32_t,  uint32_t,  uint8_t)' {aka 'void(const unsigned char *, unsigned char *, long unsigned int,  long unsigned int,  long unsigned int,  long unsigned int,  long unsigned int,  long unsigned int,  unsigned char)'}
377 |     void rotate_8bpp(const uint8_t *src, uint8_t *dst, uint32_t x_start, uint32_t y_start, uint32_t x_end, uint32_t y_end, uint32_t dst_width, uint32_t dst_height, uint8_t rotate)
[... cut ...]
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
ninja failed with exit code 1, output of the command is in the /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_6630 and /home/runner/work/micropython_esp32_firmware/micropython_esp32_firmware/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_6630
make: *** [Makefile:66: all] Error 1
```

kdschlosser, 2024-12-11:
Give it a try again. I fixed the errors above, dunno if more will pop up tho. Strange that it passed the CI..,..

0ut4t1m3, 2024-12-11:
I'm getting a similar error when using this build command:
`python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=rgb_display INDEV=gt911 --flash-size=8`

This is the first error thrown:

```
[1357/1960] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/home/mike/lvgl_micropython/ext_mod/lcd_bus/modlcd_bus.c.obj
In file included from /home/mike/lvgl_micropython/ext_mod/lcd_bus/lcd_types.h:7,
from /home/mike/lvgl_micropython/ext_mod/lcd_bus/modlcd_bus.h:4,
from /home/mike/lvgl_micropython/ext_mod/lcd_bus/modlcd_bus.c:2:
/home/mike/lvgl_micropython/lib/micropython/py/obj.h:348:23: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
348 | #define MP_ROM_PTR(p) (p)
|                       ^
/home/mike/lvgl_micropython/ext_mod/lcd_bus/modlcd_bus.c:292:48: note: in expansion of macro 'MP_ROM_PTR'
292 |     { MP_ROM_QSTR(MP_QSTR__pump_main_thread),  MP_ROM_PTR(&mp_lcd_bus__pump_main_thread_obj)       },
|                                                ^~~~~~~~~~
[1358/1960] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/home/mike/lvgl_micropython/ext_mod/
```

kdschlosser, 2024-12-11:
are you compiling on a mac?

0ut4t1m3, 2024-12-11:
No this is Ubuntu on wsl2, I've been building for a week or so with no issues it's only after 256a27a this started happening

kdschlosser, 2024-12-11:
Try it now and see what happens...

0ut4t1m3, 2024-12-11:
The warning is gone but it's still having issues, here's everything past the first error:
```
[1360/1960] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/home/mike/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c.obj
FAILED: esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/home/mike/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c.obj
/home/mike/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc -DBERKELEY_DB_CONFIG_FILE=\"/home/mike/lvgl_micropython/lib/micropython/extmod/berkeley-db/berkeley_db_config_port.h\" -DESP_PLATFORM -DFFCONF_H=\"/home/mike/lvgl_micropython/lib/micropython/lib/oofatfs/ffconf.h\" -DIDF_VER=\"11eaf41b\" -DLFS1_NO_ASSERT -DLFS1_NO_DEBUG -DLFS1_NO_ERROR -DLFS1_NO_MALLOC -DLFS1_NO_WARN -DLFS2_NO_ASSERT -DLFS2_NO_DEBUG -DLFS2_NO_ERROR -DLFS2_NO_MALLOC -DLFS2_NO_WARN -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMICROPY_ESP_IDF_4=1 -DMICROPY_HW_BOARD_NAME="\"Generic ESP32S3 module with Octal-SPIRAM\"" -DMICROPY_MODULE_FROZEN_MPY="(1)" -DMICROPY_PY_BTREE=1 -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS2=1 -DNDEBUG -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/mike/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/config -I/home/mike/lvgl_micropython/lib/micropython -I/home/mike/lvgl_micropython/lib/micropython/lib/berkeley-db-1.xx/include -I/home/mike/lvgl_micropython/ext_mod/lcd_bus -I/home/mike/lvgl_micropython/ext_mod/lcd_bus/esp32_include -I/home/mike/lvgl_micropython/lib/esp-idf/components/esp_lcd/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/esp_lcd/interface -I/home/mike/lvgl_micropython/lib -I/home/mike/lvgl_micropython/lib/lvgl -I/home/mike/lvgl_micropython/lib/micropython/ports/esp32 -I/home/mike/lvgl_micropython/lib/micropython/ports/esp32/boards/ESP32_GENERIC_S3 -I/home/mike/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT -I/home/mike/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble -I/home/mike/lvgl_micropython/lib/esp-idf/components/newlib/platform_include -I/home/mike/lvgl_micropython/lib/esp-idf/components/freertos/config/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/freertos/config/include/freertos -I/home/mike/lvgl_micropython/lib/esp-idf/components/freertos/config/xtensa/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/home/mike/lvgl_micropython/lib/esp-idf/components/freertos/esp_additions/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include/soc -I/home/mike/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/home/mike/lvgl_micropython/lib/esp-idf/components/esp_hw_support/port/esp32s3/. -I/home/mike/lvgl_micropython/lib/esp-idf/components/heap/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/log/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/soc/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/soc/esp32s3 -I/home/mike/lvgl_micropython/lib/esp-idf/components/soc/esp32s3/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/hal/platform_port/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/hal/esp32s3/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/hal/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/esp_rom/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/esp_rom/include/esp32s3 -I/home/mike/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3 -I/home/mike/lvgl_micropython/lib/esp-idf/components/esp_common/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/esp_system/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/esp_system/port/soc -I/home/mike/lvgl_micropython/lib/esp-idf/components/esp_system/port/include/private -I/home/mike/lvgl_micropython/lib/esp-idf/components/xtensa/esp32s3/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/xtensa/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/xtensa/deprecated_include -I/home/mike/lvgl_micropython/lib/esp-idf/components/lwip/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/lwip/include/apps -I/home/mike/lvgl_micropython/lib/esp-idf/components/lwip/include/apps/sntp -I/home/mike/lvgl_micropython/lib/esp-idf/components/lwip/lwip/src/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/lwip/port/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/lwip/port/freertos/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/lwip/port/esp32xx/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/mike/lvgl_micropython/lib/esp-idf/components/lwip/port/esp32xx/include/sys -I/home/mike/lvgl_micropython/lib/esp-idf/components/app_update/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/bootloader_support/include -I/home/mike/lvgl_micropython/lib/esp-idf/components/bootloader_support/bootloader_flash/include -I/home/mike/lvg [... cut ...]
ninja failed with exit code 1, output of the command is in the /home/mike/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_14506 and /home/mike/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_14506
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: Leaving directory '/home/mike/lvgl_micropython/lib/micropython/ports/esp32'
make: *** [Makefile:66: all] Error 1
```

0ut4t1m3, 2024-12-11:
If it helps at all I also tried compiling for another board with different drivers I've built before and had the same issue:
`python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM DISPLAY=st7796 INDEV=ft6x36 --flash-size=8 --usb-otg`

kdschlosser, 2024-12-11:
I will get it sorted out. I just made a change to the RGB Bus driver that should improve the speed with it and I need to fix my coding mistakes. What I am not aure of is why the CI is passing. I wonder if I disabled the ESP32 compiling portion of the CI. I am going to check on that.

217heidai, 2024-12-11:
> I will get it sorted out. I just made a change to the RGB Bus driver that should improve the speed with it and I need to fix my coding mistakes. What I am not aure of is why the CI is passing. I wonder if I disabled the ESP32 compiling portion of the CI. I am going to check on that.

It works with the last commit. tks.

kdschlosser, 2024-12-11:
good deal.

0ut4t1m3, 2024-12-11:
Working fine here for me too now, thank you.

closed 2025-01-05 by kdschlosser