lvgl-micropython/lvgl_micropython #208: centering text in dropdown button not working (lvgl #6842)

h-milz, 2024-12-10:
Has [this](https://github.com/lvgl/lvgl/issues/6842)  been addressed so far? in my recent clone of lvgl_micropython (from today), I see

```
        if(symbol_type == LV_IMAGE_SRC_SYMBOL) {
            symbol_area.y1 = obj->coords.y1 + top;
            symbol_area.y2 = symbol_area.y1 + symbol_h - 1;
            symbol_dsc.text = dropdown->symbol;
            lv_draw_label(layer, &symbol_dsc, &symbol_area);
        }

...

    lv_area_t txt_area;
    txt_area.y1 = obj->coords.y1 + top;
    txt_area.y2 = txt_area.y1 + size.y;
    /*Center align the text if no symbol*/
    if(dropdown->symbol == NULL) {
        txt_area.x1 = obj->coords.x1 + (lv_obj_get_width(obj) - size.x) / 2;
        txt_area.x2 = txt_area.x1 + size.x;
    }
```
and the button text is not centered vertically but top-mid. git status says "HEAD detached at e1c0b21".

kdschlosser, 2024-12-10:
no it has not I need to do some work on the binding script in order to get 9.2.x to compile properly. I have not been motivated to do that because of how it is written. It is a bunch of patch work that very easily breaks when messed with. I have thought about writing ti over again but it is quite a large undertaking to do that.

h-milz, 2024-12-10:
I hear your frustration. In fact, your repo isn't in any way different from lvgl:master. Very strange. 

I'll re-open the ticket there I suppose.

kdschlosser, 2024-12-12:
OK so I wanted to give you an update on this. I have altered the build script so it should not work with a newer version of LVGL. I still need to go in and update the submodule but the code generation portion of it should be working correctly. There are some things that I want to add in like inclusing the private header files in LVGL. While this does lead to a larger firmware it offers the greatest flexibility when doing custom stuff in LVGL.

I have not yet compared the binary sizes that get created between the 2 builds I am going to do that now and see what it looks like.

h-milz, 2024-12-13:
OK this appears to work now in LVGL 9.2.2.

kdschlosser, 2024-12-14:
Ok cool. I am going to close this issue then.