lvgl-micropython/lvgl_micropython #209: building for ESP32S3, rgb_display and gt911 works again, but then: bootloop

h-milz, 2024-12-11:
Just pulled and compiled the repo, but after flashing and resetting I'm stuck in a bootloop: 

> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
> 
> Core  0 register dump:
> PC      : 0x420ae2c2  PS      : 0x00060030  A0      : 0x820aec6d  A1      : 0x3fcdaca0  
> A2      : 0x3fcdb0c8  A3      : 0x00000000  A4      : 0x000bb800  A5      : 0x00000808  
> A6      : 0x00000000  A7      : 0x3fca0628  A8      : 0x820a8869  A9      : 0x00000000  
> A10     : 0x00000000  A11     : 0x00000808  A12     : 0x3c21470c  A13     : 0x00000003  
> A14     : 0x00000008  A15     : 0x00000000  SAR     : 0x00000011  EXCCAUSE: 0x0000001c  
> EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  
> 
> 
> Backtrace: 0x420ae2bf:0x3fcdaca0 0x420aec6a:0x3fcdacc0 0x420036bb:0x3fcdad40
> 
> 
> 
> 
> ELF file SHA256: f67da3c28
> 
> Rebooting...
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0xc (RTC_SW_CPU_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
> Saved PC:0x40376b7c
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3820,len:0x1108
> load:0x403c9700,len:0x4
> load:0x403c9704,len:0xbdc
> load:0x403cc700,len:0x2f4c
> entry 0x403c989c
> E (432) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
> E (432) esp_psram: PSRAM enabled but initialization failed. Bailing out.
> 

The build cmdline, unchanged from previous build runs:

`python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=4 DISPLAY=rgb_display INDEV=gt911 clean 
`

kdschlosser, 2024-12-11:
can you decode the backtrace for me so I know where the issue is.

h-milz, 2024-12-11:
I must admit that I have no idea how to. 

What I find strange is that it complains about quad psram although I compiled the image with octal, as before. And yeah, on the N4R8, the SPIRAM is octal.

h-milz, 2024-12-12:
OK, disregard all after good morning. It was a layer 8 problem. I had the wrong command line option for SPIRAM_OCT as shown above. It should be MICROPY_BOARD_VARIANT. 🤦 

But now I get a new one when building. Pristine git clone 5 minutes ago, then

    python3 make.py esp32 BOARD=ESP32_GENERIC_S3 MICROPY_BOARD_VARIANT=SPIRAM_OCT --flash-size=4 DISPLAY=rgb_display INDEV=gt911

or 

    python3 make.py unix DISPLAY=sdl_display INDEV=sdl_pointer

and then: 

>   File "/home/hm/local/lvgl_micropython/gen/fixed_gen_json.py", line 14, in <module>
>     import create_fake_lib_c  # NOQA
>     ^^^^^^^^^^^^^^^^^^^^^^^^
> ModuleNotFoundError: No module named 'create_fake_lib_c'

and right, I do not `find . -name "create_fake_lib_c.*"`  anywhere. The directory `gen/fake_libc/` is there, though, so it may just be that the single file is somehow missing. 

BTW what stands "# NOQA" for?

kdschlosser, 2024-12-12:
I am going to look into this issue over the next day. I did make some changes to the RGB driver and those changes may be causing this problem.

closed 2024-12-14 by h-milz

kdschlosser, 2024-12-14:
Is this fixed?

h-milz, 2024-12-14:
Yes. Sorry for not commenting - didn't work in the github Android app.

kdschlosser, 2024-12-14:
what do you mean by the github android app?

h-milz, 2024-12-14:
The GitHub app for Android. When you add a comment, it re-opens you last comment instead of opening a new one. I use the app mostly for 2FA but sometimes also for communicating.