lvgl-micropython/lvgl_micropython #211: Cannot create SPI instance on ESP32 S3

MussabMehboob, 2024-12-12:
I am using an ESP32 S3 with PSRAM 8 MB and flash 16 MB. I want to test the LVGL on ST 7789(240*240) but I am seeing this error when I am creating an SPI instance in thonny.
```
from machine import SPI, Pin
from micropython import const

import lcd_bus
import st7789
import lvgl

_WIDTH = const(240)
_HEIGHT = const(240)

_SPI_HOST = const(2)
_SPI_SCK = const(10)
_SPI_MOSI = const(11)
_SPI_MISO = const(12)

_LCD_FREQ = const(400_000)
_LCD_DC = const(8)
_LCD_CS = const(9)
_LCD_RST = const(14)
_LCD_BACKLIGHT = const(2)

spi_bus = SPI(
    _SPI_HOST,
    _LCD_FREQ,
    mosi=_SPI_MOSI,
    miso=_SPI_MISO,
    sck=_SPI_SCK,
    cs=_LCD_CS
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_LCD_DC,
    cs=_LCD_CS
)

    
display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7796.BYTE_ORDER_RGB,
    rgb565_byte_swap=True,
)
MPY: soft reboot
Traceback (most recent call last):
  File "boot.py", line 30, in <module>
TypeError: can't create 'SPI' instances
Traceback (most recent call last):
  File "<stdin>", line 30, in <module>
TypeError: can't create 'SPI' instances
```
I am building with this command:

`python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7789 INDEV=gt911 --flash-size=16`

I even tried using SoftSPI that worked with the compilation but I started getting memory initialization errors.I have tried to resolve those issues as said in the issue#66(https://github.com/lvgl-micropython/lvgl_micropython/issues/66) but it didn't worked for me.
Thank you!

kdschlosser, 2024-12-12:
This is wrong.

```
spi_bus = SPI(
    _SPI_HOST,
    _LCD_FREQ,
    mosi=_SPI_MOSI,
    miso=_SPI_MISO,
    sck=_SPI_SCK,
    cs=_LCD_CS
)
```

it should read

```
spi_bus = SPI.Bus(
    host=_SPI_HOST,
    freq=_LCD_FREQ,
    mosi=_SPI_MOSI,
    miso=_SPI_MISO,
    sck=_SPI_SCK,
    cs=_LCD_CS
)
```

MussabMehboob, 2024-12-13:
Is having this error normal while compiling for ESP32 s3?
`[1954/1962] Building C object CMakeFiles/micropython.elf.dir/home/mussab/Desktop/lvgl_micropython/ex[1955/1962] Building C object CMakeFiles/micropython.elf.dir/home/mussab/Desktop/lvgl_micropython/ex[1956/1962] Building C object CMakeFiles/micropython.elf.dir/home/mussab/Desktop/lvgl_micropython/ex[1957/1962] Building C object CMakeFiles/micropython.elf.dir/home/mussab/Desktop/lvgl_micropython/ex[1958/1962] Building C object CMakeFiles/micropython.elf.dir/home/mussab/Desktop/lvgl_micropython/li[1959/1962] Building C object CMakeFiles/micropython.elf.dir/home/mussab/Desktop/lvgl_micropython/li[1960/1962] Linking CXX executable micropython.elf                                                  [1961/1962] Generating binary image from built executable
esptool.py v4.8.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated /home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/micropython.bin
[1962/1962] cd /home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/esp-idf/esptool_py && /root/.espressif/python_env/idf5.2_py3.10_env/bin/python /home/mussab/Desktop/esp-idf-v5.2.1/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/partition_table/partition-table.bin /home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/micropython.bin
FAILED: esp-idf/esptool_py/CMakeFiles/app_check_size /home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/esp-idf/esptool_py/CMakeFiles/app_check_size
cd /home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/esp-idf/esptool_py && /root/.espressif/python_env/idf5.2_py3.10_env/bin/python /home/mussab/Desktop/esp-idf-v5.2.1/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/partition_table/partition-table.bin /home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/micropython.bin
Error: app partition is too small for binary micropython.bin size 0x25a3c0:
- Part 'factory' 0/0 @ 0x10000 size 0x25a000 (overflow 0x3c0)
ninja: build stopped: subcommand failed.
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: Leaving directory '/home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32'
ninja failed with exit code 1, output of the command is in the /home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_45703 and /home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_45703
make: *** [Makefile:66: all] Error 1

***** Resizing Partition *****

***** Running build again *****

make: Entering directory '/home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32'
idf.py -D MICROPY_BOARD=ESP32_GENERIC_S3 -D MICROPY_BOARD_DIR="/home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/boards/ESP32_GENERIC_S3" -DUSER_C_MODULES=../../../../../ext_mod/micropython.cmake -D MICROPY_FROZEN_MANIFEST=/home/mussab/Desktop/lvgl_micropython/build/manifest.py -D MICROPY_BOARD_VARIANT=SPIRAM_OCT -B build-ESP32_GENERIC_S3-SPIRAM_OCT build || (echo -e "See \033[1;31mhttps://github.com/micropython/micropython/wiki/Build-Troubleshooting\033[0m"; false)
Executing action: all (aliases: build)
Running ninja in directory /home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT
Executing "ninja all"...
-- Building ESP-IDF components for target esp32s3
NOTICE: Processing 4 dependencies:
NOTICE: [1/4] espressif/esp_tinyusb (1.0.3)
NOTICE: [2/4] espressif/mdns (1.1.0)
NOTICE: [3/4] espressif/tinyusb (0.17.0~1)
NOTICE: [4/4] idf (5.2.1)
-- Project sdkconfig file /home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/sdkconfig
Loading defaults file /home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/sdkconfig.combined...
/home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/sdkconfig.combined:100 CONFIG_ESP32_WIFI_IRAM_OPT was replaced with CONFIG_ESP_WIFI_IRAM_OPT
/home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/sdkconfig.combined:101 CONFIG_ESP32_WIFI_RX_IRAM_OPT was replaced with CONFIG_ESP_WIFI_RX_IRAM_OPT
/home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/sdkconfig.combined:148 CONFIG_BT_NIMBLE_TASK_STACK_SIZE was replaced with CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE
/tmp/kconfgen_tmp1sjvpjlj:167 line was updated to CONFIG_ESPTOOLPY_FLASHSIZE_4MB=n
/tmp/kconfgen_tmp1sjvpjlj:169 line was updated to CONFIG_ESPTOOLPY_FLASHSIZE_16MB=n
/tmp/kconfgen_tmp1sjvpjlj:173 line was updated to CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_40=n
/tmp/kconfgen_tmp1sjvpjlj:174 line was updated to CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_80=n
/tmp/kconfgen_tmp1sjvpjlj:175 line was updated to CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_160=n
[... cut ...]
[1/270] cd /home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/esp-idf/main_esp32s3 && /usr/bin/cmake -E make_directory /home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/genhdr && /root/.espressif/python_env/idf5.2_py3.10_env/bin/python /home/mussab/Desktop/lvgl_micropython/lib/micropython/py/makeversionhdr.py /home/mussab/Desktop/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-S[2/12] Generating ../../partition_table/partition-table.bin                                                                                                         `

kdschlosser, 2024-12-13:
The resizing is normal. 

I would need to see the whole script you are running along with the output from running that script in order to determine where the issue might be.

kdschlosser, 2024-12-13:
I can also tell you that you are running an old version of the library as well.

you are running with esp-idf v5.2.1 when the version that is packaged with the library is 5.2.3.

In order to update properly you need to delete your local copy lvgl_micropython and clone it again.

closed 2024-12-17 by kdschlosser