lvgl-micropython/lvgl_micropython #212: compile error after LVGL upgrade step

h-milz, 2024-12-12:
I figure I cloned too soon this morning: 

Pristine git clone, then

    python3 make.py esp32 BOARD=ESP32_GENERIC_S3 MICROPY_BOARD_VARIANT=SPIRAM_OCT --flash-size=4 DISPLAY=rgb_display INDEV=gt911

or 

    python3 make.py unix DISPLAY=sdl_display INDEV=sdl_pointer

and then: 

>   File "/home/hm/local/lvgl_micropython/gen/fixed_gen_json.py", line 14, in <module>
>     import create_fake_lib_c  # NOQA
>     ^^^^^^^^^^^^^^^^^^^^^^^^
> ModuleNotFoundError: No module named 'create_fake_lib_c'

and right, I do not `find . -name "create_fake_lib_c.*"`  anywhere. The directory `gen/fake_libc/` is there, though, so it may just be that the single file is somehow missing. 

BTW what stands "# NOQA" for?

kdschlosser, 2024-12-12:
the `# NOQA` stops my IDE from checking a line for errors. some things are dynamic so and the IDE is not able to follow what is happening correctly in order for intellisense to work properly. So I disable the IDE's intellisense for that line using that marker.

kdschlosser, 2024-12-12:
I wasn't done quite yet with the LVGL update. I have only done step one of it. I have to double check and see if I forgot to copy something over to the repo from the changes I made.

kdschlosser, 2024-12-12:
OK I just finished updating the submodule for LVGL. It should be on version 9.2.2 now.

kdschlosser, 2024-12-12:
This should now be fixed. the CI runs properly.

You need to install doxygen by running `apt-get install doxygen` if on a Debian based OS or `brew install doxygen` if running on Darwin (macOS)

kdschlosser, 2024-12-12:
You need to do a fresh clone of the repo for the new IDF and LVGL versions to be used.

closed 2024-12-14 by h-milz

kdschlosser, 2024-12-14:
You are all set with this? Everything is running properly?

h-milz, 2024-12-14:
Yes. Sorry for not commenting - didn't work in the github Android app.