lvgl-micropython/lvgl_micropython #214: Compile error

spienz, 2024-12-12:
Hy,

also compile error with newest git pull

```

root@debian:/home/spienz/lvgl_micropython# git pull
remote: Enumerating objects: 35, done.
remote: Counting objects: 100% (35/35), done.
remote: Compressing objects: 100% (11/11), done.
remote: Total 25 (delta 18), reused 19 (delta 13), pack-reused 0 (from 0)
Entpacke Objekte: 100% (25/25), 3.10 KiB | 211.00 KiB/s, fertig.
Von https://github.com/lvgl-micropython/lvgl_micropython
   80d88cc..311e751  main               -> origin/main
   96aac81..a4ee6df  test_updating_lvgl -> origin/test_updating_lvgl
Anfordern des Submoduls lib/lvgl
Von https://github.com/lvgl/lvgl
 * [neues Tag]       v9.1.0     -> v9.1.0
Von https://github.com/lvgl/lvgl
 * branch            7f07a129e8d77f4984fff8e623fd5be18ff42e74 -> FETCH_HEAD
Aktualisiere 80d88cc..311e751
Fast-forward
 .github/workflows/macOS.yml | 1 +
 .github/workflows/unix.yml  | 2 +-
 .gitmodules                 | 6 +++---
 README.md                   | 8 ++++++++
 gen/python_api_gen_mpy.py   | 1 +
 lib/lvgl                    | 2 +-
 6 files changed, 15 insertions(+), 5 deletions(-)
root@debian:/home/spienz/lvgl_micropython# python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=4
Cleaning build....
Getting ESP-IDF build Environment
make: Verzeichnis „/home/spienz/lvgl_micropython/lib/micropython/mpy-cross“ wird betreten
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
rm -f build/mpy-cross
rm -f build/mpy-cross.map
rm -rf build
make: Verzeichnis „/home/spienz/lvgl_micropython/lib/micropython/mpy-cross“ wird verlassen
make: Verzeichnis „/home/spienz/lvgl_micropython/lib/micropython/ports/esp32“ wird betreten
idf.py -D MICROPY_BOARD=ESP32_GENERIC_S3 -D MICROPY_BOARD_DIR="/home/spienz/lvgl_micropython/lib/micropython/ports/esp32/boards/ESP32_GENERIC_S3"  -D MICROPY_BOARD_VARIANT=SPIRAM_OCT -B build-ESP32_GENERIC_S3-SPIRAM_OCT -p /dev/ttyUSB0 -b 460800 fullclean
Executing action: fullclean
Directory '/home/spienz/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.
make: Verzeichnis „/home/spienz/lvgl_micropython/lib/micropython/ports/esp32“ wird verlassen
make: *** [Makefile:78: clean] Fehler 2
setting up ESP-IDF v5.2.0
this might take a while...
Detecting the Python interpreter
Checking "python3" ...
Python 3.11.2
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Selected targets are: esp32, esp32s2, esp32h2, esp32p4, esp32c3, esp32s3, esp32c6, esp32c2
Current system platform: linux-amd64
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@14.2_20240403 (already installed)
Skipping riscv32-esp-elf-gdb@14.2_20240403 (already installed)
Skipping xtensa-esp-elf@esp-13.2.0_20230928 (already installed)
Skipping riscv32-esp-elf@esp-13.2.0_20230928 (already installed)
Skipping esp32ulp-elf@2.35_20220830 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20240821 (already installed)
Skipping esp-rom-elfs@20230320 (already installed)
Installing Python environment and packages
Python 3.11.2
pip 24.3.1 from /root/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages/pip (python 3.11)
Requirement already satisfied: pip in /root/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages (24.3.1)
Requirement already satisfied: setuptools in /root/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages (71.0.0)
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
Requirement already satisfied: setuptools in /root/.espressif/python_env/idf5.2_py3.11_env/lib/python3.11/site-packages (from -r /home/spienz/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 3)) (71.0.0)
[... 225 lines cut ...]
File "/home/spienz/lvgl_micropython/gen/python_api_gen_mpy.py", line 134, in <module>
import fixed_gen_json  # NOQA
^^^^^^^^^^^^^^^^^^^^^
File "/home/spienz/lvgl_micropython/gen/fixed_gen_json.py", line 14, in <module>
import create_fake_lib_c  # NOQA
^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'create_fake_lib_c'
OUTPUT:
RESULT: 1
CMake Error at /home/spienz/lvgl_micropython/ext_mod/lvgl/micropython.cmake:43 (message):
Failed to generate
/home/spienz/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/lv_mp.c
Call Stack (most recent call first):
/home/spienz/lvgl_micropython/ext_mod/micropython.cmake:6 (include)
/home/spienz/lvgl_micropython/lib/micropython/py/usermod.cmake:42 (include)
esp32_common.cmake:22 (include)
main_esp32s3/CMakeLists.txt:11 (include)
cmake failed with exit code 1, output of the command is in the /home/spienz/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_5288 and /home/spienz/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_5288
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: Verzeichnis „/home/spienz/lvgl_micropython/lib/micropython/ports/esp32“ wird verlassen
make: *** [Makefile:66: all] Fehler 1

``` 
regards 
sebastian

spienz, 2024-12-12:
Tryed 

git submodule update --init --recursive won't work

after a complete new clone it works. 

don't kno why the git mechanism isn't working prob. but ok now it work

THX

Reagards

kdschlosser, 2024-12-12:
don't update anything. Delete the cloned repo and clone it again.

DO NOT add any submodule update commands, or anything. The build script takes care of what need to be done to get everything onto your system to compile. Don't do anything other than

```
git clone https://github.com/lvgl-micropython/lvgl_micropython
cd lvgl_micropython
python3 make.py esp32 etc....
```

center789, 2024-12-13:
You need to install Doxygen using apt-get install doxygen. Your issue is that it’s missing.

closed 2024-12-17 by kdschlosser