lvgl-micropython/lvgl_micropython #216: Micropy version in MICROPY_BANNER_NAME_AND_VERSION

h-milz, 2024-12-14:
In `builder/__init__.py:set_mp_version()`, the Micropython banner is fixed to version 1.23. However, I cannot see anywhere that a specific branch is checked out. Instead, the actual version can be retrieved from `lib/micropython/py/mpconfig.h`, which means the banner does not reflect the actual version number of an image. 

Also, the current LVGL version is in `lib/lvgl/lv_version.h`.

So the banner could read e.g. 

> LVGL 9.2.2 on MicroPython 3.4.0 on 14-12-24

May I suggest changing it this way? Such a change only affects `builder/__init__.py:set_mp_version()`.  Shall I make a PR?

h-milz, 2024-12-14:
Now that I think of it, version number 1.23 is referring to the version of your fork, right?

kdschlosser, 2025-01-05:
I believe I have the banner all sorted out now. let me know if it's not. I am going to close this issue, you can reopen it if the problem still exists.

closed 2025-01-05 by kdschlosser