lvgl-micropython/lvgl_micropython #217: Build fails for samd

h-milz, 2024-12-14:
I have 2 Adafruit Feather M4 Express parts here and a ST7796/XPT2046 SPI display and would like to test. Alas, although "samd" is a valid choice in make.py, it is missing in the if-else construct in main. Also, `builder/samd.py` does not exist. Was it omitted for a reason?

kdschlosser, 2024-12-14:
I know it doesn't build, it was added for future development.  I need someone that is familiar with that MCU to be able to write the bus drivers that will work on them. It needs to be written in C code.

h-milz, 2024-12-14:
Hmmm, I had hoped it would be sufficient to provide a `samd.py`. Which bus drivers are you referring to? 

In this case it just deserves to be documented in the README.md I suppose.

kdschlosser, 2024-12-14:
SPI, I8080, RGB. those bus drivers.

I have in place a bitbang I8080 driver and I can leverage the SPI to use as a driver as well. I would prefer a hardware solution instead of doing a bitbang. I don't even know if the bitbang driver even works... If the MCU supports DMA memory transfers I would also like to have that implemented as well.

h-milz, 2024-12-15:
Not a bug, taking this to the discussion section.

closed 2024-12-15 by h-milz