lvgl-micropython/lvgl_micropython #218: Build fails for rp2

h-milz, 2024-12-14:
**Describe the bug**
The build fails when compiling `lib/micropython/ports/rp2/machine_spi.c` when doing 

`python3 make.py rp2 BOARD=ADAFRUIT_FEATHER_RP2040 DISPLAY=st7796 INDEV=xpt2046 clean
`
**If you are having a problem compiling paste the build output below. (NO SCREENSHOTS)**

attached - it is just too long. 

Do you think this is an upstream problem? 

[rp2-fail.txt](https://github.com/user-attachments/files/18135626/rp2-fail.txt)

kdschlosser, 2024-12-14:
I know it doesn't build, it was added for future development. I need someone that is familiar with that MCU to be able to write the bus drivers that will work on it. It needs to be written in C code.

h-milz, 2024-12-15:
Not a bug, taking this to the discussion section #224.

closed 2024-12-15 by h-milz