lvgl-micropython/lvgl_micropython #219: Build fails for raspberry_pi

h-milz, 2024-12-14:
**Describe the bug**
Compile fails when trying to include SDL.h. I assume the proper include path is missing from the make command. The German error message says "file or directory not found". 

> machine_sdl.c:2:10: fatal error: SDL.h: Datei oder Verzeichnis nicht gefunden
> 2 | #include "SDL.h"
> |          ^~~~~~~
> compilation terminated.
> Command '['gcc', '-E', '-I/home/hm/local/lvgl_micropython/ext_mod/lcd_bus', '-I/home/hm/local/lvgl_micropython/ext_mod/lcd_bus/common_include', '-I/home/hm/local/lvgl_micropython/ext_mod/lcd_bus/sdl_bus', '-I/home/hm/local/lvgl_micropython/lib/lvgl', '-I/home/hm/local/lvgl_micropython/lib', '-DMICROPY_VFS_FAT=1', '-DMICROPY_VFS_LFS1=1', '-DMICROPY_VFS_LFS2=1', '-DMICROPY_PY_SSL=1', '-DMBEDTLS_CONFIG_FILE="mbedtls/mbedtls_config_port.h"', '-DMICROPY_SSL_MBEDTLS=1', '-I../../lib/mbedtls/include', '-DBERKELEY_DB_CONFIG_FILE="extmod/berkeley-db/berkeley_db_config_port.h"', '-DMICROPY_PY_BTREE=1', '-DFFCONF_H="lib/oofatfs/ffconf.h"', '-DLFS1_NO_MALLOC', '-DLFS1_NO_DEBUG', '-DLFS1_NO_WARN', '-DLFS1_NO_ERROR', '-DLFS1_NO_ASSERT', '-DLFS2_NO_MALLOC', '-DLFS2_NO_DEBUG', '-DLFS2_NO_WARN', '-DLFS2_NO_ERROR', '-DLFS2_NO_ASSERT', '-I../../lib/berkeley-db-1.xx/include', '-I.', '-I../..', '-Ibuild-standard', '-I../../shared/readline', '-Wall', '-Werror', '-Wextra', '-Wno-unused-parameter', '-Wpointer-arith', '-Wdouble-promotion', '-Wfloat-conversion', '-std=gnu99', '-DUNIX', '-Os', '-DNDEBUG', '-fdata-sections', '-ffunction-sections', '-Ivariants/standard', '-Wno-sign-compare', '-Wno-unused-function', '-Wno-double-promotion', '-Wno-unused-command-line-argument', '-Wno-missing-field-initializers', '-g', '-U', '_FORTIFY_SOURCE', '-DMICROPY_USE_READLINE=1', '-DMICROPY_PY_TERMIOS=1', '-DMICROPY_PY_SOCKET=1', '-DMICROPY_PY_THREAD=1', '-DMICROPY_PY_THREAD_GIL=0', '-DMICROPY_PY_FFI=1', '-DMPZ_DIG_SIZE=16', '-DMICROPY_ROM_TEXT_COMPRESSION=1', '-DNO_QSTR', 'modffi.c', 'modjni.c', 'machine_timer.c', 'machine_sdl.c', '../../shared/runtime/gchelper_generic.c', '../../shared/timeutils/timeutils.c', '../../shared/readline/readline.c']' returned non-zero exit status 1.
> make: Verzeichnis „/home/hm/local/lvgl_micropython/lib/micropython/ports/unix“ wird verlassen
> make: *** [../../py/mkrules.mk:122: build-standard/genhdr/qstr.i.last] Fehler 1
> make: *** Datei „build-standard/genhdr/qstr.i.last“ wird gelöscht
>

kdschlosser, 2024-12-14:
I had started dinking around with getting a raspberry pi port working. I honestly don't remember where I left off with it. I believe the port needs to be adjusted to compile SDL2.

picpic020960, 2024-12-15:
Maybe other error before,when the scrip copy the SDL[2] include to the build folder.
Build on PI 4 OK with raspberry pi OS (64 bits)