lvgl-micropython/lvgl_micropython #221: Calls to socket.getaddrinfo causing crash on ESP32-S3

0ut4t1m3, 2024-12-14:
Hi mate,

I've been unable to successfully run socket.getaddrinfo since the upgrades to ESP-IDF. I was having issues with hard resets during NTP updates and have been able to track it down to specifically this command. Bare minimum test and output below:

Build command:
`python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=rgb_display INDEV=gt911 --flash-size=8`
```python
import network, socket, time
wifi = network.WLAN(network.STA_IF)
wifi.active(True)

if not wifi.isconnected():
    wifi.connect("WiFi_APN", "WiFi_Password")

while not wifi.isconnected():
    time.sleep(1)

addr = socket.getaddrinfo("google.com", 80)[0][-1] # <-- crashes here
```
```
LVGL MPY version : 1.23.0 on 2024-12-14
IDF version : c9763f62
Machine     : Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4005544b  PS      : 0x00060930  A0      : 0x820d256d  A1      : 0x3fcbf020  
A2      : 0x00000000  A3      : 0x3c2d9248  A4      : 0x3fcbf070  A5      : 0x3fcbf0c0  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x820d4088  A9      : 0x3fcbefd0  
A10     : 0x00000000  A11     : 0x3fca6140  A12     : 0x00000001  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0x3fca6168  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x40055448:0x3fcbf020 0x420d256a:0x3fcbf030 0x420d2602:0x3fcbf070 0x420b2e73:0x3fcbf0f0 0x420b999d:0x3fcbf120 0x420b9a65:0x3fcbf140 0x4037a50e:0x3fcbf160 0x420b2e0b:0x3fcbf200 0x420b999d:0x3fcbf260 0x420b99b2:0x3fcbf280 0x420c7913:0x3fcbf2a0 0x420c7a56:0x3fcbf330 0x420cf0b6:0x3fcbf380

ELF file SHA256: 318a9af5d
```

I've done some looking around and found [this](https://github.com/micropython/micropython/issues/15841) issue and after applying the changes in 611d8f9ce82ab5e04fb86ab6cfc28d5ed98f33bf the command no longer causes a crash. I know you're busy so no problems if you don't want to bump uPy but just thought to let you know in case it's easy to cherry pick.

Mike

kdschlosser, 2024-12-14:
I will be updating the MicroPython version to 1.24 within the next couple of days. It will be fixed when I do that.

kdschlosser, 2024-12-17:
I am going to do the update later on today.

0ut4t1m3, 2024-12-17:
No problems, I'm happy to test when you're ready.

0ut4t1m3, 2024-12-17:
I'm having some trouble getting the build to kick off, I'm getting this consistently with different build commands:

```
-- Adding linker script /home/mike/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /home/mike/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /home/mike/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /home/mike/lvgl_micropython/lib/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
Including User C Module(s) from ../../../../../ext_mod/micropython.cmake
-- Configuring incomplete, errors occurred!
Found User C Module(s): usermod_lcd_bus, usermod_lvgl, lvgl_interface, usermod_lcd_utils
CMake Error at /home/mike/lvgl_micropython/lib/micropython/extmod/extmod.cmake:115 (message):
MICROPY_PY_BTREE is enabled but the berkeley-db submodule is not initialised.
Run 'make BOARD=ESP32_GENERIC_S3 submodules'
Call Stack (most recent call first):
esp32_common.cmake:23 (include)
main_esp32s3/CMakeLists.txt:13 (include)
cmake failed with exit code 1, output of the command is in the /home/mike/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_5669 and /home/mike/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_5669
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: Leaving directory '/home/mike/lvgl_micropython/lib/micropython/ports/esp32'
make: *** [Makefile:71: all] Error 1
```

0ut4t1m3, 2024-12-17:
Looks like all the folders in micropython/lib are empty for me, uP issues seem to suggest running make submodules but I assume we should still be avoiding that?

0ut4t1m3, 2024-12-17:
Yep that got the build working, thank you. The bug with using socket is now gone.

kdschlosser, 2024-12-18:
sweet,

h-milz, 2024-12-18:
> MicroPython is actually broken in release 1.24.1. I did manage to work around the problem. give it a try and see if it works for you this go around.

Indeed. I'm currently trying to get my WPA2 Enterprise patch to run with a vanilla MPY build using the idf.py stuff as documented (hey, it actually works once you've figured out how!). With ESP-IDF 5.2.2 and MPY master, when connecting to my home AP, I get consistent 

`E (3073) wifi:sta is connecting, return error`

errors, and MPY soft reboots in a loop when in idf.py monitor. But all works fine with MPY v1.23.0. 

`<rant>` I did not bother to find out when the error was introduced. :-(   Compilation breaks with ESP-IDF 5.2.3 or 5.3.0 and MPY v1.23.0 although they say these are supported versions. With ESP-IDF 5.3.2 compilation doesn't work at all because ESP choose to rename various CONFIG_ESP32 variables to CONFIG_ESP. This is indeed a moving target. `</rant>` 

Going to uni now, I wonder how my code works there.

closed 2025-01-05 by kdschlosser